supermerill/SuperSlicer #1069: Avoid crossing perimeters doesn't works normaly

ZhenyaKa, 2021-04-08:
### Version
Version 2.3.55

Compare with Cura slicing paths.
### Behavior
Head cross perimeters and option doesn't affect to that.
I tried different detour length - result the same

[Муфта тефаль-crossing.3mf.zip](https://github.com/supermerill/SuperSlicer/files/6276593/-crossing.3mf.zip)

[image]
[image]

ZhenyaKa, 2021-04-08:
I find when this issue appears: when option Extra wipe for external perimeter is not zero in Print setting-Extruder section.

supermerill, 2021-04-11:
#1061

supermerill, 2021-04-11:
on 2.3.55.5 nigtlhy:
[image]

ZhenyaKa, 2021-04-11:
You didn't switch "Extra wipe for external perimeter" option ON!!!

supermerill, 2021-04-11:
> You didn't switch "Extra wipe for external perimeter" option ON!!!

Hey, calm down! How can i have known that? It isn't in your project file and didn't talk about it... (and it's not a switch but a number).

> And your perimeters start points isn't RANDOM

not needed to replicate and your screenshot shows "cost-based" and not "random".

replicated:
[image]

ZhenyaKa, 2021-04-11:
> > You didn't switch "Extra wipe for external perimeter" option ON!!!
> 
> Hey, calm down! How can i have known that? It isn't in your project file and didn't talk about it... (and it's not a switch but a number).
I wrote about that in my second message
[image]

closed 2022-02-13 by supermerill