supermerill/SuperSlicer #1070: [Feature Request] Upload to MainsailOS

wildjokerss, 2021-04-08:
### Version
2.3.55.5

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Klipper + Mainsail OPS

### Behavior
* _Describe the problem_ Upload g-code does not work to mainsail OS
* _Steps needed to reproduce the problem_ Have a printer running klipper and use mainsail OS instead of octoprint

* _Expected Results_ the upload to complete and start printing
* _Actual Results_ connection error

I believe this is due to having the updated API, although it is best to have updated, this does not work with mainsail that uses a older octoprint API or something like that. Is it possible to add a option to SuperSlicer that allows us to use the old octoprint API so we can upload directly to Mainsail? It is a bit tedious to have to save the g-code and upload manually. I also noted that this works with ideamaker, guess they are using a older API.

Feel free to reach out if you need more clarification from me.

Xploder, 2021-04-09:
You just have to define [octoprint_compat] in your moonraker.conf and point SuperSlicer to your moonraker service.
Its default port is 7125, so put in http://[RaspiIP]:7125 in your printer settings and you're good to go. 

No API Key needed.

wildjokerss, 2021-04-09:
Thanks for letting me know @Xploder !

Printer's working now so can't test, the "[octoprint_compat]" was not part of my moonraker config file, I've added it and will test once the printer finishes in ~8 hours. At least without it it did not work, but I'm hoping it will once I restart it.

Will update later

wildjokerss, 2021-04-09:
@Xploder thank you once again! My somewhat long print finished and I just tried it and works fine!

closed 2021-04-09 by wildjokerss