supermerill/SuperSlicer #1075: Unexpected result on walls around holes

MLiv79, 2021-04-08:
### Version
2.3.55.5 (Latest Nightly)

### Operating system type + version
Windows 10 64 bit
### 3D printer brand / version + firmware version (if known)
Marlin FW 2.0.7.2 Thinker SE

### Behavior
As shown in the picture attached walls around holes are not printed as expected, finished print has a step like if some walls are missing or the print is undersized in these layers. 
Same file printed with prusa doesn't have this issue.

Which setting i need to check, if there is one that could cause this behaviour? Thanks

I'm using latest nightly build (5 days ago)

[image]

neophyl, 2021-04-08:
It would be best to save your project (file>save project as) and zip up the .3mf file and attach it here so we can see all the setting you are using. It’s the best way to debug issues. 

However if I had to hazard a guess the first thing I would check is if you have a solid infill below xx set (set it to 0 to turn off). Any infill with a square area less than the defined (which on some profiles seems to get set to 70) will be solid. That can cause issues with perimeters and thermal shrinking etc but usually not to that extent.

TommyC81, 2021-04-12:
Hi there, I think I might be experiencing the same issue. Printing the normal calibration cube, I get a few "thin" layers 2-3 layers above the X and Y letter, and also one "thin" layer about 2-3 layers above the first layer. Images for reference and .3mf file attached. "Solid infill threshold area" is set to 0.
[image]
[image]
[2021-04-12-Calibration_cube.zip](https://github.com/supermerill/SuperSlicer/files/6295138/2021-04-12-Calibration_cube.zip)

TommyC81, 2021-04-13:
Just a note to say I had the same results as above when printing with a different material and the "Solid infill threshold area" set to 50 mm2. I'm happy to test further if there's a suggestion on some other settings that could influence this behavior. There is no indication in the slicer itself that anything will be printed differently in the affected layers.

MLiv79, 2021-04-13:
I set that to zero but the behavior is the same

neophyl, 2021-04-13:
Wall unevenness where there is no indication in the slice preview is usually caused by thermal effects, from plastic contracting.  Different areas with different amounts of plastic will contract at different rates.  The other that ties into that is the speed of printing.  The different speeds will also exhibit the same problems for the same reasons.  So for example on the OP model those small areas could be using small perimeter speeds even if the solid infill value isnt in play.  Without a .3mf to check its impossible to say.

You also see problematic effect when walls have a mixture of perimeters and gapfill where the slicer is trying to accommodate for different wall thicknesses, although that usually causes a bulge outwards rather than inwards like pictured.

For the pictured calibration cube, I notice with the settings you have its putting 3 layers worth of solid infill at the perimeters that correspond to your inwards artefact.  Try turning off ensure vertical shell thickness to remove that and try reprinting.  Again thats a bunch of plastic that isnt really needed there that could be causing extra thermal issues.

MLiv79, 2021-04-13:
> Wall unevenness where there is no indication in the slice preview is usually caused by thermal effects, from plastic contracting. Different areas with different amounts of plastic will contract at [...]

could be this the cause?
[image]

supermerill, 2021-04-13:
> could be this the cause?

unlikely, this box just adjusts the fill flow by 2-5% .

TommyC81, 2021-04-13:
For whatever it is worth (given that I'm not sure whether my issues is related), I re-did my "Standard" settings starting from the built-in "0.20 NORMAL" settings. I recreated from the top of my memory the same settings, and it now prints perfectly. So, between the attached 2021-04-12-Calibration_cube.zip (zipped 3mf) and 2021-04-13-Calibration_cube.zip (zipped 3mf), the situation has been resolved. Not sure if there is an easy way to compare the settings to identify what would/could cause the rather big gap as seen in this photo (same as from previous post):
[image]
[2021-04-12-Calibration_cube.zip](https://github.com/supermerill/SuperSlicer/files/6305044/2021-04-12-Calibration_cube.zip)
[2021-04-13-Calibration_cube.zip](https://github.com/supermerill/SuperSlicer/files/6305045/2021-04-13-Calibration_cube.zip)

MLiv79, 2021-04-13:
> > could be this the cause?
> 
> unlikely, this box just adjusts the fill flow by 2-5% .
> 
> look at the layer time preview, see if the small layers are not the long ones.
> you can also look at the fan speed preview.

Yes, on those parts i've speed reduction because they are inside my "small perimeter speed" threshold

PS: thanks for the great slicer, latest nightly with first top layer after infill using bridge fan and speed setting is great.

TommyC81, 2021-04-14:
To update on my (possible hijacked) issue - I've now printed another 8 test cubes slowly going through settings. I believed I had figured out it had to do with "Layer time goal" for the filament eventually, but that turned out inconclusive. A re-calibration of the Linear Advance k-factor (from 0.8 to 0.5) seemed to yield good results. But I'm still not certain what caused the issue with "thin layers" that I experienced, as I got good results before the K-factor re-calibration with essentially the same setting.

In summary; despite extensive effort I'm not sure what caused my issue, it has largely been resolved by itself. If any hints can be derived from my previously provided 3mf files, I'd appreciate it! (Is there a tool available to compare settings? Cura has a "Profile analyzer" plugin that is useful, anything similar in SuperSlicer?)

supermerill, 2021-04-18:
> anything similar in SuperSlicer?

you can use a compare tool in your exported profile (like notepad++ compare)