supermerill/SuperSlicer #1077: I supporti provocano sovrapposizione layer e layer shifting

boromyr, 2021-04-11:
### Behavior
Hi, I recently had problems printing a case that required several supports. During the press I had serious layer overlapping problems, and after a dozen failed prints I managed to isolate the problem located in the support blocks. I can't share the file with my homes for business reasons, so I reproduced the problem in a common 3D print test.
As can be seen from images the use of supports causes different losses and overlaps layer. I report the previews of the GCDs, but the printed results are much worse than the previews.
I tried to modify different settings, including the one that should be the most correlated to the problem, the Z-offset for media.
[image]
[image]
#### Project File (.3MF) where problem occurs
Example file (not printed) + settings (used in printing)
The results of my case are worse than this example, involving 70% of the press about.
[3D_printer_test_mini_2.zip](https://github.com/supermerill/SuperSlicer/files/6292003/3D_printer_test_mini_2.zip)
### Version
2.3.55.5
### Operating system type + version
Windows 10 x64
### 3D printer 
Marlin 2.0.7.2
LK4 PRO

supermerill, 2021-04-11:
> As can be seen from images the use of supports causes different losses and overlaps layer. I report the previews of the GCDs, but the printed results are much worse than the previews.
[image]
I don't see where is the problem, or i can't reproduce.
Can you be more explicit?

boromyr, 2021-04-11:
Are your screens of SuperSlicer or Prusa-gcodeviewer.exe? Preview from SuperSlicer does not show problems. Try filling out the GCODE file and open it with pruse-gcodeviewer.exe, defective layers should be evident.