supermerill/SuperSlicer #1080: Flow Calibration 2% step don't print the O ring.

vdubedout, 2021-04-11:
### Version
SuperSlicer
Version:   2.2.53
Build:     SuperSlicer_2.2.53_2020-11-11

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19042), 64-bit edition
Total RAM size [MB]: 68,659MB
OpenGL installation
GL version:   4.6.0 NVIDIA 457.30
Vendor:       NVIDIA Corporation
Renderer:     GeForce GTX 1060 6GB/PCIe/SSE2
GLSL version: 4.60 NVIDIA

### 3D printer brand / version + firmware version (if known)
Ender5 (original)
Marlin 2.0.7.2

### Behavior
Doing calibrations for my printer. 
When generating the 2% flow calibration, the O ring don't appear in the slicer plater tab
I can see it in the sub-files with mX.amf o.amf

[image]
[image]

Expected: The O Ring is added

#### Project File (.3MF) where problem occurs

[filament_flow_test.zip](https://github.com/supermerill/SuperSlicer/files/6292635/filament_flow_test.zip)

lukaszzamel, 2021-04-14:
Works fine on 2.3.55
[image]

vdubedout, 2021-04-14:
Downloaded the 2.3.55, yes it's working fine there. We can consider it fixed/next release I guess. Thanks :)

closed 2021-04-14 by vdubedout