supermerill/SuperSlicer #1081: Mac version freezes on display config change (or sleep on MacBook)

sttts, 2021-04-12:
**Describe the bug**

When disconnecting a screen from a Mac(BookPro), SuperSlicer freezes. The spinning rainbow ball appear and does not go away. The window does not react anymore.

I believe this is the same issue as when the Mac goes to sleep (with attached screen). When waking up, the SuperSlicer process has disappeared.

**To Reproduce**
Steps to reproduce the behavior:
1. start SuperSlicer with external screen attached
2. disconnect external screen
3. enjoy rainbow ball

Alternative:
1. start SuperSlicer with external screen attached
2. wait for Mac to go to sleep
3. wake Mac up
4. SuperSlicer is gone.

**Desktop (please complete the following information):**
 - MacBookPro with macOS Catalina 10.15.7 and two screens attached
 - Version 2.3.55.5 nightly snapshot from Github actions (around Apr 5)

sttts, 2021-04-13:
Seems to happen only when in 3d view.

supermerill, 2021-04-13:
sadly, I don't have a max yet.

sttts, 2021-04-13:
It's also not that easy to reproduce. Now it doesn't happen while yesterday it happened multiple times.

neophyl, 2022-02-02:
There are several mac related issues on wakeup etc on the main PS github.  
like https://github.com/prusa3d/PrusaSlicer/issues/7645 
https://github.com/prusa3d/PrusaSlicer/issues/7818