supermerill/SuperSlicer #1082: build fail on Linux

Lukas1818, 2021-04-12:
**Describe the bug**
While updating the [superslicer-git AUR-package](https://aur.archlinux.org/packages/superslicer-git/) I notify that the build fail, because make try to install `Slic3r.desktop`.
```
CMake Error at cmake_install.cmake:66 (file):
  file INSTALL cannot find
  "/home/lukas/git/aur/superslicer-git/src/SuperSlicer/src/platform/unix/Slic3r.desktop":
  No such file or directory.

make: *** [Makefile:95: install] Fehler 1
```
In this folder does exist the desktop file but it extension is `.in` (`Slic3r.desktop.in`)

Furthermore the file does include some `@` which looks like variables, which was not replace during build.
```
[Desktop Entry]
Name=@GCODEVIEWER_APP_NAME@
Comment=G-code viewer of Slic3r
Exec=@SLIC3R_APP_CMD@ --gcodeviewer %F
Icon=@SLIC3R_APP_KEY@-gcodeviewer
Terminal=false
Type=Application
MimeType=text/x.gcode;
Categories=Graphics;3DGraphics;
Keywords=3D;Printing;Slicer;
```

**To Reproduce**
Steps to reproduce the behavior:
compile SuperSlicer on Linux

**Desktop (please complete the following information):**
 - OS: Manjro Linux (Based on Arch LINUX),
          but I pretty sure that will alos happen on other Linux distros
 - Version 39c95d7fedac6a8504351bc9218f3f54fb663c54

**Additional context**
I think the sames issue will also happen to `Gcodeviewer.desktop`.

Lukas1818, 2021-04-12:
same on 4abfbc1f50ee8584337b53407738199b85d36fc3

LoudSoftware, 2021-04-12:
ditto here, Manjaro, trying to build aur/superslicer and aur/superslicer-git and both failed identically.

Lukas1818, 2021-04-12:
Strange aur/superslice use version 2.2.54.2, which does compile for along time without issue.
I have done a downgrade a few hour ago, because I have accidentally upgrade to the latest pre-release. May a cleanbuild is a good idea now.

But I will check this again. Sadly building SuperSlicer take some time, so you must wait a bit.

LoudSoftware, 2021-04-12:
> Strange aur/superslice use version 2.2.54.2
Hmm, just caught that. and I am having different issues actually. Seems to be with boost library. I'll investigate further, maybe try the superslicer-prerelease package instead.

Lukas1818, 2021-04-12:
Boost again :D
I was able to build aur/superslicer with boost 1.75.0-2.
Which boost version did you use? Above 1.73.0?

LoudSoftware, 2021-04-12:
> Boost again :D
> I was able to build aur/superslicer with boost 1.75.0-2.
> Which boost version did you use? Above 1.73.0?

Using boost 1.75.0

Lukas1818, 2021-04-12:
> > Strange aur/superslice use version 2.2.54.2
> 
> Hmm, just caught that. and I am having different issues actually. Seems to be with boost library. I'll investigate further, maybe try the superslicer-prerelease package instead.
> 
> Edit: It looks like it's issues with boost mainly:
> `` more undefined references to `boost::detail::set_tss_data( `` etc over and over all related to boost library references

Könntest du bitte mal den ganzen Log hochladen? Habe das gefühl, dass das was hinter der `(`kommt wichtig sein könnte.

Lukas1818, 2021-04-12:
I have compile aur/superslicer-prerelease too, without any issue now.

LoudSoftware, 2021-04-12:
Hmm, That's strange.

Since I dont' really know where to find the build logs I started another fresh build of superslicer-git with the -L flag that should leave a log file in the build folder. I'll update when it's done failing to compile

Lukas1818, 2021-04-12:
> Hmm, That's strange.
> 
> Since I dont' really know where to find the build logs I started another fresh build of superslicer-git with the -L flag that should leave a log file in the build folder. I'll update when it's done failing to compile

I mean the output, which is showing in the terminal.
I do not think build superslicer-git make scene now, because if the [ci does fail](https://github.com/supermerill/SuperSlicer/runs/2327209232).

LoudSoftware, 2021-04-12:
https://pastebin.com/41d6yrQ6
This is my full log for superslicer-git from the aur

Lukas1818, 2021-04-12:
Did you have install boost-libs?

LoudSoftware, 2021-04-12:
yep, boost-libs-1.75.0-2 already installed

at this point I might just reboot back into windows

Lukas1818, 2021-04-12:
I have find another very similarly [issue](https://github.com/boostorg/log/issues/46). In this Issue the linker does link to a different version, as the user had build for. They offer some solution, may one help you. But because it was caused by bug in cmake, I think the bug is fixed now and the solutions may not work.

Are you sure you have done a clean build?

I do not have any addition ideas anymore.

Furthermore this issue get al little bit off topic, you may would like to create your own.

LoudSoftware, 2021-04-12:
Thanks a bunch @Lukas1818 for your insight. I do agree my issue is somewhat off topic for this one, I'll create a new one whenever I have time for it.

supermerill, 2021-04-13:
Sorry, i'm updating the cmake files.
I tough it would have worked, but I was wrong.

Lukas1818, 2021-04-13:
> Sorry, i'm updating the cmake files.
> I tough it would have worked, but I was wrong.

no problem. could happen

Lukas1818, 2021-05-01:
master (2e9562d5d7b6e05cad56d0d9a7203722520b9d3d) does compile again

thanks

closed 2021-05-01 by Lukas1818