supermerill/SuperSlicer #1085: Allow extrusion spacing to be specified instead of extrusion widths

NameOfTheDragon, 2021-04-15:
**Is your feature request related to a problem? Please describe.**

The problem is this: you want to be able to get solid walls of a known thickness without any gaps or infill. This is quite a common requirement in my experience, printing models with thin walls such as drawers, or in this case as a real-world example, parts for a Voron 2.4 3D printer, which has many parts designed to have 1.6 mm solid walls. The Voron build manual says to print "0.4 mm extrusions (forced)". The goal here is clearly to have exactly 4 extrusions with an overall width of exactly 1.6 mm with no gaps or infill. From the build manual...
[image]

But this does not work in Prusa/Super Slicer, or at least it is very difficult to achieve. There is no way to state that you want "4 permiters that come out to exactly 1.6 mm". Instead, you have to use these settings:
[image]

I put it to the jury that this is somewhat counter-intuitive. These numbers have to be computed using formulae that are buried in the user guide. It is quite hard to understand and get right.

What is needed, I think, is for the specified extrusion widths to take account of the spacing between extrusions, so that you can more easily comprehend the results. So for example if you specify 3 perimiters and 0.5mm extrusion widths, then you can confidently predict that you will obtain `3 x 0.5 = 1.5 mm` walls. This relationship is intuitively simple and easy to work out.

As an example of a part where this would be useful, consider the Voron 2.4 "skirt" panels which use a hexagon grid pattern with 1.6 mm solid walls. Setting all of the extrusion widths to the intuitively correct value of 0.4 mm produces this result (note the gaps in the walls and the infill in the corners):
[image]
[image]

Whereas, the correct result is far less easily obtained using the following settings (no gaps, far less infill in the corners):
[image]
[image]

How were those widths arrived at? No idea! I copied them from another user who copied them from another user who presumably did it by trial and error.

**Describe the solution you'd like**
I would like the specified extrusion widths to work intuitively such that the relationship `X perimiters with Y extrusion width = X * Y mm thick walls` holds true. Obviously it's more nuanced than that because the algorithm has to take account of overlap and external perimeters, but it should be intuitively obvious for the user.

Currently the extrusion widths specified do not take account of things like overlap, so that it is hard to come up with settings that result in an exact printed width.

**Describe alternatives you've considered**
The alternative as it exists today requires doing math and solving equations to come up with the correct extrusion widths if the goal is to get an exact number of extrusions in a given thickness of the model. Doing math is really what the software is good at and should be doing.

**Additional context**
The thinking behind this request came out of [this thread on Facebook](https://www.facebook.com/groups/voroncorexy/permalink/1160395217722376), in the `Voron CoreXY 3D Printer` group. 

You may not be a member so here are some screen shots...
[image]
[image]

julianschill, 2021-04-16:
If I think about it, the Voron manual is probably not precise enough. It is probably coming from the way it has to be set in Cura. I think the intention is to have an extrusion **spacing** of 0.4 with a reasonable extrusion **width** (like 0.46 or so) to have some overlap between the the extrusions. An extrusion width of 0.4 is not a thing you want to do with a 0.4 nozzle. So actually we need a way to set the **spacing** instead of it being derived from the width and the overlap.
This is all under the assumption, that the calculation is done in the way described [in the Slic3r documentation](https://manual.slic3r.org/advanced/flow-math)

EagleeyeKai, 2021-04-16:
I ran into this issue as well. I would really appreciate a solution like described above (TO)

supermerill, 2021-04-24:
I agree, it's needed.
Kind of the reverse of the perimeter tooltip, right?
something like that?
[image]

Note that it will works only on layers where you keep the current layer height, as the spacing depends on width & layer height.

also already discussed here:
https://github.com/supermerill/SuperSlicer/issues/921#issuecomment-804239326

julianschill, 2021-04-24:
This looks great to me. The question is how to handle the overlap settings. Will they be deactivated or overridden, when entering the spacing and the other way round? So if I enter 0 or -1 in spacing it takes the overlap?

supermerill, 2021-04-24:
If I set the value in a field, the companion one will turn inactive, no need of -1. The companion will then display his computed value.
I have to alter the gui to display an "inactive" icon or something on the "computed" field.

closed 2021-06-04 by supermerill

JD-Harding, 2022-09-07:
Hate to drag this up again, But did anyone get a good profile running? Would love to know what you went with as I'm about to start printing parts for my first Voron!

shortbus, 2022-09-07:
most start with the 2.4 settings that are in superslicer. adjust beed size
and then tune.

supermerill, 2022-09-07:
If you have a wall thickness you want to print without gapfill, you can use the "wall thickness" widget.
If you have multiple walls thickness, all a multiple of X. You can put X in the perimeter & external perimeter Spacing.