supermerill/SuperSlicer #1087: Superslicer ignoring origin

timj79, 2021-04-16:
### Do not report Prusa3D Slic3r bugs here without confirming it is a problem on a development release of Slic3r, or your issue will be closed. [Development builds of Slic3r](https://dl.slic3r.org/dev). [Prusa3D Issue Tracker](https://github.com/Prusa3D/Slic3r/issues)

**Describe the bug**
Superslicer is not respecting the origin set under 'bed shape' resulting in 'move out of range'

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'bed shape, set an origin of x15'
2. Place an object near the origin marked on the bed
3. slice and send to print
4. check gcode output

**3D Model and Slic3r Configuration Export**
Please upload a ZIP archive containing a copy of the 3D model you are seeing this bug with and a CTRL-E export of the configuration used.

**Expected behavior**
Expected Superslicer to an an offset of +15 to the gcode, this was working in previous versions

**Screenshots**

[image]
[image]
[image]
[image]
[X.stl.zip](https://github.com/supermerill/SuperSlicer/files/6325537/X.stl.zip)

**Desktop (please complete the following information):**
 Version: 2.3.56.0
Build: Slic3r_2.3.56.0
Operating System: Macintosh
System Architecture: 64 bit
System Version: macOS Big Sur Version 11.2.3 (Build 20D91)
Total RAM size [MB]: 8,590MB

**Additional context**
Add any other context about the problem here.

supermerill, 2021-04-24:
the offset is 15 mm in the plater, as shown in the 3D view.
I check and everything works as intended.
Maybe you wanted to have the origin 15 mm off the plater? In this case you have to enter `-15`

timj79, 2021-04-24:
Thanks for checking, maybe I’m just not understanding how the setting is meant to work?

In my firmware I have x-min set to 15 because the probe hits my frame at less than this. 

So I set my origin to x15 In superslicer and the 3d view shows where x15 is it is now effectively 0?  

so I expected  superslicer not to be send gcode under x15 when the mode is placed on the player further right than the origin?

supermerill, 2021-04-25:
The origin is where the "0" is.

closed 2021-04-25 by timj79