supermerill/SuperSlicer #1089: Problem with tabs on Mac OS

timj79, 2021-04-16:
**Describe the bug**
Clicking from sliced preview to Gcode preview causes a completely blank screen

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Click on 'configuration > preferences > gui > regular layer with the tab bar''
2. load any stl, click on sliced preview
3. click on gcode preview 
4. See error

**Expected behavior**
To properly display the gcode preview screen

**Screenshots**
[image]

**Desktop (please complete the following information):**
Version: 2.3.56.0
Build: Slic3r_2.3.56.0
Operating System: Macintosh
System Architecture: 64 bit
System Version: macOS Big Sur Version 11.2.3 (Build 20D91)
Total RAM size [MB]: 8,590MB

**Additional context**
Add any other context about the problem here.

liftbag, 2021-04-16:
I can confirm the issue.

liftbag, 2021-04-16:
@supermerill 
Here the log from macOS terminal running the last nightly with the option --loglevel 5.

**Sequence of operations:**

1. Open the app
2. Configuration > Preferences > GUI > Regular layout with the tab bar
3. Right click > Add shape > Box
4. Slice now
5. 3D view
6. Sliced preview
7. Gcode preview
8. Configuration > Preferences > GUI > Old PrusaSlicer layout
9. 3D editor view
10. close the app

[2.3.56.0_issue#1089_debug_output.zip](https://github.com/supermerill/SuperSlicer/files/6327702/2.3.56.0_issue.1089_debug_output.zip)

jallen810, 2021-04-20:
I will also confirm this issue. I can make it work by clicking back/forth between tabs & hide/unhide the right-side tool bar.
Running 2020 MBP 16" + AMD Radeon Pro 5300M 4Gb macOS Big Sur 11.2.3 and SS version 2.3.55.5

supermerill, 2021-04-24:
Understood.
I don't think i'll be able to resolve it with only logs.
My advice is to use an other gui layout on macos, for the time being.
I hope someone that can write c++ and has a mac can help to resolve this problem.