supermerill/SuperSlicer #1090: Extensive solid infill causing printer nozzle hits

TommyC81, 2021-04-16:
**Describe the bug**
Not sure if bug or intended, however, SuperSlicer creates an extensive amount of solid infill that (due to other printer settings or otherwise) sets the scene for extensive situations where the printer nozzle is likely to hit the infill (as experienced by myself). Not sure if setting missed somewhere, but I'm unable to reduce the amount of solid infill. The same does not occur in Cura (Arachne beta).

**To Reproduce**
Attached projects and gcode from both SuperSlicer and Cura (Arachne beta), check files to note infill. Additional images below.

**3D Model and Slic3r Configuration Export**
[SuperSlicer - Penholder.zip](https://github.com/supermerill/SuperSlicer/files/6328273/SuperSlicer.-.Penholder.zip)
[SuperSlicer - Penholder.gcode.zip](https://github.com/supermerill/SuperSlicer/files/6328277/SuperSlicer.-.Penholder.gcode.zip)

[Cura (Arachne) - Penholder.zip](https://github.com/supermerill/SuperSlicer/files/6328274/Cura.Arachne.-.Penholder.zip)
[Cura (Arachne) - Penholder.gcode.zip](https://github.com/supermerill/SuperSlicer/files/6328278/Cura.Arachne.-.Penholder.gcode.zip)

**Expected behavior**
Less solid infill and better use of thicker lines where appropriate (thus also reducing the the chance of over extrusion and other issues).

**Screenshots**
SuperSlicer:
[image]

Cura (Arachne beta):
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version: 20H2

**Additional context**
Add any other context about the problem here.

TommyC81, 2021-04-17:
Is this possibly related to #253 and https://github.com/slic3r/Slic3r/issues/5066?

jesterhead82, 2021-04-29:
Would it be possible to define a different infill type for solid infill which got generated by the slicer to ensure hor./vert. thickness/unfilled shells?

e.g. with concentric infill the issue is less intrusive (less wobble, less overextrusion of material).

Solid infill Rect:
[image]

Solid infill concentric:
[image]

neophyl, 2021-04-29:
I agree this is a problem but if all you do is replace it with concentric then you may as well just increase the perimeter count.  Which often removes the solid infill anyway.

supermerill, 2021-04-29:
> e.g. with concentric infill the issue is less intrusive (less wobble, less overextrusion of material).

you can also try 'rectilinear w gapfill'

I won't work on it right now, have to think of a way to "fix" it. Maybe add a setting for bottom layer from the build plate only?

OldCurmudgeon3DP, 2021-07-26:
I just noticed (while researching a FB post on this issue) that the slicer seems to still be enforcing some form of the Ensure Vertical Shell Thickness option even when unchecked. I created a cylinder and rotated it 60deg from vertical. The solid infill pattern appeared on "overhang" colored areas until I increased the perimeter count to 7. Below 6, the area circled in green begins to grow. I have the shell thickness option unchecked. When checked, gap fill areas are mixed in with the solid infill even at 7 perimeters.

[image]

[image]

[image]

Zeblote, 2021-07-29:
> Solid infill concentric:
> [image]

@jesterhead82 how did you get it to do that?

Zeblote, 2021-07-29:
> but if all you do is replace it with concentric then you may as well just increase the perimeter count. Which often removes the solid infill anyway.

@neophyl no, that's not it. The idea is to only to this for slopes, not the entire part. Currently that is only possible with slow wiggle infill.

neophyl, 2021-07-29:
I think you missed my point.  Yes I am familiar with the solid infill (what you call the wiggle infill) being added in an undesirable manner on sloping surfaces.  Its been raised as an issue for quite awhile on the Prusa slicer github as well as several times here.   I agree this behavior should really be changed.  I am not saying it doesn't need looking at.   The problem is that from reading both here and on the prusa git that it doesnt seem like a simple fix.  Its one of those ones that has been around for ages.

The point I was making is that if you change your slicer settings to replace it with concentric infill to try and get rid of it then from a practical standpoint theres not much difference with extra perimeters.  Seeing as in SOME case increasing perimeter count can reduce or remove the unwanted infill it seems like a first option to try when slicing to possibly get a better result.