supermerill/SuperSlicer #1092: Modifier parts create problems in the external perimeters

robustini, 2021-04-17:
When modifiers are added they create breaks in the outer perimeters.
More than a bug it is a boring thing.
It would be more useful instead that the external perimeters were respected and maintained their linearity, also for a better solidity of the printed item.
For example, if the declared external perimeters are 2 and in modifier 3 the modified section should have only two internal perimeters, and use an external one.

Slice the project and check the issue.

SuperSlicer V2.3.55.5 on Windows 10 Pro

[image]

[FUSOLIERA2.zip](https://github.com/supermerill/SuperSlicer/files/6329222/FUSOLIERA2.zip)

neophyl, 2021-04-17:
This is the way PS slices when you use modifiers.  When you apply a modifier you are effectively splitting the model into 2 models.  It slices them separately which causes these issues.  Its always done that.  
I agree that its problematic though and its been reported many time before at the main Prusa Slicer github repository.  I think the most recent one is https://github.com/prusa3d/PrusaSlicer/issues/5988  and from that post it looks like something is being modified in the current PS2.4 master.  
As changes in PS usually end up filtering down to SuSie this issue should eventually be resolved upstream.

robustini, 2021-04-17:
I hope so, because the modifiers are very comfortable but when placed on the external perimeters it is a disaster.

supermerill, 2021-04-25:
For the perimeters, the split is currently done the easy way.
It's possible to not split it for some options, but a significant effort is needed for that.

robustini, 2021-04-27:
I see, it is also a known problem in Prusa Slicer.
I'm noticing that Cura is smarter in this and leaves the outer perimeters unchanged.

robustini, 2021-04-27:
Here is the comparison.

[image]

[image]

robustini, 2021-12-23:
@supermerill thanks for all the current improvements and fixes, SuperSlicer has long since become my favorite slicer, I have deinstalled all the others.
Haven't planned a fix for this big problem?
Merry Christmas!

supermerill, 2021-12-23:
not on short term.

tonysmartantzoulatos, 2022-05-23:
@supermerill i can see the same problem as @robustini is saying on my parts also.
[image]
This is far from perfect print but the stl prior to slice with modifiers is solid and if sliced without modifiers it does not create this gap. Having the perimeters intact is key almost for all types of prints.