supermerill/SuperSlicer #1093: Compatibility with wxWidgets 3.1.5

gkatev, 2021-04-17:
Hi,
I'd like to give heads up on the new Wx 3.1.5 release:
https://github.com/wxWidgets/wxWidgets/releases

When compiling and running superslicer with Wx 3.1.5, the Plater and the 3D View, Sliced Preview, GCode Preview are "blank", and a bunch of these errors spew out:
```
[0x00007f6655c64080] [error]   Unable to init glew library
```
What I mean by "blank" (the image you see is my desktop's background):
[image]

Platform: Arch Linux
Version(s): 2.3.55.5 and git-master

Furthermore, on master this assertion fails as soon as superslicer w/ wx 3.1.5 opens (and not with 3.1.4):
```
ASSERT INFO:
/tmp/makepkg/superslicer-git/src/SuperSlicer/src/slic3r/GUI/ConfigWizard.cpp(2595): assert ""prusa_it != p->bundles.cend()"" failed in ConfigWizard(): Vendor PrusaResearch not found

BACKTRACE:
[1] wxEvtHandler::TryHereOnly(wxEvent&)
[2] wxEvtHandler::ProcessEventLocally(wxEvent&)
[3] wxEvtHandler::ProcessEvent(wxEvent&)
[4] wxEvtHandler::ProcessPendingEvents()
[5] wxAppConsoleBase::ProcessPendingEvents()
[6] wxApp::DoIdle()
[7] g_main_context_dispatch
[8] g_main_loop_run
[9] gtk_main
[10] wxGUIEventLoop::DoRun()
[11] wxEventLoopBase::Run()
[12] wxAppConsoleBase::MainLoop()
[13] wxEntry(int&, wchar_t**)
[14] __libc_start_main
```
I hope not to overwhelm you with bugs, just putting this out here, as I can only suppose it will come up sooner or later!

Noticed you are asking for verifications if bugs also exist on slic3r-dev (not even sure how slic3r interacts with perl-wx/wx, or if slic3r supports wx-3.1.x). If necessary/useful, I can try to build slic3r-dev with wx-3.1.5 and see what comes up.

Thank you, and let me know if I can be of further help.

sl1pkn07, 2021-04-22:
for me, wuth 2.2.54.2 and wxwidgets 3.1.5

~~~
[ 98%] Linking CXX executable superslicer
/usr/bin/ld: slic3r/liblibslic3r_gui.a(GUI_ObjectList.cpp.o):(.data.rel.ro._ZTVN6Slic3r3GUI10ObjectListE[_ZTVN6Slic3r3GUI10ObjectListE]+0x798): undefined reference to `wxDataViewCtrl::DoExpand(wxDataViewItem const&)'
/usr/bin/ld: slic3r/liblibslic3r_gui.a(wxExtensions.cpp.o):(.data.rel.ro._ZTV28wxDataViewTreeCtrlComboPopup[_ZTV28wxDataViewTreeCtrlComboPopup]+0x798): undefined reference to `wxDataViewCtrl::DoExpand(wxDataViewItem const&)'
/usr/bin/ld: slic3r/liblibslic3r_gui.a(SysInfoDialog.cpp.o): in function `Slic3r::GUI::get_main_info[abi:cxx11](bool)':
SysInfoDialog.cpp:(.text+0x1847): undefined reference to `wxPlatformInfo::GetArchName(wxArchitecture)'
collect2: error: ld returned 1 exit status
make[2]: *** [src/CMakeFiles/slic3r.dir/build.make:139: src/superslicer] Error 1
make[1]: *** [CMakeFiles/Makefile2:503: src/CMakeFiles/slic3r.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
~~~

greetings

supermerill, 2021-04-25:
As it can't init the glew (opengl) , then it can't draw the 3D view, hence the transparent panel.
I don't know why it isn't working, maybe wxwidget 3.1.5 need a newer version of glew or something else?

> Furthermore, on master this assertion fails as soon as superslicer w/ wx 3.1.5 opens (and not with 3.1.4):

Prusa put its settings in front, and has some code for prusa -specific wizard. I deactivated most of it, but sometimes i forgot to disable new portions. Like here. Nothing to do with your issue.

> I hope not to overwhelm you with bugs

no problem.

> Noticed you are asking for verifications if bugs also exist on slic3r-dev (not even sure how slic3r interacts with perl-wx/wx, or if slic3r supports wx-3.1.x). If necessary/useful, I can try to build slic3r-dev with wx-3.1.5 and see what comes up.

nah, it's just that i forgot to change the github file, as i merge things with the slic3r repo, and i forgot to re-personalize this for my repo. Don't bother

fiechr, 2021-04-26:
Just FYI, I'm having the same issue with Pursaslicer: 
https://github.com/prusa3d/PrusaSlicer/issues/6396

wavexx, 2022-02-19:
FIY, looks like by default wxWidgets 3.1.5 enables EGL support, which also needs to be enabled in GLEW. GLEW 2.2.0 EGL support seems partial, so it's not enabled by default so far (at least on debian, but it seems that some distros have this non-matching settings too).

Since I wanted to have a few fixes from wx 3.1.5, the easiest way is to build wxwidgets with ``--disable-glcanvasegl`` at the moment. This works with the default GLEW 2.2.0 and/or with the bundled GLEW.

jwise, 2023-01-11:
GLEW 2.2.0 works for me in Bambu Studio: https://github.com/bambulab/BambuStudio/pull/1053

richfelker, 2025-03-12:
I hit this issue with self-built superslicer using system libs (no deps dir) on Apline Linux. Building an alternate `libGLEW.so` with `SYSTEM=linux-egl` and putting it in `LD_LIBRARY_PATH` made the superslicer binary (which previously had no display) work unmodified.

I'm not sure what the right fix is on systems where some things expect to be using GLX and other things (wxwidgets) expect to be using EGL, since GLEW seems to only support one or the other...