supermerill/SuperSlicer #1098: When the specified number of perimeters will not fit in an area of a model, only a single perimeter is printed

NameOfTheDragon, 2021-04-19:
### Version
2.3.55.5

### Operating system type + version
Windows 10

### Behavior
Consider the attached project
[Issue Repro.zip](https://github.com/supermerill/SuperSlicer/files/6338537/Issue.Repro.zip)

Examine the teeth in the small gear. When set to 4 perimeters, the teeth are solid (desired result):
[image]

However, if the number of perimeter's is increased to 5, the teeth print with just a single perimeter and sparse infill:
[image]

If the model feature is too small to accept the specified number of perimeters, I would expect it to print as solid (as many perimeters as will fit). Otherwise this can lead to unexpectedly weak small features.

NameOfTheDragon, 2021-04-19:
May be related to #991

supermerill, 2021-04-25:
remove the "only one perimeter on top" or the min eara for it

closed 2021-06-04 by supermerill