supermerill/SuperSlicer #1101: Make the setting difficulty indicators more colorblind accessible

PlantLily, 2021-04-21:
**Is your feature request related to a problem? Please describe.**
the indicators next to a setting showing whether they're simple, advanced, or expert aren't very colorblind friendly.   

**Describe the solution you'd like**
by simply adding a letter inside the hexagon(S for simple,  A for advanced, and E for expert), or having different amounts of dots on them would make the slicer more colorblind accessible.

**Additional context**
for context, these are the indicators im talking about:
[image]

supermerill, 2021-04-25:
kinda related to #356
I'm going to replace these by "labels" instead of "groups", so I wan't do extensive modification before that.

But in the meantime, I can put the colors used in the ui_layout/colors.ini file to be able to edit them.

supermerill, 2022-03-23:
you can now change their color in the ui_layout colors.ini (the 'tag' at the end)
For the letter, I'll try to add a colorblind mode to switch them on.