supermerill/SuperSlicer #1102: Top infill on unexpected places when using MMU

Prutsium, 2021-04-22:
Windows version Version 2.3.55
Running Prusa with MMU (in this case 2 colors)

When i slice the attached file suddenly unexpected top infill appears.

Model:
[image]

After Slicing:
[image]

First few layers are as expected:
[image]

Then 1 took the multicolor part out from 1 part and sliced again the following can be seen (the one on the lower right)
[image]

[Merged.zip](https://github.com/supermerill/SuperSlicer/files/6358084/Merged.zip)
Attached the project file

supermerill, 2021-04-25:
probably a problem with the stl file.

I can't replicate on my nightly. You use the version form what date?
[image]

Prutsium, 2021-04-25:
In Ps this problem doesnt happen.