supermerill/SuperSlicer #1106: Missing z_offset causes a slicing crash for SL printers

Pavulon87, 2021-04-25:
**Describe the bug**
update_layers_slider causes error and app crash.
`const float z_shift = wxGetApp().preset_bundle->printers.get_edited_preset().config.opt_float("z_offset");`

> Exception thrown: read access violation.
> **Slic3r::DynamicConfig::opt_float**(...) returned 0x8.

**To Reproduce**
Slice any model with SL printer profile.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - SuperSlicer Version 2.3.56

**Additional context**
Adding z_offset = 0 to the profile allows you to slice the model.

supermerill, 2021-04-25:
Can you send me a project where it's crashing?
I'm ok with your patch, but I'm wary of adding even more "if sla", as if I add another printer type, it can be a mess.
The simpler version should be to test for the existence of z_offset  instead of the printer type.

Pavulon87, 2021-04-25:
These are the basic settings and box shape.
[Shape-Box.3mf.zip](https://github.com/supermerill/SuperSlicer/files/6372709/Shape-Box.3mf.zip)
It doesn't matter what file and settings.

PR #1107 (2021-04-25): Fix requesting z_offset from sla printer
This should fix #1106
closed 2021-04-26 by Pavulon87