supermerill/SuperSlicer #1108: [dev/master] recent renaming breakages

nerdCopter, 2021-04-25:
**Describe the bug**
* Linux version: Referencing master [d4f5d64] and several commits prior, the config now uses `~/.config/SuperSlicer` rather than the prior `~/.config/SuperSlicer-alpha` (which is fine, but was confusing to solve).
* Upon moving all my settings into the new folder, I've also found that the bed-plate graphics no longer shows.
* Using the Configuration Wizard to reinstall my printer, does not resolve the problem.
* Graphical representation of the available printers do not show in the wizard.
* probably directly related to software and file-system renaming commits of late.

I'm surprised changes are being merged directly to master rather than utilizing the Pull-Request system for checks and balances. This would also help with documentation and "atomic" updates/merges.

supermerill, 2021-04-25:
> I'm surprised changes are being merged directly to master rather than utilizing the Pull-Request system for checks and balances.

I'm the one that creates the commits, check them and decide it it's ok so...

> Graphical representation of the available printers do not show in the wizard.

I've updated the doc, now you need to git submodule update the profiles, as they were moved in their own repo.
I think it's your problem.

nerdCopter, 2021-04-25:
TYVM, i use the already established GitHub-Actions build system, fwiw.