supermerill/SuperSlicer #1109: Benchy slicing problems. Bridging/ Overhangs/ infill

PinGuiNPL, 2021-04-26:
**Describe the bug**

The problems appear in many projects, but everyone knows the Benchy and it tends to several of these problems.

Let's start with the 2nd layer.

[image]
[image]
[image]

PrusaSlicer takes a lot more distance to bridge the spots. Cura takes the whole layer right away, which works best IMO.
SuperSlicer only takes a bit more than the letters themselves, that takes the most time and therefore also leads the most heat into the object.

Is there a way to set the bridging so that it looks like it does in Cura?

Let's go to the 3rd layer. What is SuperSlicer doing here? While PrusaSlicer and Cura are printing the entire layer, does SuperSlicer continue to outline the letters.

[image]
[image]
[image]

As with the second layer, the problem is more time, more heat and a lot more movement.
The problem also exists with the 4th layer.

The higher the more problems. Only Cura can do it properly here (as an example layer 13). Prusa and SuperSlicer put a lot of material on the sides to support the overhangs. But everyone who has already printed a Benchy knows that it is completely unnecessary in these places.

[image]
[image]
[image]

Is there perhaps a solution for this already?

With layer 123 you only get a reasonable bridge if you use the setting "Remove Parimeters" in the advanced settings. Otherwise, the SuperSlicer tries to make circles in the air.

[image]
[image]
[image]
[image]

Since you can do it with one Setting, I don't see that as a problem.
Nevertheless, I see several problems with the whole algorithm that not only cost time unnecessarily but also use up material senselessly in some places.
Maybe I am overlooking some setting, hence the question of whether you can somehow work around the problems and if not then that would be something for a future update.

**To Reproduce**
Download Benchy
Slice it in different Slicers
Compare Layers: 2, 3, 4, 13, 123 at 0,3 Layer hight

**3D Model and Slic3r Configuration Export**
[Config_and_Benchy.zip](https://github.com/supermerill/SuperSlicer/files/6379781/Config_and_Benchy.zip)

**Desktop (please complete the following information):**
 - OS: Windows 10
 - SuperSlicer 2.3.56.0

supermerill, 2021-04-26:
> Is there a way to set the bridging so that it looks like it does in Cura?

you can increase the bridging anchor area with infill -> anchor solid infill by X mm -> bridged
I don't think you can remove the bridging. You can reduce the flow, so it behaves a bit more like a not-bridging.

> Let's go to the 3rd layer. What is SuperSlicer doing here? While PrusaSlicer and Cura are printing the entire layer, does SuperSlicer continue to outline the letters.

It's to apply the over-bridge ratio. If you set it to 100%, your difference will disappear.

> The higher the more problems. Only Cura can do it properly here (as an example layer 13). Prusa and SuperSlicer put a lot of material on the sides to support the overhangs. But everyone who has already printed a Benchy knows that it is completely unnecessary in these places.

you may have to reduce the number of bottom layers.

> With layer 123 you only get a reasonable bridge if you use the setting "Remove Parimeters" in the advanced settings. Otherwise, the SuperSlicer tries to make circles in the air.

that one is weird, have to look at it.

PinGuiNPL, 2021-04-27:
Thanks @supermerill for the very quick reply. <3

> you can increase the bridging anchor area with infill -> anchor solid infill by X mm -> bridged
I don't think you can remove the bridging. You can reduce the flow, so it behaves a bit more like a not-bridging.

The setting "anchor solid infill by X mm" works perfectly!
I set it to 10mm and layers 2 and 3 are perfect!

[image]
[image]

Unfortunately this only works if you set "No parimeters on bridge areas" to "Disabled".

But if you set this up then there is the problem with layer 123 and the bridge in the cabin. Circles in mid-air.

[image]

However, if you set "No parimeters on bridge areas" to "Remove Parimeters" so that the bridge is correctly positioned in the cabin, then again layers 2 and 3 will not be printed correctly.

[image]

So you have to decide which part of the benchy you want to print perfectly and give up the other.

> It's to apply the over-bridge ratio. If you set it to 100%, your difference will disappear.

I can't find the setting "over- bridge ratio"

With the overhang, it's a bit more complicated in the case of Benchy.
There is a challenge under the name SpeedBenchy, there are also fixed rules. 2 of them are: Top and Bottom Layers: 3 and minimum parimeters: 2.

> you may have to reduce the number of bottom layers.

Yes, you can get rid of the superfluous material on the overhangs by setting 3 parimeters or 1 bottom layer.

Unfortunately, neither is allowed.

It is not really a problem of the slicer, since it offers the possibility to solve the problem and any challenge is not a problem of the slicer.
It's just a shame that I only have to change the slicer because of some challanges because someone else does it a little better.
I have an extremely modified Voron CoreXY printer that has no problems spitting out a sub 15 minute Benchy.
The goal is now sub 10 minutes.
I have invested a lot of time in setting the profiles in the slicer perfect and when I'm are ready for a pair of challanges i have to change the slicer.
I don't want to do that because I like the SuperSlicer a lot ... PrusaSlicer doesn't have many settings that I need and it doesn't make a lot of things better anyway. Cura, on the other hand, has other problems and I would have to create completely new profiles.

As I said, it's not just about Benchy, but it shows some problems with the algorithm very well. If you set it so that the lower bridges are perfect then the bridge in the cabin will be wrong. Or the other way around. I've had similar problems before on other projects.
I think the SuperSlicer is the right one for custom printers, even more so when it comes to fast, competetive printers. It's not always about beautiful prints. Unfortunately, the problems start exactly where every minute is concerned.

Lots of text, for a short question: How do you set up the slicer so that layers 2 and 3 (in this case) are printed correctly but the overhang is still properly anchored in the cabin?

neophyl, 2021-04-27:
Your problem with the smokestack on layer 123 is caused by you having 'only 1 perimeter on top surfaces' enabled.  That's known to cause some issues with the current versions.  Turn it off and it bridges ok.

PinGuiNPL, 2021-04-27:
I've created a much simpler object that has the same problems. So that you don't get the impression that the problem only exists with the Benchys.

*Settings and Files to be able to reproduce it:*
[Config_and_TestObject.zip](https://github.com/supermerill/SuperSlicer/files/6384626/Config_and_TestObject.zip)
Layer hight: 0.3
Layer width: 0.4

The problem is mainly the setting "No parimeters on bridge areas"
If you select the "Disabled" option here, the slicer tries again to make circles in the air as a bridge (layer 7). The layers after that are properly sliced.

[image]
[image]

If you select the "Remove Parimeters" option, the bridge is created correctly, but the layers above are then sliced ​​again in several parts instead of as a whole layer.

[image]
[image]
[image]

In short, it would be best if there was a mix of both settings.

PinGuiNPL, 2021-04-27:
> Your problem with the smokestack on layer 123 is caused by you having 'only 1 perimeter on top surfaces' enabled. That's known to cause some issues with the current versions. Turn it off and it bridges ok.

Oh wow, this is a very simple fix now that seems to work! Thank you very much!

neophyl, 2021-04-27:
Btw instead of zipping up your file and your config just zip up and attach your saved .3mf project file (File>Save Project as).
The 3mf will contain your model(s) and ALL your settings, the model positions, any modifiers in use etc etc.  Its basically a snapshot of your environment.  Its a much easier way of doing it.

supermerill, 2021-04-27:
> So you have to decide which part of the benchy you want to print perfectly and give up the other.

you can set a setting only on some layers with a layer range modifier.

PinGuiNPL, 2021-04-27:
> you can set a setting only on some layers with a layer range modifier.

Thanks everyone for help! If I find more problems, I'll open a new topic :)

closed 2021-04-27 by PinGuiNPL