supermerill/SuperSlicer #1110: SuperSlicer crashes on launch.

lbibass, 2021-04-26:
### Do not report Prusa3D Slic3r bugs here without confirming it is a problem on a development release of Slic3r, or your issue will be closed. [Development builds of Slic3r](https://dl.slic3r.org/dev). [Prusa3D Issue Tracker](https://github.com/Prusa3D/Slic3r/issues)

**Describe the bug**
SuperSlicer Crashes on launch. 
**To Reproduce**
Steps to reproduce the behavior:
Attempt to open SuperSlicer.

**3D Model and Slic3r Configuration Export**
N/A
**Expected behavior**
App Launches. 
**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: MacOS
 - Version: MacOS Nightly #230. 

**Additional context**
Add any other context about the problem here.
Here's the debug log reported by macos. This crash also occurs when running a build compiled on my native machine. This was not a problem as of commit d4f5d643f3c2b62b724651818b4037f0fd8dd164. 

`Process:               SuperSlicer [34192]
Path:                  /private/var/folders/*/SuperSlicer.app/Contents/MacOS/SuperSlicer
Identifier:            com.merill.slic3r/
Version:               SuperSlicer Slic3r_2.3.56.0 (Slic3r_2.3.56.0)
Code Type:             X86-64 (Translated)
Parent Process:        ??? [1]
Responsible:           SuperSlicer [34192]
User ID:               501

Date/Time:             2021-04-26 17:38:00.388 -0400
OS Version:            macOS 11.2.3 (20D91)
Report Version:        12
Anonymous UUID:        A7484206-1361-AE4B-248C-74F12DFEDDC0

Sleep/Wake UUID:       DF1B5D9F-5453-47E1-A1D5-70D70F84CF2A

Time Awake Since Boot: 70000 seconds
Time Since Wake:       8800 seconds

System Integrity Protection: enabled

Notes:                 Translocated Process

Crashed Thread:        0  slic3r_main  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0xc49dffff88371bf9 -> 0xffffffff88371bf9 (possible pointer authentication failure)
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [34192]

VM Regions Near 0xffffffff88371bf9:
--> mapped file              7fffe0000000-7fffe22cc000 [ 34.8M] r--/r-- SM=COW  Object_id=f3266aa1
    

Thread 0 Crashed:: slic3r_main  Dispatch queue: com.apple.main-thread
0   com.merill.slic3r/            	0x0000000102cea604 Slic3r::GUI::Tab::load_current_preset() + 964
1   com.merill.slic3r/            	0x0000000102bf0cc8 Slic3r::GUI::GUI_App::load_current_presets(bool) + 248
2   com.merill.slic3r/            	0x0000000102bec4e8 Slic3r::GUI::GUI_App::on_init_inner() + 3912
3   com.merill.slic3r/            	0x0000000102beb55b Slic3r::GUI::GUI_App::OnInit() + 11
4   com.merill.slic3r/            	0x0000000103a3ef51 wxApp::CallOnInit() + 417
5   com.merill.slic3r/            	0x0000000103bd1976 wxEntry(int&, wchar_t**) + 198
6   com.merill.slic3r/            	0x0000000103bd1bab wxEntry(int&, char**) + 59
7   com.merill.slic3r/            	0x0000000102bdc69c Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) + 236
8   com.merill.slic3r/            	0x00000001023359ab Slic3r::CLI::run(int, char**) + 17435
[... 422 lines cut ...]
Rosetta Thread Context             160K       16 
STACK GUARD                         24K        6 
Stack                             11.6M        8 
Stack Guard                       56.0M        2 
VM_ALLOCATE                       3480K       46 
VM_ALLOCATE (reserved)               4K        1         reserved VM address space (unallocated)
__CTF                               759        1 
__DATA                            14.8M      328 
__DATA_CONST                      16.1M      188 
__DATA_DIRTY                       664K      111 
__FONT_DATA                          4K        1 
__GLSLBUILTINS                    5176K        1 
__LINKEDIT                       527.6M       13 
__OBJC_RO                         60.6M        1 
__OBJC_RW                         2449K        2 
__TEXT                           321.4M      333 
__UNICODE                          588K        1 
mapped file                        4.1G      413 
shared memory                       64K        4 
unshared pmap                     8544K        5 
===========                     =======  ======= 
TOTAL                              6.2G     1952 
TOTAL, minus reserved VM space     5.5G     1952 

`

supermerill, 2021-04-26:
can you try to recompile and re run, but with modifying
src\slic3r\GUI\GUI_Preview.cpp@744
`if ( wxGetApp().preset_bundle->printers.get_edited_preset().config.option("z_offset") != nullptr )`
to
`if ( wxGetApp().preset_bundle && wxGetApp().preset_bundle->printers.get_edited_preset().config.option("z_offset"))`
?

supermerill, 2021-04-27:
Or can you test the latest nightly, it has the change.

lbibass, 2021-04-27:
> Or can you test the latest nightly, it has the change.

There is no difference. There is still an immediate crash as soon as the splash screen disappears.

supermerill, 2021-04-27:
can you try to build at different commit, to see what is the wrong commit?
first trying with dfc9671dc604a8b7154e288053929cd47cae3516
If it works and the latest nightly crash, then try  3d1cd5c7602f2e30b0dd5e9b29f99165a9524b9d
or it doesn't works, can you try with 715d58da888e591116a82f70ba6ff30020275ced
(note that these two last commits will change the name of the app, so they will trigger / need a new cmake)

akospalfi, 2021-04-29:
Just a quick info: d4f5d64 works for me correctly, but dfc9671 doesn't work, it fails with a segfault on startup. However, I observed, that sometimes it doesn't crash and the app loads fine. I would say 3 of 5 startup crashes.
I'll try to bisect the problem to a specific commit...

supermerill, 2021-04-29:
@akospalfi thanks
there is only 

daa2c66546ac83eff67fc1f24f15158e7b2dc775
0fe1664f19f65a2d6ebd9fa402f54c06792e2dde
b6d278a36f9917482a85e5786445cae49859cf95
715d58da888e591116a82f70ba6ff30020275ced

between them.
daa2c66546ac83eff67fc1f24f15158e7b2dc775 fix the macosx icon (some little changes in the Info.plist inside the dmg. Very low probability
0fe1664f19f65a2d6ebd9fa402f54c06792e2dde is little fix for gcode export. Also 0 chance taht's the problem
b6d278a36f9917482a85e5786445cae49859cf95 add a menu entry. Low probability
715d58da888e591116a82f70ba6ff30020275ced add spacing settings. Very high probability that's the problem. But why only on macos?

As it seems that's random ( you can make it launch), I'm going to do a pass on it to check for thread-unsafe things.
I'm going to output a new nightly in some hours. If it succeeds to compile, it contains some threadsafing code for settings, so maybe it contains the fix. If you can try.

akospalfi, 2021-04-29:
Yes, I've cherry picked only 715d58d on top of d4f5d64 and the app started to crash randomly on startups.
I'll try to make a debug build tomorrow and send you a proper backtrace.

akospalfi, 2021-04-29:
> As it seems that's random ( you can make it launch), I'm going to do a pass on it to check for thread-unsafe things.
> I'm going to output a new nightly in some hours. If it succeeds to compile, it contains some threadsafing code for settings, so maybe it contains the fix. If you can try.

Sure, I'll try it happily.

supermerill, 2021-04-29:
I protected the only line where a problem can occur (in 715d58d)
I'll launch a build on that (it's at the current CI branch)

akospalfi, 2021-04-29:
Seems the CI build is failed, but I think I've got some progress..
I've managed to build a debug build and this build always crashes on startup (strange...). Anyways I was able to capture a backtrace, I hope that helps somehow:

```
(lldb) target create ./superslicer 
Current executable set to '/Volumes/WORK/myprojects/SuperSlicer/SuperSlicer/build/src/superslicer' (x86_64).
(lldb) process launch -- --loglevel 5 --datadir "/Users/palfia/Library/Application Support/SuperSlicer-2"
Process 97947 launched: '/Volumes/WORK/myprojects/SuperSlicer/SuperSlicer/build/src/superslicer' (x86_64)
[2021-04-29 13:44:55.497835] [0x00000001112abe00] [debug]   full path: 16849348648804176793
[2021-04-29 13:44:55.498102] [0x00000001112abe00] [debug]   single instance: undefined. other params: /Volumes/WORK/myprojects/SuperSlicer/SuperSlicer/build/src/superslicer;--loglevel;5;--datadir;"/Users/palfia/Library/Application Support/SuperSlicer-2"
[2021-04-29 13:44:55.498114] [0x00000001112abe00] [debug]   full lock path: /Users/palfia/Library/Application Support/SuperSlicer-2/cache/16849348648804176793.lock
[2021-04-29 13:44:55.498166] [0x00000001112abe00] [info]    instance check: Another instance not found or single-instance not set.
2021-04-29 13:44:55.503353+0200 superslicer[97947:3194938] SecTaskLoadEntitlements failed error=22 cs_flags=20, pid=97947
2021-04-29 13:44:55.503421+0200 superslicer[97947:3194938] SecTaskCopyDebugDescription: superslicer[97947]/0#-1 LF=0
[2021-04-29 13:44:55.877823] [0x00000001112abe00] [info]    Checking if indices need to be installed from resources...
13:44:55: Debug: Adding duplicate image handler for 'JPEG file'
[2021-04-29 13:44:55.888798] [0x00000001112abe00] [trace]   Best translation language detected (may be different from user locales): en_GB
[2021-04-29 13:44:55.888835] [0x00000001112abe00] [trace]   Switching wxLocales to en_GB
/Volumes/WORK/myprojects/SuperSlicer/SuperSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/osx/core/bitmap.cpp(1376): assert "GetWidth() == w" failed in SetWidth(): Changing the bitmap width is not supported
Collecting stack trace information, please wait...2021-04-29 13:45:08.964517+0200 superslicer[97947:3194938] SecTaskLoadEntitlements failed error=22 cs_flags=20, pid=97947
2021-04-29 13:45:08.964601+0200 superslicer[97947:3194938] SecTaskCopyDebugDescription: superslicer[97947]/0#-1 LF=0
/Volumes/WORK/myprojects/SuperSlicer/SuperSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/osx/core/bitmap.cpp(1376): assert "GetWidth() == w" failed in SetWidth(): Changing the bitmap width is not supported
Collecting stack trace information, please wait...[2021-04-29 13:45:31.824694] [0x00000001112abe00] [info]    3dx mac handler starts
[2021-04-29 13:45:31.824935] [0x00000001112abe00] [error]   3dx drivers module loading error: dlopen(/Library/Frameworks/3DconnexionClient.framework/3DconnexionClient, 5): image not found
[2021-04-29 13:45:31.824947] [0x00000001112abe00] [info]    3dx drivers loaded: NO
[2021-04-29 13:45:31.826865] [0x00000001112abe00] [info]    Undo / Redo snapshot taken: New Project, Undo / Redo stack memory: 0MB Resident memory: 146MB; Peak memory usage: 151MB
[2021-04-29 13:45:31.975787] [0x00000001112abe00] [debug]   GLCanvas3D: Use Retina OpenGL: true
[2021-04-29 13:45:31.975812] [0x00000001112abe00] [debug]   GLCanvas3D: Use Retina OpenGL: true
Assertion failed: (false), function select_technology, file /Volumes/WORK/myprojects/SuperSlicer/SuperSlicer/src/slic3r/GUI/BackgroundSlicingProcess.cpp, line 117.
Process 97947 stopped
* thread #1, name = 'slic3r_main', queue = 'com.apple.main-thread', stop reason = hit program assert
    frame #4: 0x000000010141ff36 superslicer`Slic3r::BackgroundSlicingProcess::select_technology(this=0x00000001138a1c90, tech=0) at BackgroundSlicingProcess.cpp:117:18
   114 			case ptFFF: m_print = m_fff_print; break;
   115 			case ptSLA: m_print = m_sla_print; break;
   116 	        case ptSLS: m_print = m_fff_print; break;
-> 117 	        default: assert(false); break;
   118 			}
   119 			changed = true;
   120 		}
Target 0: (superslicer) stopped.
(lldb) bt
* thread #1, name = 'slic3r_main', queue = 'com.apple.main-thread', stop reason = hit program assert
    frame #0: 0x00007fff20389462 libsystem_kernel.dylib`__pthread_kill + 10
    frame #1: 0x00007fff203b7610 libsystem_pthread.dylib`pthread_kill + 263
    frame #2: 0x00007fff2030a720 libsystem_c.dylib`abort + 120
    frame #3: 0x00007fff203099d6 libsystem_c.dylib`__assert_rtn + 314
  * frame #4: 0x000000010141ff36 superslicer`Slic3r::BackgroundSlicingProcess::select_technology(this=0x00000001138a1c90, tech=0) at BackgroundSlicingProcess.cpp:117:18
    frame #5: 0x0000000101a353ad superslicer`Slic3r::GUI::Plater::set_printer_technology(this=0x00000001130ae770, printer_technology=0) at Plater.cpp:6219:38
    frame #6: 0x00000001018afe69 superslicer`Slic3r::GUI::GUI_App::load_current_presets(this=0x000000011202aa00, check_printer_presets_=true) at GUI_App.cpp:1971:18
    frame #7: 0x00000001018ac270 superslicer`Slic3r::GUI::GUI_App::on_init_inner(this=0x000000011202aa00) at GUI_App.cpp:928:9
    frame #8: 0x00000001018aaaa5 superslicer`Slic3r::GUI::GUI_App::OnInit(this=0x000000011202aa00) at GUI_App.cpp:765:16
    frame #9: 0x000000010317ffc1 superslicer`wxApp::CallOnInit(this=0x000000011202aa00) at utils.mm:439:22
    frame #10: 0x00000001033391a6 superslicer`wxEntry(argc=0x0000000103ba66b0, argv=0x0000000113004590) at init.cpp:491:25
[... cut ...]
    frame #13: 0x000000010000d403 superslicer`Slic3r::CLI::run(this=0x00007ffeefbff7c0, argc=5, argv=0x00007ffeefbff8e8) at PrusaSlicer.cpp:572:16
    frame #14: 0x0000000100012059 superslicer`main(argc=5, argv=0x00007ffeefbff8e8) at PrusaSlicer.cpp:773:18
    frame #15: 0x00007fff203d2621 libdyld.dylib`start + 1
    frame #16: 0x00007fff203d2621 libdyld.dylib`start + 1
(lldb) 
```

So it seems `printer_technology` becomes null  `at Plater.cpp:6219:38` after 715d58d (with d4f5d64 it has a non-null value).

akospalfi, 2021-04-29:
I just picked your fix (102c882d) to my branch, but the appstart fails at the same point in 
```
superslicer`Slic3r::BackgroundSlicingProcess::select_technology(this=0x000000011e060a90, tech=0) at BackgroundSlicingProcess.cpp:117:18
```

mvieleers, 2021-04-29:
Just to add to this issue, I have tried all the pre-release 2.3 versions from the releases page. They all crash immediately after the splash screen. MacOS doesn't report a crash as with some other software. The app simply closes immediately. So it doesn't provide me with a possibility to get a crash report.
The stable 2.2 version from the releases page does work correctly for me.

supermerill, 2021-04-29:
@akospalfi 
it's an assert error, probably not related to the issue.
can someone on mac test the [action build 235](https://github.com/supermerill/SuperSlicer/suites/2613051641/artifacts/57249381) ?
It should output some text on the console, can you sopyt it here?

mvieleers, 2021-04-29:
@supermerill I've tried this action build 235. This one sometimes crashes and sometimes doesn't.

No crash console output:
`fout	18:18:30.423181+0200	mDNSResponder	GetLargeResourceRecord: SetRData failed for <mask.hash: 'TY9CWFjq2hDsvJMnQ/DODQ=='> (NSEC)
fout	18:18:30.548921+0200	mDNSResponder	GetLargeResourceRecord: SetRData failed for <mask.hash: 'TY9CWFjq2hDsvJMnQ/DODQ=='> (NSEC)
fout	18:18:30.738919+0200	mDNSResponder	GetLargeResourceRecord: SetRData failed for <mask.hash: 'TY9CWFjq2hDsvJMnQ/DODQ=='> (NSEC)
fout	18:18:30.739039+0200	mDNSResponder	GetLargeResourceRecord: SetRData failed for <mask.hash: 'TY9CWFjq2hDsvJMnQ/DODQ=='> (NSEC)
fout	18:18:31.473690+0200	lsd	cannot open file at line 44595 of [02c344acea]
fout	18:18:31.473706+0200	lsd	os_unix.c:44595: (2) open(/var/db/DetachedSignatures) - No such file or directory
fout	18:18:31.521512+0200	taskgated	cannot open file at line 44595 of [02c344acea]
fout	18:18:31.521553+0200	taskgated	os_unix.c:44595: (2) open(/var/db/DetachedSignatures) - No such file or directory
fout	18:18:31.568340+0200	lsd	cannot open file at line 44595 of [02c344acea]
fout	18:18:31.568367+0200	lsd	os_unix.c:44595: (2) open(/var/db/DetachedSignatures) - No such file or directory
fout	18:18:31.570106+0200	CoreServicesUIAgent	Error -60005 creating authorization
fout	18:18:31.574738+0200	sharingd	netfs_GetServerInfo returned 5
fout	18:18:31.603713+0200	sharingd	netfs_GetServerInfo returned 5
fout	18:18:31.636771+0200	launchservicesd	cannot open file at line 44595 of [02c344acea]
fout	18:18:31.636791+0200	launchservicesd	os_unix.c:44595: (2) open(/var/db/DetachedSignatures) - No such file or directory
fout	18:18:31.643887+0200	tccd	FAIL: PID[52429]: SecTaskCopySigningIdentifier(): [22: Invalid argument]
fout	18:18:31.644206+0200	tccd	FAIL: PID[52429]: SecTaskCopySigningIdentifier(): [22: Invalid argument]
fout	18:18:31.646059+0200	tccd	cannot open file at line 44595 of [02c344acea]
fout	18:18:31.646076+0200	tccd	os_unix.c:44595: (2) open(/var/db/DetachedSignatures) - No such file or directory
fout	18:18:31.648782+0200	tccd	cannot open file at line 44595 of [02c344acea]
fout	18:18:31.648804+0200	tccd	os_unix.c:44595: (2) open(/var/db/DetachedSignatures) - No such file or directory
fout	18:18:31.650445+0200	tccd	FAIL: PID[52429]: SecTaskCopySigningIdentifier(): [22: Invalid argument]
fout	18:18:31.651329+0200	tccd	FAIL: PID[52429]: SecTaskCopySigningIdentifier(): [22: Invalid argument]
fout	18:18:31.654118+0200	tccd	cannot open file at line 44595 of [02c344acea]
fout	18:18:31.654234+0200	tccd	os_unix.c:44595: (2) open(/var/db/DetachedSignatures) - No such file or directory
fout	18:18:31.656877+0200	tccd	cannot open file at line 44595 of [02c344acea]
fout	18:18:31.656899+0200	tccd	os_unix.c:44595: (2) open(/var/db/DetachedSignatures) - No such file or directory
fout	18:18:31.658906+0200	lsd	Failed to enumerate URLs under /private/var/folders/zw/f386642s0_j4bxq4nqy934bm0000gn/T/AppTranslocation/C0C5AE92-5308-4D2A-8131-928666918CF5 for SnapshotStorage reuse with error Error Domain=NSCocoaErrorDomain Code=256 "Het bestand 'C0C5AE92-5308-4D2A-8131-928666918CF5' kan niet worden geopend." UserInfo={NSURL=file:///private/var/folders/zw/f386642s0_j4bxq4nqy934bm0000gn/T/AppTranslocation/C0C5AE92-5308-4D2A-8131-928666918CF5, NSFilePath=/private/var/folders/zw/f386642s0_j4bxq4nqy934bm0000gn/T/AppTranslocation/C0C5AE92-5308-4D2A-8131-928666918CF5, NSUnderlyingError=0x7f895c42ffb0 {Error Domain=NSPOSIXErrorDomain Code=34 "Result too large"}}
fout	18:18:31.667722+0200	lsd	cannot open file at line 44595 of [02c344acea]
fout	18:18:31.667749+0200	lsd	os_unix.c:44595: (2) open(/var/db/DetachedSignatures) - No such file or directory
fout	18:18:31.672262+0200	ContextStoreAgent	cannot open file at line 44595 of [02c344acea]
fout	18:18:31.672396+0200	ContextStoreAgent	os_unix.c:44595: (2) open(/var/db/DetachedSignatures) - No such file or directory
fout	18:18:31.677670+0200	lsd	Failed to enumerate URLs under /private/var/folders/zw/f386642s0_j4bxq4nqy934bm0000gn/T/AppTranslocation/C0C5AE92-5308-4D2A-8131-928666918CF5 for SnapshotStorage reuse with error Error Domain=NSCocoaErrorDomain Code=256 "Het bestand 'C0C5AE92-5308-4D2A-8131-928666918CF5' kan niet worden geopend." UserInfo={NSURL=file:///private/var/folders/zw/f386642s0_j4bxq4nqy934bm0000gn/T/AppTranslocation/C0C5AE92-5308-4D2A-8131-928666918CF5, NSFilePath=/private/var/folders/zw/f386642s0_j4bxq4nqy934bm0000gn/T/AppTranslocation/C0C5AE92-5308-4D2A-8131-928666918CF5, NSUnderlyingError=0x7fa93fd12f30 {Error Domain=NSPOSIXErrorDomain Code=34 "Result too large"}}
fout	18:18:31.688875+0200	lsd	cannot open file at line 44595 of [02c344acea]
fout	18:18:31.688920+0200	lsd	os_unix.c:44595: (2) open(/var/db/DetachedSignatures) - No such file or directory
fout	18:18:31.723881+0200	tccd	cannot open file at line 44595 of [02c344acea]
fout	18:18:31.723911+0200	tccd	os_unix.c:44595: (2) open(/var/db/DetachedSignatures) - No such file or directory
fout	18:18:31.732669+0200	tccd	cannot open file at line 44595 of [02c344acea]
fout	18:18:31.732701+0200	tccd	os_unix.c:44595: (2) open(/var/db/DetachedSignatures) - No such file or directory
fout	18:18:31.757257+0200	kernel	Sandbox: com.apple.WebKit(25351) deny(1) mach-lookup com.apple.diagnosticd
fout	18:18:32.154225+0200	imklaunchagent	IMKLaunchAgent -requestIMKXPCEndpointInvalid: received notification Request for Endpoint Invalid (from pid_t: 52429) for bundleID: <private>
[... cut ...]
fout	18:19:21.117492+0200	kernel	Sandbox: com.apple.WebKit(27633) deny(1) mach-lookup com.apple.diagnosticd
fout	18:19:21.189752+0200	Jabra Bria Integration	nw_socket_handle_socket_event <private> Socket SO_ERROR [61: Connection refused]
fout	18:19:21.190016+0200	Jabra Bria Integration	nw_connection_get_connected_socket [C2360] Client called nw_connection_get_connected_socket on unconnected nw_connection
fout	18:19:21.190112+0200	Jabra Bria Integration	TCP Conn 0x7ff5de3b9090 Failed : error 0:61 [61]
`

lbibass, 2021-04-29:
> @akospalfi
> it's an assert error, probably not related to the issue. I just add some log to understand what's going on here.
> can someone on mac test the [action build 235](https://github.com/supermerill/SuperSlicer/suites/2613051641/artifacts/57249381) ?
> It should output some text on the console, can you copy them here?

Process:               SuperSlicer [49157]
Path:                  /Volumes/VOLUME/*/SuperSlicer.app/Contents/MacOS/SuperSlicer
Identifier:            com.merill.slic3r/
Version:               SuperSlicer Slic3r_2.3.56.0 (Slic3r_2.3.56.0)
Code Type:             X86-64 (Translated)
Parent Process:        ??? [1]
Responsible:           SuperSlicer [49157]
User ID:               501

Date/Time:             2021-04-29 13:54:52.956 -0400
OS Version:            macOS 11.3 (20E232)
Report Version:        12
Anonymous UUID:        A7484206-1361-AE4B-248C-74F12DFEDDC0

Sleep/Wake UUID:       7D7152AA-6161-49E1-A0CE-7B1AB72F13C3

Time Awake Since Boot: 77000 seconds
Time Since Wake:       150 seconds

System Integrity Protection: disabled

Notes:                 Translocated Process

Crashed Thread:        0  slic3r_main  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0xa49dffff800826a9 -> 0xffffffff800826a9 (possible pointer authentication failure)
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [49157]

VM Regions Near 0xffffffff800826a9:
--> mapped file              7fffe0000000-7fffe2e18000 [ 46.1M] r--/r-- SM=COW  Object_id=9d9061f7
    

Thread 0 Crashed:: slic3r_main  Dispatch queue: com.apple.main-thread
0   com.merill.slic3r/            	0x0000000104ae51d4 Slic3r::GUI::Tab::load_current_preset() + 980
1   com.merill.slic3r/            	0x00000001049eb7b8 Slic3r::GUI::GUI_App::load_current_presets(bool) + 376
2   com.merill.slic3r/            	0x00000001049e6e25 Slic3r::GUI::GUI_App::on_init_inner() + 4821
3   com.merill.slic3r/            	0x00000001049e5b0b Slic3r::GUI::GUI_App::OnInit() + 11
4   com.merill.slic3r/            	0x000000010583fda1 wxApp::CallOnInit() + 417
5   com.merill.slic3r/            	0x00000001059d27c6 wxEntry(int&, wchar_t**) + 198
6   com.merill.slic3r/            	0x00000001059d29fb wxEntry(int&, char**) + 59
7   com.merill.slic3r/            	0x00000001049d6c3c Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) + 236
8   com.merill.slic3r/            	0x0000000104130aab Slic3r::CLI::run(int, char**) + 17435
9   com.merill.slic3r/            	0x0000000104135337 main + 439
10  libdyld.dylib                 	0x00007fff20381f3d start + 1

Thread 1:: com.apple.rosetta.exceptionserver
0   runtime                       	0x00007ffdffd3f878 0x7ffdffd3e000 + 6264
1   runtime                       	0x00007ffdffd49d70 0x7ffdffd3e000 + 48496
2   runtime                       	0x00007ffdffd4aa40 0x7ffdffd3e000 + 51776

Thread 2:
0   runtime                       	0x00007ffdffd9899c 0x7ffdffd3e000 + 371100

Thread 3:
0   runtime                       	0x00007ffdffd9899c 0x7ffdffd3e000 + 371100

Thread 4:
0   runtime                       	0x00007ffdffd9899c 0x7ffdffd3e000 + 371100

Thread 5:
[... 423 lines cut ...]
__DATA_DIRTY                       659K      114 
__FONT_DATA                          4K        1 
__GLSLBUILTINS                    5176K        1 
__LINKEDIT                       527.4M       12 
__OBJC_RO                         70.1M        1 
__OBJC_RW                         2480K        2 
__TEXT                           319.2M      335 
__UNICODE                          588K        1 
mapped file                        4.1G      434 
shared memory                      800K       15 
unshared pmap                     8720K        5 
===========                     =======  ======= 
TOTAL                              6.6G     2022 
TOTAL, minus reserved VM space     5.6G     2022 

Model: MacBookPro17,1, BootROM 6723.101.4, proc 8:4:4 processors, 8 GB, SMC 
Graphics: kHW_AppleM1Item, Apple M1, spdisplays_builtin
Memory Module: LPDDR4
AirPort: spairport_wireless_card_type_airport_extreme, wl0: Apr  2 2021 00:48:12 version 18.52.1.2.7.8.116 FWID 01-9f7b4413
Bluetooth: Version 8.0.4d18, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB 3.1 Bus
USB Device: USB 3.1 Bus
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.

lbibass, 2021-04-29:
I also am now unable to launch the Brim Ears version as well, even though it used to launch correctly. 

```Process:               SuperSlicer [49434]
Path:                  /Users/USER/*/SuperSlicer.app/Contents/MacOS/SuperSlicer
Identifier:            com.merill.slic3r/
Version:               SuperSlicer SuperSlicer_2.3.55.5 (SuperSlicer_2.3.55.5)
Code Type:             X86-64 (Translated)
Parent Process:        ??? [1]
Responsible:           SuperSlicer [49434]
User ID:               501

Date/Time:             2021-04-29 13:58:49.436 -0400
OS Version:            macOS 11.3 (20E232)
Report Version:        12
Anonymous UUID:        A7484206-1361-AE4B-248C-74F12DFEDDC0

Sleep/Wake UUID:       7D7152AA-6161-49E1-A0CE-7B1AB72F13C3

Time Awake Since Boot: 77000 seconds
Time Since Wake:       380 seconds

System Integrity Protection: disabled

Crashed Thread:        0  slic3r_main  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000008
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [49434]

VM Regions Near 0x8:
--> 
    __TEXT                      1024c9000-104235000    [ 29.4M] r-x/r-x SM=COW  /Users/*/SuperSlicer.app/Contents/MacOS/SuperSlicer

Thread 0 Crashed:: slic3r_main  Dispatch queue: com.apple.main-thread
0   com.merill.slic3r/            	0x0000000103c1a94b wxMacCoreGraphicsPenBrushDataBase::CalculateShadingValues(void*, double const*, double*) + 59
1   com.apple.CoreGraphics        	0x00007fff24ea7971 CGFunctionEvaluate + 252
2   com.apple.CoreGraphics        	0x00007fff24ea77cd function_evaluate + 368
3   com.apple.CoreGraphics        	0x00007fff24ea6f52 ripc_AcquireFunction + 968
4   com.apple.CoreGraphics        	0x00007fff24ea5e8a ripc_DrawShading + 7598
5   com.apple.CoreGraphics        	0x00007fff2521fb76 CG::DisplayListExecutor::drawShading(CG::DisplayListEntryShading const*) + 894
6   com.apple.CoreGraphics        	0x00007fff252d7c68 CG::DisplayList::executeEntries(std::__1::__wrap_iter<std::__1::unique_ptr<CG::DisplayListEntry const, std::__1::default_delete<CG::DisplayListEntry const> >*>, std::__1::__wrap_iter<std::__1::unique_ptr<CG::DisplayListEntry const, std::__1::default_delete<CG::DisplayListEntry const> >*>, CGContextDelegate*, CGRenderingState*, CGGStack*, CGRect const*, __CFDictionary const*, bool) + 228
7   com.apple.CoreGraphics        	0x00007fff24f39193 CG::DisplayList::execute(CGContextDelegate*, CGRenderingState*, CGGStack*, CGRect const*, __CFDictionary const*) + 363
8   com.apple.CoreGraphics        	0x00007fff24f38d75 CGDisplayListDrawInContextDelegate + 701
9   com.apple.QuartzCore          	0x00007fff26e30f40 CABackingStoreUpdate_ + 570
10  com.apple.QuartzCore          	0x00007fff26e980f9 invocation function for block in CA::Layer::display_() + 53
11  com.apple.QuartzCore          	0x00007fff26e302f3 -[CALayer _display] + 2247
12  com.apple.AppKit              	0x00007fff22cf2ea5 -[_NSBackingLayer display] + 475
13  com.apple.AppKit              	0x00007fff22c5e00f -[_NSViewBackingLayer display] + 555
14  com.apple.QuartzCore          	0x00007fff26e2f25f CA::Layer::display_if_needed(CA::Transaction*) + 863
15  com.apple.QuartzCore          	0x00007fff26f7a44a CA::Context::commit_transaction(CA::Transaction*, double, double*) + 666
16  com.apple.QuartzCore          	0x00007fff26e1074f CA::Transaction::commit() + 713
17  com.apple.AppKit              	0x00007fff22d081dc __62+[CATransaction(NSCATransaction) NS_setFlushesWithDisplayLink]_block_invoke + 285
18  com.apple.AppKit              	0x00007fff2345c443 ___NSRunLoopObserverCreateWithHandler_block_invoke + 41
19  com.apple.CoreFoundation      	0x00007fff2045eb7d __CFRUNLOOP_IS_CALLING_OUT_TO_AN_OBSERVER_CALLBACK_FUNCTION__ + 23
20  com.apple.CoreFoundation      	0x00007fff2045ea0d __CFRunLoopDoObservers + 543
21  com.apple.CoreFoundation      	0x00007fff2045de94 __CFRunLoopRun + 845
22  com.apple.CoreFoundation      	0x00007fff2045d480 CFRunLoopRunSpecific + 567
23  com.apple.HIToolbox           	0x00007fff288e1203 RunCurrentEventLoopInMode + 292
24  com.apple.HIToolbox           	0x00007fff288e0e36 ReceiveNextEventCommon + 284
25  com.apple.HIToolbox           	0x00007fff288e0d03 _BlockUntilNextEventMatchingListInModeWithFilter + 70
26  com.apple.AppKit              	0x00007fff22b83b32 _DPSNextEvent + 864
27  com.apple.AppKit              	0x00007fff22b82305 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1364
28  com.apple.AppKit              	0x00007fff22b74679 -[NSApplication run] + 586
29  com.merill.slic3r/            	0x0000000103c6d6ee wxGUIEventLoop::OSXDoRun() + 174
30  com.merill.slic3r/            	0x0000000103dca7c7 wxCFEventLoop::DoRun() + 39
31  com.merill.slic3r/            	0x0000000103d03e7f wxEventLoopBase::Run() + 143
[... 420 lines cut ...]
Rosetta Return Stack               140K       14 
Rosetta Thread Context             140K       14 
STACK GUARD                         20K        5 
Stack                             11.1M        7 
Stack Guard                       56.0M        2 
VM_ALLOCATE                       4244K       30 
VM_ALLOCATE (reserved)               4K        1         reserved VM address space (unallocated)
__DATA                            15.2M      322 
__DATA_CONST                      16.2M      186 
[... cut ...]
__OBJC_RO                         70.1M        1 
__OBJC_RW                         2480K        2 
__TEXT                           315.9M      327 
__UNICODE                          588K        1 
mapped file                        4.1G      407 
shared memory                      800K       15 
unshared pmap                     7184K        4 
===========                     =======  ======= 
TOTAL                              6.3G     1929 
TOTAL, minus reserved VM space     5.6G     1929 

```

akospalfi, 2021-04-29:
@supermerill Oh, yes, sorry these asserts seemed so suspicious to me I forgot that I wanted to track the root of the crash...
I've checked out the same branch that was built by action 235, turned off the asserts and rerun the app in the debugger:

```
palfia@Akoss-iMac build % lldb ./src/superslicer
(lldb) target create "./src/superslicer"
Current executable set to '/Volumes/WORK/myprojects/SuperSlicer/SuperSlicer/build/src/superslicer' (x86_64).
(lldb) process launch -- --loglevel 5 --datadir "/Users/palfia/Library/Application Support/SuperSlicer"
Process 11654 launched: '/Volumes/WORK/myprojects/SuperSlicer/SuperSlicer/build/src/superslicer' (x86_64)
[2021-04-29 20:47:03.307465] [0x0000000111327e00] [debug]   full path: 16849348648804176793
[2021-04-29 20:47:03.307724] [0x0000000111327e00] [debug]   single instance: undefined. other params: /Volumes/WORK/myprojects/SuperSlicer/SuperSlicer/build/src/superslicer;--loglevel;5;--datadir;"/Users/palfia/Library/Application Support/SuperSlicer"
[2021-04-29 20:47:03.307738] [0x0000000111327e00] [debug]   full lock path: /Users/palfia/Library/Application Support/SuperSlicer/cache/16849348648804176793.lock
[2021-04-29 20:47:03.307792] [0x0000000111327e00] [info]    instance check: Another instance not found or single-instance not set.
2021-04-29 20:47:03.314583+0200 superslicer[11654:3334020] SecTaskLoadEntitlements failed error=22 cs_flags=20, pid=11654
2021-04-29 20:47:03.314653+0200 superslicer[11654:3334020] SecTaskCopyDebugDescription: superslicer[11654]/0#-1 LF=0
[2021-04-29 20:47:03.714388] [0x0000000111327e00] [info]    Checking if indices need to be installed from resources...
20:47:03: Debug: Adding duplicate image handler for 'JPEG file'
[2021-04-29 20:47:03.726337] [0x0000000111327e00] [trace]   Best translation language detected (may be different from user locales): en_GB
[2021-04-29 20:47:03.726379] [0x0000000111327e00] [trace]   Switching wxLocales to en_GB
BackgroundSlicingProcess:select_technology(1)
[2021-04-29 20:47:04.029943] [0x0000000111327e00] [info]    3dx mac handler starts
[2021-04-29 20:47:04.030196] [0x0000000111327e00] [error]   3dx drivers module loading error: dlopen(/Library/Frameworks/3DconnexionClient.framework/3DconnexionClient, 5): image not found
[2021-04-29 20:47:04.030208] [0x0000000111327e00] [info]    3dx drivers loaded: NO
[2021-04-29 20:47:04.032302] [0x0000000111327e00] [info]    Undo / Redo snapshot taken: New Project, Undo / Redo stack memory: 0MB Resident memory: 139MB; Peak memory usage: 139MB
BackgroundSlicingProcess:select_technology(2)
/Volumes/WORK/myprojects/SuperSlicer/SuperSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/osx/notebook_osx.cpp(148): assert ""((nPage) < GetPageCount())"" failed in SetPageImage(): SetPageImage: invalid notebook page
Collecting stack trace information, please wait...2021-04-29 20:47:17.760584+0200 superslicer[11654:3334020] SecTaskLoadEntitlements failed error=22 cs_flags=20, pid=11654
2021-04-29 20:47:17.760685+0200 superslicer[11654:3334020] SecTaskCopyDebugDescription: superslicer[11654]/0#-1 LF=0
/Volumes/WORK/myprojects/SuperSlicer/SuperSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/osx/notebook_osx.cpp(148): assert ""((nPage) < GetPageCount())"" failed in SetPageImage(): SetPageImage: invalid notebook page
Collecting stack trace information, please wait...[2021-04-29 20:47:32.717735] [0x0000000111327e00] [debug]   GLCanvas3D: Use Retina OpenGL: true
[2021-04-29 20:47:32.717810] [0x0000000111327e00] [debug]   GLCanvas3D: Use Retina OpenGL: true
printer_technology:from_preset(- default FFF -) = 44
BackgroundSlicingProcess:select_technology(44)
BackgroundSlicingProcess:select_technology(1)
/Volumes/WORK/myprojects/SuperSlicer/SuperSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/include/wx/vector.h(482): assert "idx < m_size" failed in at().
Collecting stack trace information, please wait...[2021-04-29 20:47:44.804145] [0x0000000111327e00] [debug]   GLCanvas3D: Use Retina OpenGL: true
[2021-04-29 20:47:44.804188] [0x0000000111327e00] [debug]   GLCanvas3D: Use Retina OpenGL: true
Process 11654 stopped
* thread #1, name = 'slic3r_main', queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x0)
    frame #0: 0x0000000101cc34b2 superslicer`Slic3r::GUI::Tab::load_current_preset(this=0x000000011198a800) at Tab.cpp:3033:67
   3030	                    }
   3031	                    else {
   3032	                        int page_id = wxGetApp().tab_panel()->FindPage(tab);
-> 3033	                        wxGetApp().tab_panel()->GetPage(page_id)->Show(false);
   3034	                        wxGetApp().tab_panel()->RemovePage(page_id);
   3035	                    }
   3036	                }
Target 0: (superslicer) stopped.
(lldb) po wxGetApp().tab_panel()->FindPage(tab)
warning: (x86_64) /Volumes/WORK/myprojects/SuperSlicer/SuperSlicer/build/src/slic3r/liblibslic3r_gui.a(Tab.cpp.o) 0xa900244050: DW_AT_specification(0x0016b3ac) has no decl
warning: (x86_64) /Volumes/WORK/myprojects/SuperSlicer/SuperSlicer/build/src/slic3r/liblibslic3r_gui.a(Tab.cpp.o) 0xa900244c4a: DW_AT_specification(0x0016b41a) has no decl
warning: (x86_64) /Volumes/WORK/myprojects/SuperSlicer/SuperSlicer/build/src/slic3r/liblibslic3r_gui.a(Tab.cpp.o) 0xa90024b400: DW_AT_specification(0x0016b43c) has no decl
-1

(lldb) po wxGetApp().tab_panel()->GetPage(page_id)
[... cut ...]
So the `page_id` is -1 at this point in `Tab.cpp:3033` and then the `wxGetApp().tab_panel()->GetPage(page_id)` returns a nullptr.
My other observation is that if I start the app with a totally empty profile then the app crashes on every startup at the point I quoted above. However if I use a populated profile then _sometimes_ the app can start (randomly crashes). But even when it loads then the print settings preset and filament settings preset combos are empty (and the tabs for these are also missing).

[image]

lbibass, 2021-04-29:
An update: commit 93e5584 works perfectly. I will try progressively more recent commits until it breaks.

supermerill, 2021-04-30:
@lbibass 
thank you, but akospalfi already did that, see https://github.com/supermerill/SuperSlicer/issues/1110#issuecomment-828871855

I submitted the nightly #236: log macos bug, to let you run it and post the output.
If the content is again insoluble, I'll try with partial merge of 715d58da888e591116a82f70ba6ff30020275ced

> edit: 3d1cd5c doesn't compile. ld: can't write output file to 'src/Slic3r' because that path is a directory

3d1cd5c  will fail, it's after dfc9671dc604a8b7154e288053929cd47cae3516
Can you try 715d58da888e591116a82f70ba6ff30020275ced ? merge 715d58da888e591116a82f70ba6ff30020275ced
 with d4f5d643f3c2b62b724651818b4037f0fd8dd164 for easier compilation if needed.

supermerill, 2021-04-30:
In the action menu, there is no the build 237 and 238
if someone can run them to see if they crash.

mvieleers, 2021-05-01:
238:
Loaded 10 out of 10 times without issues

237:
Loaded 6 out of 10 times. 
Crash log for one of the times it failed:
`Process:               SuperSlicer [78091]
Path:                  /private/var/folders/*/SuperSlicer.app/Contents/MacOS/SuperSlicer
Identifier:            com.merill.slic3r/
Version:               SuperSlicer Slic3r_2.3.56.0 (Slic3r_2.3.56.0)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           SuperSlicer [78091]
User ID:               501

Date/Time:             2021-05-01 11:41:18.814 +0200
OS Version:            macOS 11.3 (20E232)
Report Version:        12
Anonymous UUID:        397EF261-D7A2-469E-9453-47B88C171C7A

Time Awake Since Boot: 210000 seconds

System Integrity Protection: enabled

Notes:                 Translocated Process

Crashed Thread:        0  slic3r_main  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       EXC_I386_GPFLT
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [78091]

Thread 0 Crashed:: slic3r_main  Dispatch queue: com.apple.main-thread
0   com.merill.slic3r/            	0x00000001039f09d4 Slic3r::GUI::Tab::load_current_preset() + 964
1   com.merill.slic3r/            	0x00000001038f7518 Slic3r::GUI::GUI_App::load_current_presets(bool) + 248
2   com.merill.slic3r/            	0x00000001038f2d38 Slic3r::GUI::GUI_App::on_init_inner() + 3912
3   com.merill.slic3r/            	0x00000001038f1dab Slic3r::GUI::GUI_App::OnInit() + 11
4   com.merill.slic3r/            	0x00000001047452d1 wxApp::CallOnInit() + 417
5   com.merill.slic3r/            	0x00000001048d7cf6 wxEntry(int&, wchar_t**) + 198
6   com.merill.slic3r/            	0x00000001048d7f2b wxEntry(int&, char**) + 59
7   com.merill.slic3r/            	0x00000001038e2f3c Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) + 236
8   com.merill.slic3r/            	0x000000010303c66b Slic3r::CLI::run(int, char**) + 17435
9   com.merill.slic3r/            	0x0000000103040ef7 main + 439
10  libdyld.dylib                 	0x00007fff206c2f3d start + 1

Thread 1:
0   libsystem_pthread.dylib       	0x00007fff206a3484 start_wqthread + 0

Thread 2:
0   libsystem_pthread.dylib       	0x00007fff206a3484 start_wqthread + 0

Thread 3:
0   libsystem_pthread.dylib       	0x00007fff206a3484 start_wqthread + 0

Thread 4:
0   libsystem_pthread.dylib       	0x00007fff206a3484 start_wqthread + 0

Thread 5:: com.apple.NSEventThread
0   libsystem_kernel.dylib        	0x00007fff2067232a mach_msg_trap + 10
1   libsystem_kernel.dylib        	0x00007fff2067269c mach_msg + 60
2   com.apple.CoreFoundation      	0x00007fff207a0c6f __CFRunLoopServiceMachPort + 316
3   com.apple.CoreFoundation      	0x00007fff2079f34f __CFRunLoopRun + 1328
4   com.apple.CoreFoundation      	0x00007fff2079e75c CFRunLoopRunSpecific + 563
5   com.apple.AppKit              	0x00007fff23120baa _NSEventThread + 124
6   libsystem_pthread.dylib       	0x00007fff206a7954 _pthread_start + 224
7   libsystem_pthread.dylib       	0x00007fff206a34a7 thread_start + 15
[... 406 lines cut ...]
mapped file                      225.7M       18 
shared memory                     3440K       16 
===========                     =======  ======= 
TOTAL                              1.9G     1280 
TOTAL, minus reserved VM space     1.5G     1280 

Model: iMac18,3, BootROM 429.100.7.0.0, 4 processors, Quad-Core Intel Core i7, 4,2 GHz, 16 GB, SMC 2.41f2
Graphics: kHW_AMDRadeonPro580Item, Radeon Pro 580, spdisplays_pcie_device, 8 GB
Memory Module: BANK 0/DIMM0, 8 GB, DDR4 SO-DIMM, 2400 MHz, 0x802C, 0x3841544631473634485A2D324733423220202020
Memory Module: BANK 1/DIMM0, 8 GB, DDR4 SO-DIMM, 2400 MHz, 0x802C, 0x3841544631473634485A2D324733423220202020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x16F), Broadcom BCM43xx 1.0 (7.77.111.1 AirPortDriverBrcmNIC-1680.7)
Bluetooth: Version 8.0.4d18, 3 services, 27 devices, 1 incoming serial ports
Network Service: Ethernet, Ethernet, en0
Network Service: Wi-Fi, AirPort, en1
Network Service: Thunderbolt Ethernet Slot 1, Ethernet, en5
PCI Card: Aquantia AQC107, sppci_ethernet, Thunderbolt@11,0,0
PCI Card: pci144d,a808, sppci_nvme, Thunderbolt@135,0,0
Serial ATA Device: APPLE HDD ST3000DM001, 3 TB
USB Device: USB 3.0 Bus
USB Device: Bluetooth USB Host Controller
USB Device: FaceTime HD Camera (Built-in)
Thunderbolt Bus: iMac, Apple Inc., 41.4
Thunderbolt Device: Solo 10G Thunderbolt 3 Edition, Sonnet Technologies, Inc, 1, 33.1
Thunderbolt Device: Portable SSD X5, SAMSUNG ELECTRONICS CO., LTD, 3, 35.1
`

mvieleers, 2021-05-01:
Unrelated to the issue, but when I try to import my config bundle from Prusa Slicer, I get this error:

`ConfigBase::set_deserialize()` failed for parameter "gcode_flavor", value "reprapfirmware"

supermerill, 2021-05-01:
thank you @mvieleers

now, can somebody run the nightly build
 
* 239
* 240 if 239 is working
* 241 if 240 is working

and also give me the log of a run of the #236 build

> Unrelated to the issue, but when I try to import my config bundle from Prusa Slicer, I get this error:

k, will fix.

mvieleers, 2021-05-01:
239:
Attempt 1: Crashed, attempt 2: ok, attempt 3: crashed, attempt 4: ok, attempt5: crash, attempt6: ok, attempt 7: crashed: attempt 8: ok, attempt 9: crashed, attempt: 10: ok
Crash log:
`Process:               SuperSlicer [93056]
Path:                  /private/var/folders/*/SuperSlicer.app/Contents/MacOS/SuperSlicer
Identifier:            com.merill.slic3r/
Version:               SuperSlicer Slic3r_2.3.56.0 (Slic3r_2.3.56.0)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           SuperSlicer [93056]
User ID:               501

Date/Time:             2021-05-01 16:22:59.494 +0200
OS Version:            macOS 11.3 (20E232)
Report Version:        12
Anonymous UUID:        397EF261-D7A2-469E-9453-47B88C171C7A

Time Awake Since Boot: 230000 seconds

System Integrity Protection: enabled

Notes:                 Translocated Process

Crashed Thread:        0  slic3r_main  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       EXC_I386_GPFLT
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [93056]

Thread 0 Crashed:: slic3r_main  Dispatch queue: com.apple.main-thread
0   com.merill.slic3r/            	0x0000000106566aa4 Slic3r::GUI::Tab::load_current_preset() + 964
1   com.merill.slic3r/            	0x000000010646d8a8 Slic3r::GUI::GUI_App::load_current_presets(bool) + 248
2   com.merill.slic3r/            	0x00000001064690c8 Slic3r::GUI::GUI_App::on_init_inner() + 3912
3   com.merill.slic3r/            	0x000000010646813b Slic3r::GUI::GUI_App::OnInit() + 11
4   com.merill.slic3r/            	0x00000001072bb391 wxApp::CallOnInit() + 417
5   com.merill.slic3r/            	0x000000010744ddb6 wxEntry(int&, wchar_t**) + 198
6   com.merill.slic3r/            	0x000000010744dfeb wxEntry(int&, char**) + 59
7   com.merill.slic3r/            	0x00000001064592cc Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) + 236
8   com.merill.slic3r/            	0x0000000105bb32eb Slic3r::CLI::run(int, char**) + 17435
9   com.merill.slic3r/            	0x0000000105bb7b77 main + 439
10  libdyld.dylib                 	0x00007fff206c2f3d start + 1

Thread 1:
0   libsystem_pthread.dylib       	0x00007fff206a3484 start_wqthread + 0

Thread 2:
0   libsystem_pthread.dylib       	0x00007fff206a3484 start_wqthread + 0

Thread 3:
0   libsystem_pthread.dylib       	0x00007fff206a3484 start_wqthread + 0

Thread 4:
0   libsystem_pthread.dylib       	0x00007fff206a3484 start_wqthread + 0

Thread 5:: com.apple.NSEventThread
0   libsystem_kernel.dylib        	0x00007fff2067232a mach_msg_trap + 10
1   libsystem_kernel.dylib        	0x00007fff2067269c mach_msg + 60
2   com.apple.CoreFoundation      	0x00007fff207a0c6f __CFRunLoopServiceMachPort + 316
3   com.apple.CoreFoundation      	0x00007fff2079f34f __CFRunLoopRun + 1328
4   com.apple.CoreFoundation      	0x00007fff2079e75c CFRunLoopRunSpecific + 563
5   com.apple.AppKit              	0x00007fff23120baa _NSEventThread + 124
6   libsystem_pthread.dylib       	0x00007fff206a7954 _pthread_start + 224
7   libsystem_pthread.dylib       	0x00007fff206a34a7 thread_start + 15

Thread 0 crashed with X86 Thread State (64-bit):
  rax: 0x0000600003684400  rbx: 0xffffffffffffffff  rcx: 0xea9dffff803c34f1  rdx: 0x0000000000000012
[... 382 lines cut ...]
Foundation                          16K        1 
Kernel Alloc Once                    8K        1 
MALLOC                           240.2M       63 
MALLOC guard page                   32K        7 
MALLOC_NANO (reserved)           384.0M        1         reserved VM address space (unallocated)
STACK GUARD                       56.0M        6 
Stack                             10.5M        6 
VM_ALLOCATE                        372K       27 
__DATA                            27.7M      324 
__DATA_CONST                      16.2M      187 
__DATA_DIRTY                       657K      112 
__FONT_DATA                          4K        1 
__GLSLBUILTINS                    5176K        1 
__LINKEDIT                       520.5M        8 
__OBJC_RO                         70.2M        1 
__OBJC_RW                         2480K        2 
__TEXT                           316.8M      326 
__UNICODE                          588K        1 
mapped file                      226.0M       18 
shared memory                     2928K       16 
===========                     =======  ======= 
TOTAL                              1.9G     1258 
TOTAL, minus reserved VM space     1.5G     1258 

`

mvieleers, 2021-05-01:
Version 240: 
Able to open 3 out of 10 times.
Crash log:
`Process:               SuperSlicer [93236]
Path:                  /private/var/folders/*/SuperSlicer.app/Contents/MacOS/SuperSlicer
Identifier:            com.merill.slic3r/
Version:               SuperSlicer Slic3r_2.3.56.0 (Slic3r_2.3.56.0)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           SuperSlicer [93236]
User ID:               501

Date/Time:             2021-05-01 16:25:37.296 +0200
OS Version:            macOS 11.3 (20E232)
Report Version:        12
Anonymous UUID:        397EF261-D7A2-469E-9453-47B88C171C7A

Time Awake Since Boot: 230000 seconds

System Integrity Protection: enabled

Notes:                 Translocated Process

Crashed Thread:        0  slic3r_main  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       EXC_I386_GPFLT
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [93236]

Thread 0 Crashed:: slic3r_main  Dispatch queue: com.apple.main-thread
0   com.merill.slic3r/            	0x000000010113ca34 Slic3r::GUI::Tab::load_current_preset() + 964
1   com.merill.slic3r/            	0x0000000101043838 Slic3r::GUI::GUI_App::load_current_presets(bool) + 248
2   com.merill.slic3r/            	0x000000010103f058 Slic3r::GUI::GUI_App::on_init_inner() + 3912
3   com.merill.slic3r/            	0x000000010103e0cb Slic3r::GUI::GUI_App::OnInit() + 11
4   com.merill.slic3r/            	0x0000000101e91311 wxApp::CallOnInit() + 417
5   com.merill.slic3r/            	0x0000000102023d36 wxEntry(int&, wchar_t**) + 198
6   com.merill.slic3r/            	0x0000000102023f6b wxEntry(int&, char**) + 59
7   com.merill.slic3r/            	0x000000010102f25c Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) + 236
8   com.merill.slic3r/            	0x00000001007889eb Slic3r::CLI::run(int, char**) + 17435
9   com.merill.slic3r/            	0x000000010078d277 main + 439
10  libdyld.dylib                 	0x00007fff206c2f3d start + 1

Thread 1:
0   libsystem_pthread.dylib       	0x00007fff206a3484 start_wqthread + 0

Thread 2:
0   libsystem_pthread.dylib       	0x00007fff206a3484 start_wqthread + 0

Thread 3:
0   libsystem_pthread.dylib       	0x00007fff206a3484 start_wqthread + 0

Thread 4:
0   libsystem_pthread.dylib       	0x00007fff206a3484 start_wqthread + 0

Thread 5:: com.apple.NSEventThread
0   libsystem_kernel.dylib        	0x00007fff2067232a mach_msg_trap + 10
1   libsystem_kernel.dylib        	0x00007fff2067269c mach_msg + 60
2   com.apple.CoreFoundation      	0x00007fff207a0c6f __CFRunLoopServiceMachPort + 316
3   com.apple.CoreFoundation      	0x00007fff2079f34f __CFRunLoopRun + 1328
4   com.apple.CoreFoundation      	0x00007fff2079e75c CFRunLoopRunSpecific + 563
5   com.apple.AppKit              	0x00007fff23120baa _NSEventThread + 124
6   libsystem_pthread.dylib       	0x00007fff206a7954 _pthread_start + 224
7   libsystem_pthread.dylib       	0x00007fff206a34a7 thread_start + 15

Thread 6:
0   libsystem_pthread.dylib       	0x00007fff206a3484 start_wqthread + 0
[... 403 lines cut ...]
mapped file                      226.1M       18 
shared memory                     2928K       16 
===========                     =======  ======= 
TOTAL                              1.9G     1262 
TOTAL, minus reserved VM space     1.5G     1262 

Model: iMac18,3, BootROM 429.100.7.0.0, 4 processors, Quad-Core Intel Core i7, 4,2 GHz, 16 GB, SMC 2.41f2
Graphics: kHW_AMDRadeonPro580Item, Radeon Pro 580, spdisplays_pcie_device, 8 GB
Memory Module: BANK 0/DIMM0, 8 GB, DDR4 SO-DIMM, 2400 MHz, 0x802C, 0x3841544631473634485A2D324733423220202020
Memory Module: BANK 1/DIMM0, 8 GB, DDR4 SO-DIMM, 2400 MHz, 0x802C, 0x3841544631473634485A2D324733423220202020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x16F), Broadcom BCM43xx 1.0 (7.77.111.1 AirPortDriverBrcmNIC-1680.7)
Bluetooth: Version 8.0.4d18, 3 services, 27 devices, 1 incoming serial ports
Network Service: Ethernet, Ethernet, en0
Network Service: Wi-Fi, AirPort, en1
Network Service: Thunderbolt Ethernet Slot 1, Ethernet, en5
PCI Card: Aquantia AQC107, sppci_ethernet, Thunderbolt@11,0,0
PCI Card: pci144d,a808, sppci_nvme, Thunderbolt@135,0,0
Serial ATA Device: APPLE HDD ST3000DM001, 3 TB
USB Device: USB 3.0 Bus
USB Device: Bluetooth USB Host Controller
USB Device: FaceTime HD Camera (Built-in)
Thunderbolt Bus: iMac, Apple Inc., 41.4
Thunderbolt Device: Solo 10G Thunderbolt 3 Edition, Sonnet Technologies, Inc, 1, 33.1
Thunderbolt Device: Portable SSD X5, SAMSUNG ELECTRONICS CO., LTD, 3, 35.1
`

mvieleers, 2021-05-01:
Version 241:
Crashed 5 out of 10 times.
Crash log:
`Process:               SuperSlicer [93342]
Path:                  /private/var/folders/*/SuperSlicer.app/Contents/MacOS/SuperSlicer
Identifier:            com.merill.slic3r/
Version:               SuperSlicer Slic3r_2.3.56.0 (Slic3r_2.3.56.0)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           SuperSlicer [93342]
User ID:               501

Date/Time:             2021-05-01 16:27:13.138 +0200
OS Version:            macOS 11.3 (20E232)
Report Version:        12
Anonymous UUID:        397EF261-D7A2-469E-9453-47B88C171C7A

Time Awake Since Boot: 230000 seconds

System Integrity Protection: enabled

Notes:                 Translocated Process

Crashed Thread:        0  slic3r_main  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       EXC_I386_GPFLT
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [93342]

Thread 0 Crashed:: slic3r_main  Dispatch queue: com.apple.main-thread
0   com.merill.slic3r/            	0x000000010cf40a44 Slic3r::GUI::Tab::load_current_preset() + 964
1   com.merill.slic3r/            	0x000000010ce47598 Slic3r::GUI::GUI_App::load_current_presets(bool) + 248
2   com.merill.slic3r/            	0x000000010ce42db8 Slic3r::GUI::GUI_App::on_init_inner() + 3912
3   com.merill.slic3r/            	0x000000010ce41e2b Slic3r::GUI::GUI_App::OnInit() + 11
4   com.merill.slic3r/            	0x000000010dc95311 wxApp::CallOnInit() + 417
5   com.merill.slic3r/            	0x000000010de27d36 wxEntry(int&, wchar_t**) + 198
6   com.merill.slic3r/            	0x000000010de27f6b wxEntry(int&, char**) + 59
7   com.merill.slic3r/            	0x000000010ce32fbc Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) + 236
8   com.merill.slic3r/            	0x000000010c58c72b Slic3r::CLI::run(int, char**) + 17435
9   com.merill.slic3r/            	0x000000010c590fb7 main + 439
10  libdyld.dylib                 	0x00007fff206c2f3d start + 1

Thread 1:
0   libsystem_pthread.dylib       	0x00007fff206a3484 start_wqthread + 0

Thread 2:
0   libsystem_pthread.dylib       	0x00007fff206a3484 start_wqthread + 0

Thread 3:
0   libsystem_pthread.dylib       	0x00007fff206a3484 start_wqthread + 0

Thread 4:
0   libsystem_pthread.dylib       	0x00007fff206a3484 start_wqthread + 0

Thread 5:: com.apple.NSEventThread
0   libsystem_kernel.dylib        	0x00007fff2067232a mach_msg_trap + 10
1   libsystem_kernel.dylib        	0x00007fff2067269c mach_msg + 60
2   com.apple.CoreFoundation      	0x00007fff207a0c6f __CFRunLoopServiceMachPort + 316
3   com.apple.CoreFoundation      	0x00007fff2079f34f __CFRunLoopRun + 1328
4   com.apple.CoreFoundation      	0x00007fff2079e75c CFRunLoopRunSpecific + 563
5   com.apple.AppKit              	0x00007fff23120baa _NSEventThread + 124
6   libsystem_pthread.dylib       	0x00007fff206a7954 _pthread_start + 224
7   libsystem_pthread.dylib       	0x00007fff206a34a7 thread_start + 15

Thread 0 crashed with X86 Thread State (64-bit):
  rax: 0x0000600001e86000  rbx: 0xffffffffffffffff  rcx: 0xfd9dffff803c34f1  rdx: 0x00000000a5c48827
[... 382 lines cut ...]
Foundation                          16K        1 
Kernel Alloc Once                    8K        1 
MALLOC                           241.2M       64 
MALLOC guard page                   32K        8 
MALLOC_NANO (reserved)           384.0M        1         reserved VM address space (unallocated)
STACK GUARD                       56.0M        6 
Stack                             10.5M        6 
VM_ALLOCATE                        368K       26 
__DATA                            27.7M      324 
__DATA_CONST                      16.2M      187 
__DATA_DIRTY                       657K      112 
__FONT_DATA                          4K        1 
__GLSLBUILTINS                    5176K        1 
__LINKEDIT                       520.5M        8 
__OBJC_RO                         70.2M        1 
__OBJC_RW                         2480K        2 
__TEXT                           316.8M      326 
__UNICODE                          588K        1 
mapped file                      225.5M       18 
shared memory                     2928K       16 
===========                     =======  ======= 
TOTAL                              1.9G     1285 
TOTAL, minus reserved VM space     1.5G     1285 

`

mvieleers, 2021-05-01:
> and also give me the log of a run of the 236 build

Happy to do this, but not really sure how I can provide you this log....

supermerill, 2021-05-01:
> Happy to do this, but not really sure how I can provide you this log....

on linux, I would start the software in a console and wrote here the lines that would appear on the console.

mvieleers, 2021-05-01:
I don't have linux. On a Mac. I have a the console.app. This is what I was able to capture in that regarding 236. Hope it helps in some way.
`fout	17:08:33.230231+0200	kernel	Sandbox: com.apple.WebKit(85458) deny(1) mach-lookup com.apple.diagnosticd
standaard	17:08:33.648046+0200	lsd	*** Registration warnings for <private>:
standaard	17:08:33.648072+0200	lsd	    Missing iTunes metadata
standaard	17:08:33.651030+0200	lsd	Non-fatal error enumerating at <private>, continuing: Error Domain=NSCocoaErrorDomain Code=260 "Het bestand 'PlugIns' kan niet worden geopend, omdat het bestand niet bestaat." UserInfo={NSURL=PlugIns/ -- file:///Volumes/com.apple.TimeMachine.localsnapshots/Backups.backupdb/iMac%20van%20Marc/2021-05-01-170310/Harddisk%20-%20Data/Users/marc/Library/Containers/com.microsoft.openxml.excel.app/, NSFilePath=/Volumes/com.apple.TimeMachine.localsnapshots/Backups.backupdb/iMac van Marc/2021-05-01-170310/Harddisk - Data/Users/marc/Library/Containers/com.microsoft.openxml.excel.app/PlugIns, NSUnderlyingError=0x7fa941deaa00 {Error Domain=NSPOSIXErrorDomain Code=2 "No such file or directory"}}
standaard	17:08:33.651109+0200	lsd	(null):5:4:1:0:_LSServerRegisterItemInfo begin
standaard	17:08:33.652978+0200	lsd	MacOS error: -67028
standaard	17:08:33.653976+0200	lsd	(null):5:5:1:1:Building bundle record for app
standaard	17:08:33.654028+0200	lsd	Parsed Info.plist for (null)
standaard	17:08:33.654043+0200	lsd	Parsed MobileInstallation data for (null)
standaard	17:08:33.654061+0200	lsd	(null):5:5:2:1:Built bundle record for app
standaard	17:08:33.654942+0200	lsd	(null):5:4:2:1:_LSServerRegisterItemInfo result = 0
standaard	17:08:33.655348+0200	UserEventAgent	Fired subscription for notification: <private>
standaard	17:08:33.655486+0200	UserEventAgent	Fired subscription for notification: <private>
standaard	17:08:33.655606+0200	UserEventAgent	Fired subscription for notification: <private>
standaard	17:08:33.655709+0200	nehelper	Handling an apps installed notification with bundle IDs (null)
standaard	17:08:33.655724+0200	UserEventAgent	Fired subscription for notification: <private>
standaard	17:08:33.655775+0200	nehelper	UUID cache generation changed from 3273 to 3274
standaard	17:08:33.744079+0200	nehelper	Setting UUID cache generation to 3275
standaard	17:08:34.081049+0200	Jabra Bria Integration	TCP Conn 0x7ff5cdc54240 canceled
standaard	17:08:34.081096+0200	Jabra Bria Integration	[C13596 9170E7F3-A79E-49E0-BCAE-49C072CF45D7 127.0.0.1:9002 tcp, legacy-socket] cancel
standaard	17:08:34.081126+0200	Jabra Bria Integration	[C13596 127.0.0.1:9002 tcp, legacy-socket] cancelled
standaard	17:08:34.081186+0200	Jabra Bria Integration	TCP Conn [13597:0x7ff5cdc55af0] using empty proxy configuration
standaard	17:08:34.081197+0200	Jabra Bria Integration	Stream client bypassing proxies on TCP Conn [13597:0x7ff5cdc55af0]
standaard	17:08:34.081209+0200	Jabra Bria Integration	nw_connection_report_state_with_handler_on_nw_queue [C13596] reporting state cancelled error Connection refused
standaard	17:08:34.081227+0200	Jabra Bria Integration	TCP Conn 0x7ff5cdc55af0 started
standaard	17:08:34.081496+0200	Jabra Bria Integration	[C13597 6202C0F7-F5E4-4F44-AEE3-D0775B713B81 127.0.0.1:9002 tcp, context: Default Network Context, proc: 036DB53E-85C8-3A04-8A1E-1FD60EFCF3B6, no proxy, no fallback, allow socket access] start
standaard	17:08:34.081537+0200	Jabra Bria Integration	[C13597 127.0.0.1:9002 initial path ((null))] event: path:start @0.000s
standaard	17:08:34.081581+0200	Jabra Bria Integration	[C13597 127.0.0.1:9002 waiting path (satisfied (Path is satisfied), interface: lo0)] event: path:satisfied @0.000s, uuid: 63FA1F33-8048-4065-AD50-D3E3D7292983
standaard	17:08:34.081799+0200	Jabra Bria Integration	[C13597 127.0.0.1:9002 in_progress socket-flow (satisfied (Path is satisfied), interface: lo0)] event: flow:start_connect @0.000s
standaard	17:08:34.081811+0200	Jabra Bria Integration	nw_connection_report_state_with_handler_on_nw_queue [C13597] reporting state preparing
standaard	17:08:34.081988+0200	Jabra Bria Integration	nw_socket_handle_socket_event [C13597:1] Socket received CONNRESET event
standaard	17:08:34.081999+0200	Jabra Bria Integration	nw_socket_handle_socket_event [C13597:1] Socket received DISCONNECTED event
fout	17:08:34.082012+0200	Jabra Bria Integration	nw_socket_handle_socket_event <private> Socket SO_ERROR [61: Connection refused]
standaard	17:08:34.082043+0200	Jabra Bria Integration	nw_flow_disconnected [C13597 127.0.0.1:9002 in_progress socket-flow (satisfied (Path is satisfied), viable, interface: lo0)] Output protocol disconnected
standaard	17:08:34.082087+0200	Jabra Bria Integration	[C13597 127.0.0.1:9002 failed socket-flow (satisfied (Path is satisfied), viable, interface: lo0)] event: flow:failed_connect @0.000s, error Connection refused
standaard	17:08:34.082103+0200	Jabra Bria Integration	nw_connection_report_state_with_handler_on_nw_queue [C13597] reporting state failed error Connection refused
standaard	17:08:34.082189+0200	Jabra Bria Integration	TCP Conn 0x7ff5cdc55af0 event 3. err: 61
fout	17:08:34.082214+0200	Jabra Bria Integration	nw_connection_get_connected_socket [C13597] Client called nw_connection_get_connected_socket on unconnected nw_connection
standaard	17:08:34.082279+0200	Jabra Bria Integration	TCP Conn 0x7ff5cdc55af0 complete. fd: -1, err: 61
fout	17:08:34.082317+0200	Jabra Bria Integration	TCP Conn 0x7ff5cdc55af0 Failed : error 0:61 [61]
[... cut ...]
standaard	17:08:42.051755+0200	kernel	com.apple.p2p: monitorAWDLState: Number of peers found: 1
standaard	17:08:42.051842+0200	kernel	com.apple.p2p: LQM-WiFi:AWDL State #15 Data(5)  DutyCycle 0  StateDuration 2lu.007s[233919.894 - 233921.902]   StateComplete  1
standaard	17:08:42.051845+0200	kernel	com.apple.p2p: LQM-WiFi:AWDL State #16 Idle(3)  DutyCycle 40  StateDuration 3lu.105s[233921.902 - 0.000]   StateComplete  0
standaard	17:08:42.051846+0200	kernel	com.apple.p2p: LQM-WiFi:AWDL Active Time 1242ms(24%) 5.003s[233920.004s - 233925.008s]
`

akospalfi, 2021-05-01:
@mvieleers @supermerill On macOS you can do this by opening the terminal and paste this line:

`/Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer --loglevel 5`

akospalfi, 2021-05-01:
Output of build 236:

```
palfia@Akoss-MacBook-Pro ~ % /Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer --loglevel 5
set enum to monotonic: 0=>23
set enum to concentric: 0=>6
set enum to object: 0=>0
set enum to stars: 0=>3
set enum to sprinter: 0=>1
set enum to connected: 0=>0
set enum to connected: 0=>0
set enum to connected: 0=>0
set enum to connected: 0=>0
set enum to autoenlarged: 0=>2
set enum to top: 0=>0
set enum to time_estimate_only: 0=>1
set enum to none: 0=>0
set enum to rear: 0=>3
set enum to nearest: 0=>1
set enum to rectilineargapfill: 0=>22
set enum to plane: 0=>1
set enum to rectilinear: 0=>0
set enum to rectilinear: 0=>0
set enum to monotonic: 0=>23
set enum to linear: 0=>0
set enum to portrait: 0=>1
set enum to mCWS: 0=>1
set enum to FFF: 0=>1
set enum to dynamic: 0=>2
[2021-05-01 17:14:43.109304] [0x00000001113dde00] [debug]   full path: 15840400373692917632
[2021-05-01 17:14:43.109683] [0x00000001113dde00] [debug]   single instance: undefined. other params: /Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer;--loglevel;5
[2021-05-01 17:14:43.109691] [0x00000001113dde00] [debug]   full lock path: /Users/palfia/Library/Application Support/SuperSlicer/cache/15840400373692917632.lock
[2021-05-01 17:14:43.109706] [0x00000001113dde00] [debug]   get_lock(): datadir does not exist yet, creating...
[2021-05-01 17:14:43.110012] [0x00000001113dde00] [info]    instance check: Another instance not found or single-instance not set.
GUI_App Bundle::new
create PresetBundle with default printer as 105
GUI_App:on_init_inner test point 1 : (- default FFF -) = 140
GUI_App:on_init_inner test point 2 : (- default FFF -) = 140
GUI_App:on_init_inner test point 4 : (- default FFF -) = 140
[2021-05-01 17:14:43.362465] [0x00000001113dde00] [info]    Checking if indices need to be installed from resources...
GUI_App:on_init_inner test point 6 : (- default FFF -) = 140
GUI_App:on_init_inner test point 7 : (- default FFF -) = 140
[2021-05-01 17:14:43.365489] [0x00000001113dde00] [trace]   Best translation language detected (may be different from user locales): en_GB
[2021-05-01 17:14:43.365512] [0x00000001113dde00] [trace]   Switching wxLocales to en_GB
GUI_App:on_init_inner test point 8 : (- default FFF -) = 140
GUI_App:on_init_inner test point 9 : (- default FFF -) = 136
GUI_App:on_init_inner test point 10 : (- default FFF -) = 136
set enum to object: 0=>0
set enum to sprinter: 0=>1
set enum to time_estimate_only: 0=>1
set enum to linear: 0=>0
set enum to concentric: 0=>6
set enum to nearest: 0=>1
set enum to plane: 0=>1
set enum to rectilinear: 0=>0
set enum to rectilinear: 0=>0
set enum to monotonic: 0=>23
set enum to stars: 0=>3
set enum to connected: 0=>0
set enum to connected: 0=>0
set enum to connected: 0=>0
set enum to connected: 0=>0
set enum to autoenlarged: 0=>2
set enum to top: 0=>0
set enum to none: 0=>0
set enum to rear: 0=>3
set enum to rectilineargapfill: 0=>22
set enum to monotonic: 0=>23
set enum to portrait: 0=>1
set enum to mCWS: 0=>1
set enum to FFF: 0=>1
set enum to dynamic: 0=>2
BackgroundSlicingProcess:select_technology(1)
[2021-05-01 17:14:43.536434] [0x00000001113dde00] [info]    3dx mac handler starts
[2021-05-01 17:14:43.536995] [0x00000001113dde00] [error]   3dx drivers module loading error: dlopen(/Library/Frameworks/3DconnexionClient.framework/3DconnexionClient, 5): image not found
[2021-05-01 17:14:43.537042] [0x00000001113dde00] [info]    3dx drivers loaded: NO
[2021-05-01 17:14:43.539852] [0x00000001113dde00] [info]    Undo / Redo snapshot taken: New Project, Undo / Redo stack memory: 0MB Resident memory: 106MB; Peak memory usage: 106MB
BackgroundSlicingProcess:select_technology(2)
[2021-05-01 17:14:43.642101] [0x00000001113dde00] [debug]   GLCanvas3D: Use Retina OpenGL: true
[2021-05-01 17:14:43.642134] [0x00000001113dde00] [debug]   GLCanvas3D: Use Retina OpenGL: true
GUI_App:on_init_inner test point 11 : (- default FFF -) = 136
GUI_App:on_init_inner test point 12 : (- default FFF -) = 136
GUI_App:on_init_inner test point 13 : (- default FFF -) = 136
GUI_App:on_init_inner test point 14 : (- default FFF -) = 136
GUI_App:load_current_presets start : (- default FFF -) = 136
GUI_App:load_current_presets check_printer_presets : (- default FFF -) = 136
GUI_App:load_current_presets check_printer_presets : (- default FFF -) = 136
BackgroundSlicingProcess:select_technology(136)
BackgroundSlicingProcess:select_technology(1)
[2021-05-01 17:14:43.902191] [0x00000001113dde00] [debug]   TriangleMesh::repair() started
[2021-05-01 17:14:43.902538] [0x00000001113dde00] [debug]   TriangleMesh::repair() finished
[2021-05-01 17:14:43.902547] [0x00000001113dde00] [trace]   TriangleMeshSlicer::require_shared_vertices - start
[2021-05-01 17:14:43.902551] [0x00000001113dde00] [trace]   TriangleMeshSlicer::require_shared_vertices - stl_generate_shared_vertices
[2021-05-01 17:14:43.902568] [0x00000001113dde00] [trace]   TriangleMeshSlicer::require_shared_vertices - end
[2021-05-01 17:14:43.902658] [0x00000001113dde00] [debug]   TriangleMesh::repair() started
[2021-05-01 17:14:43.902851] [0x00000001113dde00] [debug]   TriangleMesh::repair() finished
[2021-05-01 17:14:43.902857] [0x00000001113dde00] [trace]   TriangleMeshSlicer::require_shared_vertices - start
[2021-05-01 17:14:43.902879] [0x00000001113dde00] [trace]   TriangleMeshSlicer::require_shared_vertices - stl_generate_shared_vertices
[2021-05-01 17:14:43.902892] [0x00000001113dde00] [trace]   TriangleMeshSlicer::require_shared_vertices - end
[2021-05-01 17:14:44.357920] [0x00000001113dde00] [info]    Running ConfigWizard, reason: 0, start_page: 0
[2021-05-01 17:14:44.498594] [0x00000001113dde00] [debug]   GLCanvas3D: Use Retina OpenGL: true
[2021-05-01 17:14:44.498622] [0x00000001113dde00] [debug]   GLCanvas3D: Use Retina OpenGL: true
```

mvieleers, 2021-05-01:
@akospalfi thanks for the instructions.

@supermerill if it helps to have a second output, here is my output of 236:

`marc@iMac-van-Marc ~ % /Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer --loglevel 5
set enum to monotonic: 0=>23
set enum to concentric: 0=>6
set enum to object: 0=>0
set enum to stars: 0=>3
set enum to sprinter: 0=>1
set enum to connected: 0=>0
set enum to connected: 0=>0
set enum to connected: 0=>0
set enum to connected: 0=>0
set enum to autoenlarged: 0=>2
set enum to top: 0=>0
set enum to time_estimate_only: 0=>1
set enum to none: 0=>0
set enum to rear: 0=>3
set enum to nearest: 0=>1
set enum to rectilineargapfill: 0=>22
set enum to plane: 0=>1
set enum to rectilinear: 0=>0
set enum to rectilinear: 0=>0
set enum to monotonic: 0=>23
set enum to linear: 0=>0
set enum to portrait: 0=>1
set enum to mCWS: 0=>1
set enum to FFF: 0=>1
set enum to dynamic: 0=>2
[2021-05-01 17:14:55.330115] [0x0000000116c38e00] [debug]   full path: 15840400373692917632
[2021-05-01 17:14:55.330384] [0x0000000116c38e00] [debug]   single instance: undefined. other params: /Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer;--loglevel;5
[2021-05-01 17:14:55.330391] [0x0000000116c38e00] [debug]   full lock path: /Users/marc/Library/Application Support/SuperSlicer/cache/15840400373692917632.lock
[2021-05-01 17:14:55.330442] [0x0000000116c38e00] [info]    instance check: Another instance not found or single-instance not set.
GUI_App Bundle::new
create PresetBundle with default printer as 0
GUI_App:on_init_inner test point 1 : (- default FFF -) = 0
GUI_App:on_init_inner test point 2 : (- default FFF -) = 0
GUI_App:on_init_inner test point 4 : (- default FFF -) = 0
[2021-05-01 17:14:55.520556] [0x0000000116c38e00] [info]    Checking if indices need to be installed from resources...
GUI_App:on_init_inner test point 6 : (- default FFF -) = 0
GUI_App:on_init_inner test point 7 : (- default FFF -) = 0
[2021-05-01 17:14:55.521542] [0x0000000116c38e00] [trace]   System language detected (user locales and such): nl_NL
[2021-05-01 17:14:55.523729] [0x0000000116c38e00] [trace]   Best translation language detected (may be different from user locales): en_GB
[2021-05-01 17:14:55.523748] [0x0000000116c38e00] [trace]   Switching wxLocales to en_GB
GUI_App:on_init_inner test point 8 : (- default FFF -) = 0
create PresetBundle with default printer as 0
set enum to FFF: 1=>1
set enum to octoprint: 0=>0
set enum to key: 0=>0
set enum to FFF: 1=>1
set enum to octoprint: 0=>0
set enum to key: 0=>0
set enum to FFF: 1=>1
set enum to octoprint: 0=>0
set enum to key: 0=>0
set enum to FFF: 1=>1
set enum to octoprint: 0=>0
set enum to key: 0=>0
GUI_App:on_init_inner test point 9 : (0.4mm nozzle - Voron) = 0
GUI_App:on_init_inner test point 10 : (0.4mm nozzle - Voron) = 0
set enum to object: 0=>0
set enum to sprinter: 0=>1
set enum to time_estimate_only: 0=>1
set enum to linear: 0=>0
set enum to concentric: 0=>6
set enum to nearest: 0=>1
set enum to plane: 0=>1
set enum to rectilinear: 0=>0
set enum to rectilinear: 0=>0
[... 77 lines cut ...]
< X-Frame-Options: deny
< X-Content-Type-Options: nosniff
< X-XSS-Protection: 0
< Referrer-Policy: origin-when-cross-origin, strict-origin-when-cross-origin
< Content-Security-Policy: default-src 'none'
< Content-Type: application/json; charset=utf-8
< X-GitHub-Media-Type: github.v3; format=json
< Link: <https://api.github.com/repositories/116708603/releases?page=2>; rel="next", <https://api.github.com/repositories/116708603/releases?page=2>; rel="last"
< ETag: W/"c4f4500f0cfd09e43e5d72e68f5bb840a1d0213c0a64be7c91d002f06a1e1b3c"
< X-RateLimit-Limit: 60
< X-RateLimit-Remaining: 41
< X-RateLimit-Reset: 1619882608
< X-RateLimit-Used: 19
< Accept-Ranges: bytes
< Content-Length: 214479
< X-GitHub-Request-Id: C64C:1BB0:B29488:BB5957:608D7072
< 
* Connection #0 to host api.github.com left intact
[2021-05-01 17:14:56.806467] [0x000070000e4a9000] [info]    Syncing configuration cache
[2021-05-01 17:14:56.806482] [0x000070000e4a9000] [warning] No such vendor: BIBO
[2021-05-01 17:14:56.806485] [0x000070000e4a9000] [warning] No such vendor: LulzBot
[2021-05-01 17:14:56.806489] [0x000070000e4a9000] [info]    Vendor has no config_update_url: Prusa {technology}
[2021-05-01 17:14:56.806492] [0x000070000e4a9000] [info]    Vendor has no config_update_url: Voron
[2021-05-01 17:14:56.806494] [0x000070000e4a9000] [warning] No such vendor: Creality
[2021-05-01 17:15:05.696668] [0x0000000116c38e00] [debug]   message handler shutdown().`

supermerill, 2021-05-01:
while I'm trying to make sense of the results, can someone make the build 242 fail?

mvieleers, 2021-05-01:
Broke it on the first try ;-)

Here's the crash report for 242
```
Process:               SuperSlicer [1334]
Path:                  /private/var/folders/*/SuperSlicer.app/Contents/MacOS/SuperSlicer
Identifier:            com.merill.slic3r/
Version:               SuperSlicer Slic3r_2.3.56.0 (Slic3r_2.3.56.0)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           SuperSlicer [1334]
User ID:               501

Date/Time:             2021-05-01 18:58:49.492 +0200
OS Version:            macOS 11.3 (20E232)
Report Version:        12
Anonymous UUID:        397EF261-D7A2-469E-9453-47B88C171C7A

Time Awake Since Boot: 240000 seconds

System Integrity Protection: enabled

Notes:                 Translocated Process

Crashed Thread:        0  slic3r_main  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       EXC_I386_GPFLT
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [1334]

Thread 0 Crashed:: slic3r_main  Dispatch queue: com.apple.main-thread
0   com.merill.slic3r/            	0x000000010f973ad4 Slic3r::GUI::Tab::load_current_preset() + 964
1   com.merill.slic3r/            	0x000000010f87a8d8 Slic3r::GUI::GUI_App::load_current_presets(bool) + 248
2   com.merill.slic3r/            	0x000000010f8760f8 Slic3r::GUI::GUI_App::on_init_inner() + 3912
3   com.merill.slic3r/            	0x000000010f87516b Slic3r::GUI::GUI_App::OnInit() + 11
4   com.merill.slic3r/            	0x00000001106c8391 wxApp::CallOnInit() + 417
5   com.merill.slic3r/            	0x000000011085adb6 wxEntry(int&, wchar_t**) + 198
6   com.merill.slic3r/            	0x000000011085afeb wxEntry(int&, char**) + 59
7   com.merill.slic3r/            	0x000000010f8662fc Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) + 236
8   com.merill.slic3r/            	0x000000010efc02eb Slic3r::CLI::run(int, char**) + 17435
9   com.merill.slic3r/            	0x000000010efc4b77 main + 439
10  libdyld.dylib                 	0x00007fff206c2f3d start + 1

Thread 1:
0   libsystem_pthread.dylib       	0x00007fff206a3484 start_wqthread + 0

Thread 2:
0   libsystem_pthread.dylib       	0x00007fff206a3484 start_wqthread + 0

Thread 3:
0   libsystem_pthread.dylib       	0x00007fff206a3484 start_wqthread + 0

Thread 4:
0   libsystem_pthread.dylib       	0x00007fff206a3484 start_wqthread + 0

Thread 5:: com.apple.NSEventThread
0   libsystem_kernel.dylib        	0x00007fff2067232a mach_msg_trap + 10
1   libsystem_kernel.dylib        	0x00007fff2067269c mach_msg + 60
2   com.apple.CoreFoundation      	0x00007fff207a0c6f __CFRunLoopServiceMachPort + 316
3   com.apple.CoreFoundation      	0x00007fff2079f34f __CFRunLoopRun + 1328
4   com.apple.CoreFoundation      	0x00007fff2079e75c CFRunLoopRunSpecific + 563
5   com.apple.AppKit              	0x00007fff23120baa _NSEventThread + 124
6   libsystem_pthread.dylib       	0x00007fff206a7954 _pthread_start + 224
7   libsystem_pthread.dylib       	0x00007fff206a34a7 thread_start + 15

Thread 0 crashed with X86 Thread State (64-bit):
[... 381 lines cut ...]
CoreGraphics                        12K        2 
Foundation                          16K        1 
Kernel Alloc Once                    8K        1 
MALLOC                           240.5M       65 
MALLOC guard page                   32K        8 
MALLOC_NANO (reserved)           384.0M        1         reserved VM address space (unallocated)
STACK GUARD                       56.0M        6 
Stack                             10.5M        6 
VM_ALLOCATE                        368K       26 
__DATA                            27.7M      324 
__DATA_CONST                      16.2M      187 
__DATA_DIRTY                       657K      112 
__FONT_DATA                          4K        1 
__GLSLBUILTINS                    5176K        1 
__LINKEDIT                       520.5M        8 
__OBJC_RO                         70.2M        1 
__OBJC_RW                         2480K        2 
__TEXT                           316.8M      326 
__UNICODE                          588K        1 
mapped file                      225.8M       18 
shared memory                     2928K       16 
===========                     =======  ======= 
TOTAL                              1.9G     1263 
TOTAL, minus reserved VM space     1.5G     1263 
```

supermerill, 2021-05-01:
ok, here's my theory:
- by adding a bool to the ConfigOption object, we grow the amount of byte it takes.
- somewhere in the software, something is writing something where it shouldn't.
- previously, it doesn't do anything because nothing important was at this location
- now there is printer_technology at this location and it switches value randomly because of that.

So now I have to search for the commit/lines that silently&randomely do bad stuff. And it's almost impossible to find.

I just find a possible thread-unsafe code block in config init. I'm going to fix that and submit an other build, in 2-3 hours.

supermerill, 2021-05-02:
ok, can someone run the build 246?

also, @akospalfi, how did you manage to create a debug build? on the gihub macos10.15 VM, something is wrong and it stop with `symbol(s) not found for architecture x86_64` and it's about `wxOnAssert`. Did you encounter that? If not, what are the sdk version you compile with? the command lines?

akospalfi, 2021-05-02:
>  Did you encounter that? If not, what are the sdk version you compile with? the command lines?

Yes, I saw the same problem when I built in debug mode and it turned out that these asserts are turned off by an ifdef in wxwidgets by default.
I can't copy the exact lines right now (I only have my laptop with me right now), but I invoked cmake for deps with this flag:

```
cmake .. -DCMAKE_BUILD_TYPE=Debug
```

and for the app itself:

```
cmake .. -DCMAKE_BUILD_TYPE=Debug -DCMAKE_PREFIX_PATH="$PWD/../deps/build/destdir/usr/local" -DSLIC3R_STATIC=1
```

And I had to apply this patch to the deps to fix the linker error that you saw:
```diff
diff --git a/deps/wxWidgets/wxWidgets.cmake b/deps/wxWidgets/wxWidgets.cmake
index bf23698cf..18ecd5d78 100644
--- a/deps/wxWidgets/wxWidgets.cmake
+++ b/deps/wxWidgets/wxWidgets.cmake
@@ -26,7 +26,7 @@ prusaslicer_add_cmake_project(wxWidgets
         -DwxBUILD_PRECOMP=ON
         ${_wx_toolkit}
         "-DCMAKE_DEBUG_POSTFIX:STRING="
-        -DwxBUILD_DEBUG_LEVEL=0
+        -DwxBUILD_DEBUG_LEVEL=1
         -DwxUSE_DETECT_SM=OFF
         -DwxUSE_UNICODE=ON
         -DwxUSE_OPENGL=ON
```

The rest of the build process was the same that the CI action does.

I use macOS 11.2.3 and Xcode 12.5 with Apple clang version 12.0.5 (clang-1205.0.22.9).

akospalfi, 2021-05-02:
> ok, here's my theory:
>
> * by adding a bool to the ConfigOption object, we grow the amount of byte it takes.
> * somewhere in the software, something is writing something where it shouldn't.
> * [...]

I think I can confirm your theory. If I'm on d4f5d643f  and I only add the `bool phony` the the `ConfigOption` class then the app starts to crash, but interestingly if I change the `bool` type to `int` then the app doesn't crash anymore. Since it only crashes on macOS it could be something os or clang specific problem as well, maybe some weird aligment/padding problem?

supermerill, 2021-05-02:
thank you

and the build 246?

akospalfi, 2021-05-02:
> thank you
> 
> and the build 246?

I’ll test it in the evening and post the results.

lbibass, 2021-05-02:
The Debug build 6 fails at launch. 

[image]

supermerill, 2021-05-02:
This error is not related to this issue!
Maybe i can see the end of this long grind.

supermerill, 2021-05-02:
I've merged the bits from prusaslicer that will fix that last issue.

akospalfi, 2021-05-02:
@supermerill This is the run log of build 246:

```palfia@Akoss-iMac SuperSlicer % /Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer --loglevel 5
set enum 140672087737840 to monotonic: 0=>23
set enum 140672087720400 to concentric: 0=>6
set enum 140672087728512 to object: 0=>0
set enum 140672087808384 to stars: 0=>3
set enum 140672087828832 to sprinter: 0=>1
set enum 140672087839232 to connected: 0=>0
set enum 140672087841392 to connected: 0=>0
set enum 140672087842512 to connected: 0=>0
set enum 140672087840272 to connected: 0=>0
set enum 140672087843504 to autoenlarged: 0=>2
set enum 140672087853024 to top: 0=>0
set enum 140672087863088 to time_estimate_only: 0=>1
set enum 140672087894256 to none: 0=>0
set enum 140672087760400 to rear: 0=>3
set enum 140672087920272 to nearest: 0=>1
set enum 140672087737856 to rectilineargapfill: 0=>22
set enum 140672087953248 to plane: 0=>1
set enum 140672087962112 to rectilinear: 0=>0
set enum 140672087962096 to rectilinear: 0=>0
set enum 140672087737904 to monotonic: 0=>23
set enum 140672087993792 to linear: 0=>0
set enum 140672088021312 to portrait: 0=>1
set enum 140672088064368 to mCWS: 0=>1
set enum 140672087668096 to FFF: 0=>1
set enum 140672088042672 to dynamic: 0=>2
create default config @140672107662848 with option printer_technology @140672107662856 value 1
[2021-05-02 22:28:03.374037] [0x0000000111ecde00] [debug]   full path: 15840400373692917632
[2021-05-02 22:28:03.374277] [0x0000000111ecde00] [debug]   single instance: undefined. other params: /Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer;--loglevel;5
[2021-05-02 22:28:03.374286] [0x0000000111ecde00] [debug]   full lock path: /Users/palfia/Library/Application Support/SuperSlicer/cache/15840400373692917632.lock
[2021-05-02 22:28:03.374337] [0x0000000111ecde00] [info]    instance check: Another instance not found or single-instance not set.
GUI_App Bundle::new
Config:140672099465328::apply_only printer_technology from @140672122361760 value 0 to nothing because it doesn't exist
Config:140672124636704::apply_only printer_technology from @140672122382368 value 0 to @140672122361760 value:0
create PresetBundle, does it has the printer tech? yes, value: 0 @140672122382368
create PresetBundle with default printer as 0
Config:140672099465504::apply_only printer_technology from @140672116250128 value 0 to @140672107662856 value:1
GUI_App:on_init_inner test point 1 : (- default FFF -) = 0
GUI_App:on_init_inner test point 2 : (- default FFF -) = 0
GUI_App:on_init_inner test point 4 : (- default FFF -) = 0
[2021-05-02 22:28:03.521554] [0x0000000111ecde00] [info]    Checking if indices need to be installed from resources...
GUI_App:on_init_inner test point 6 : (- default FFF -) = 0
GUI_App:on_init_inner test point 7 : (- default FFF -) = 0
[2021-05-02 22:28:03.524804] [0x0000000111ecde00] [trace]   Best translation language detected (may be different from user locales): en_GB
[2021-05-02 22:28:03.524826] [0x0000000111ecde00] [trace]   Switching wxLocales to en_GB
GUI_App:on_init_inner test point 8 : (- default FFF -) = 0
GUI_App:on_init_inner test point 8.1 : (- default FFF -) = 0
PresetBundle::load_presets test point 1 : (- default FFF -) = 0
PresetBundle::load_presets test point 2 : (- default FFF -) = 0
PresetBundle::load_presets test point 5 : (- default FFF -) = 0
PresetBundle::load_presets test point 6 : (- default FFF -) = 0
PresetBundle::load_presets test point 7 : (- default FFF -) = 0
PresetBundle::load_presets test point 8 : (- default FFF -) = 0
PresetBundle::load_presets test point 9 : (- default FFF -) = 0
PresetBundle::load_presets test point 10 : (- default FFF -) = 142
GUI_App:on_init_inner test point 9 : (- default FFF -) = 142
GUI_App:on_init_inner test point 10 : (- default FFF -) = 142
Config:140672116329344::apply_only printer_technology from @140672116332080 value 0 to nothing because it doesn't exist
set enum 140672087728512 to object: 0=>0
set enum 140672087828832 to sprinter: 0=>1
set enum 140672087863088 to time_estimate_only: 0=>1
set enum 140672087993792 to linear: 0=>0
set enum 140672087720400 to concentric: 0=>6
set enum 140672087920272 to nearest: 0=>1
set enum 140672087953248 to plane: 0=>1
set enum 140672087962112 to rectilinear: 0=>0
set enum 140672087962096 to rectilinear: 0=>0
set enum 140672087737840 to monotonic: 0=>23
[... 67 lines cut ...]
< Vary: Accept, Accept-Encoding, Accept, X-Requested-With
< ETag: W/"7d4f9c76fd89d1be81503b849b31b3144f98e2028b5f59c3e66ca7b5f9736081"
< X-GitHub-Media-Type: github.v3; format=json
< Link: <https://api.github.com/repositories/116708603/releases?page=2>; rel="next", <https://api.github.com/repositories/116708603/releases?page=2>; rel="last"
< Access-Control-Expose-Headers: ETag, Link, Location, Retry-After, X-GitHub-OTP, X-RateLimit-Limit, X-RateLimit-Remaining, X-RateLimit-Used, X-RateLimit-Reset, X-OAuth-Scopes, X-Accepted-OAuth-Scopes, X-Poll-Interval, X-GitHub-Media-Type, Deprecation, Sunset
< Access-Control-Allow-Origin: *
< Strict-Transport-Security: max-age=31536000; includeSubdomains; preload
< X-Frame-Options: deny
< X-Content-Type-Options: nosniff
< X-XSS-Protection: 0
< Referrer-Policy: origin-when-cross-origin, strict-origin-when-cross-origin
< Content-Security-Policy: default-src 'none'
< X-RateLimit-Limit: 60
< X-RateLimit-Remaining: 56
< X-RateLimit-Reset: 1619989587
< X-RateLimit-Used: 4
< Accept-Ranges: bytes
< Transfer-Encoding: chunked
< X-GitHub-Request-Id: CD98:863E:4A059A0:4B5D018:608F0B54
< 
* Connection #0 to host api.github.com left intact
[2021-05-02 22:28:04.566239] [0x000070000ed76000] [info]    Syncing configuration cache
```

The app loads without crash, but all the print settings and filament settings presets are missing (the combos are empty).

supermerill, 2021-05-02:
erf

I'm working on understanding more how the printer_tech and init of config works.
In the meantime, if @akospalfi can run the latest build.
@lbibass the latest build should have the fix for your assert bug

akospalfi, 2021-05-02:
@supermerill Build 252:

```
palfia@Akoss-iMac SuperSlicer % /Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer --loglevel 5 --datadir ~/Library/Application\ Support/SuperSlicer-2
set enum 140388765647392 to monotonic: 0=>23
set enum 140388765629952 to concentric: 0=>6
set enum 140388765638064 to object: 0=>0
set enum 140388765717936 to stars: 0=>3
set enum 140388765738384 to sprinter: 0=>1
set enum 140388765748784 to connected: 0=>0
set enum 140388765750944 to connected: 0=>0
set enum 140388765752064 to connected: 0=>0
set enum 140388765749824 to connected: 0=>0
set enum 140388765753056 to autoenlarged: 0=>2
set enum 140388765762576 to top: 0=>0
set enum 140388765772640 to time_estimate_only: 0=>1
set enum 140388765803808 to none: 0=>0
set enum 140388765669952 to rear: 0=>3
set enum 140388765829824 to nearest: 0=>1
set enum 140388765647408 to rectilineargapfill: 0=>22
set enum 140388765862800 to plane: 0=>1
set enum 140388765871664 to rectilinear: 0=>0
set enum 140388765871648 to rectilinear: 0=>0
set enum 140388765647456 to monotonic: 0=>23
set enum 140388765903344 to linear: 0=>0
set enum 140388765931808 to portrait: 0=>1
set enum 140388765974864 to mCWS: 0=>1
set enum 140388765577648 to FFF: 0=>1
set enum 140388765953168 to dynamic: 0=>2
create default config @140388782351360 with option printer_technology @140388782351368 value 1
[2021-05-03 01:14:14.454287] [0x000000011786ee00] [debug]   full path: 15840400373692917632
[2021-05-03 01:14:14.454529] [0x000000011786ee00] [debug]   single instance: undefined. other params: /Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer;--loglevel;5;--datadir;"/Users/palfia/Library/Application Support/SuperSlicer-2"
[2021-05-03 01:14:14.454538] [0x000000011786ee00] [debug]   full lock path: /Users/palfia/Library/Application Support/SuperSlicer-2/cache/15840400373692917632.lock
[2021-05-03 01:14:14.454587] [0x000000011786ee00] [info]    instance check: Another instance not found or single-instance not set.
GUI_App Bundle::new
Config:140388799230576::apply_only printer_technology from @140388754216304 value 0 to nothing because it doesn't exist
Config:140388799134240::apply_only printer_technology from @140388754236864 value 0 to @140388754216304 value:0
create PresetBundle, does it has the printer tech? yes, value: 0 @140388754236864
create PresetBundle with default printer as 0
Config:140388799230752::apply_only printer_technology from @140388754243440 value 0 to @140388782351368 value:1
end PresetBundle creation with default printer as 0
GUI_App:on_init_inner test point 1 : (- default FFF -) = 0
GUI_App:on_init_inner test point 2 : (- default FFF -) = 0
GUI_App:on_init_inner test point 4 : (- default FFF -) = 0
[2021-05-03 01:14:14.590251] [0x000000011786ee00] [info]    Checking if indices need to be installed from resources...
GUI_App:on_init_inner test point 6 : (- default FFF -) = 0
GUI_App:on_init_inner test point 7 : (- default FFF -) = 0
[2021-05-03 01:14:14.593505] [0x000000011786ee00] [trace]   Best translation language detected (may be different from user locales): en_GB
[2021-05-03 01:14:14.593525] [0x000000011786ee00] [trace]   Switching wxLocales to en_GB
GUI_App:on_init_inner test point 8 : (- default FFF -) = 0
GUI_App:on_init_inner test point 8.1 : (- default FFF -) = 0
PresetBundle::load_presets test point 1 : (- default FFF -) = 0
Config::set_deserialize_nothrow printer_technology from 'unfound option'  to FFF == FFF
Config::set_deserialize printer_technology from value 0 to FFF
Config:140732820422320::apply_only printer_technology from @140388848856944 value 0 to @140388848841648 value:0
Config::set_deserialize_nothrow printer_technology from 'unfound option'  to FFF == FFF
Config::set_deserialize printer_technology from value 0 to FFF
Config:140732820422320::apply_only printer_technology from @140388848870592 value 0 to @140388848841632 value:0
Config::set_deserialize_nothrow printer_technology from 'unfound option'  to FFF == FFF
Config::set_deserialize printer_technology from value 156 to FFF
Config:140732820422320::apply_only printer_technology from @140388849022304 value 0 to @140388849007056 value:156
Config::set_deserialize_nothrow printer_technology from 'unfound option'  to FFF == FFF
Config::set_deserialize printer_technology from value 221 to FFF
Config:140732820422320::apply_only printer_technology from @140388849288704 value 0 to @140388849262896 value:221
Config::set_deserialize_nothrow printer_technology from 'unfound option'  to FFF == FFF
Config::set_deserialize printer_technology from value 218 to FFF
Config:140732820422320::apply_only printer_technology from @140388849302912 value 0 to @140388849274928 value:218
Config::set_deserialize_nothrow printer_technology from 'unfound option'  to FFF == FFF
Config::set_deserialize printer_technology from value 0 to FFF
Config:140732820422320::apply_only printer_technology from @140388757055936 value 0 to @140388757040720 value:0
[... 515 lines cut ...]
* Connected to files.prusa3d.com (185.115.1.124) port 443 (#0)
* TLS 1.2 connection using TLS_ECDHE_RSA_WITH_AES_256_GCM_SHA384
* Server certificate: files.prusa3d.com
* Server certificate: R3
* Server certificate: DST Root CA X3
> GET /wp-content/uploads/repository/PrusaSlicer-settings-master/live/Anycubic//index.idx HTTP/1.1
Host: files.prusa3d.com
User-Agent: SuperSlicer/2.3.56
Accept: */*

< HTTP/1.1 200 OK
< Server: nginx
< Date: Sun, 02 May 2021 23:14:15 GMT
< Content-Length: 628
< Connection: keep-alive
< Last-Modified: Sun, 02 May 2021 23:00:45 GMT
< ETag: "274-5c160d0ae0b02"
< Accept-Ranges: bytes
< 
* Connection #0 to host files.prusa3d.com left intact
[2021-05-03 01:14:15.691730] [0x000070000d82c000] [debug]   Got index for vendor: Anycubic: current version: 0.0.9, recommended version: 0.0.9
[2021-05-03 01:14:15.691743] [0x000070000d82c000] [warning] No such vendor: Creality
```

No crash and missing print&filament presets.

supermerill, 2021-05-03:
@akospalfi 
it seems like your test with build 246 it without any preset (empty setting directory)
and your run with build 252 has a 'Anycubic Kossel Linear Plus' inside it.
Is it because your setting directory see its content change? Or it is because of my changes (I only add logs, so I don't think sp).
It's very difficult to debug something that isn't consistent.
note that you should be able to run superslicer on a local setting path with the option `--datadir`
example: `superslicer --datadir my_local_setting_directory`

supermerill, 2021-05-03:
hi
can someone test the #253 build and post the output. Ideally, without (and with if you have the time) an already populated setting directory.

the build 'debug macos #10' is here to launch in case of a crash

Current investigation state, after dozen and dozen of hours lost in that issue: the software doesn't get the right value from the memory. And I don't know why.

for reference, here is a good log:
<details>
 <summary>Summary</summary>

```
After creation, the default FullPrintConfig:2274924233248 has no printer tech
set enum 2274884243504 to monotonic: 0=>23
set enum 2274845048160 to concentric: 0=>6
set enum 2274844904784 to object: 0=>0
set enum 2274884509552 to stars: 0=>3
set enum 2274884510320 to sprinter: 0=>1
set enum 2274884591040 to connected: 0=>0
set enum 2274884589888 to connected: 0=>0
set enum 2274884587680 to connected: 0=>0
set enum 2274884588832 to connected: 0=>0
set enum 2274884585184 to autoenlarged: 0=>2
set enum 2274884592960 to top: 0=>0
set enum 2274884593056 to time_estimate_only: 0=>1
set enum 2274884815312 to none: 0=>0
set enum 2274884341248 to rear: 0=>3
set enum 2274884829328 to nearest: 0=>1
set enum 2274884245424 to rectilineargapfill: 0=>22
set enum 2274885000336 to plane: 0=>1
set enum 2274884999760 to rectilinear: 0=>0
set enum 2274884998704 to rectilinear: 0=>0
set enum 2274884243792 to monotonic: 0=>23
set enum 2274885163904 to linear: 0=>0
try to infer printer_technology
Don't find anything, return unknown
set enum 2274885291040 to portrait: 0=>1
set enum 2274885313120 to mCWS: 0=>1
set enum 2274844995760 to FFF: 0=>1
set enum 2274885296512 to dynamic: 0=>2
create default config @2274923936832 with option printer_technology @2274923936848 value 1
GUI_App::PresetBundle::new
ConfigOptionDef::create_default_option printer_technology
ConfigOptionDef::create_default_option it extists a printer_technology
ConfigOptionDef::create_default_option create a defautl with value 1
Config:2274924724040::apply_only printer_technology for the first time to default @2274925289968 value 1 to nothing because it doesn't exist
ConfigOptionDef::create_default_option printer_technology
ConfigOptionDef::create_default_option it extists a printer_technology
ConfigOptionDef::create_default_option create a defautl with value 1
Config:2274925146216::apply_only printer_technology for the first time to default @2274925309648 value 1 to value:1by conf @-912318280 : 2274925289968
create PresetBundle, does it has the printer tech? yes, value: 1 @2274925309648
create PresetBundle with default printer as 1
ConfigOptionDef::create_default_option printer_technology
ConfigOptionDef::create_default_option it extists a printer_technology
ConfigOptionDef::create_default_option create a defautl with value 1
Config:2274924721608::apply_only printer_technology for the first time to default @2274925323184 value 1 to value:1by conf @-912318456 : 2274923936848
set &this->printers.preset(1).printer_technology_ref() to SLA (2
PresetCollection::select_preset  => - default -
PresetCollection::select_preset  => - default -
PresetCollection::select_preset  => - default -
PresetCollection::select_preset  => - default -
PresetCollection::select_preset  => - default FFF -
end PresetBundle creation with default printer as 1
GUI_App:on_init_inner test point 1 : (- default FFF -) = 1
GUI_App:on_init_inner test point 2 : (- default FFF -) = 1
GUI_App:on_init_inner test point 3 : (- default FFF -) = 1
GUI_App:on_init_inner test point 4 : (- default FFF -) = 1
GUI_App:on_init_inner test point 6 : (- default FFF -) = 1
GUI_App:on_init_inner test point 7 : (- default FFF -) = 1
GUI_App:on_init_inner test point 8 : (- default FFF -) = 1
[... 966 lines cut ...]
set enum 2274884245424 to rectilineargapfill: 0=>22
set enum 2274884243792 to monotonic: 0=>23
set enum 2274885291040 to portrait: 0=>1
set enum 2274885313120 to mCWS: 0=>1
set enum 2274844995760 to FFF: 0=>1
create default config @2274925584880 with option printer_technology @2274925584896 value 1
set enum 2274885296512 to dynamic: 0=>2
 set fff print to background_process, this print has the config -1349637960
BackgroundSlicingProcess:select_technology(1)
ConfigOptionDef::create_default_option printer_technology
ConfigOptionDef::create_default_option it extists a printer_technology
ConfigOptionDef::create_default_option create a defautl with value 1
Config:978340224144::apply_only printer_technology for the first time to default @2274927444416 value 1 to value:1by conf @-912320104 : 2274962171648
GUI_App:on_init_inner test point 11 : (Original Prusa i3 MK2S MMU1) = 1
GUI_App:on_init_inner test point 12 : (Original Prusa i3 MK2S MMU1) = 1
GUI_App:on_init_inner test point 13 : (Original Prusa i3 MK2S MMU1) = 1
GUI_App:on_init_inner test point 14 : (Original Prusa i3 MK2S MMU1) = 1
GUI_App:load_current_presets start : (Original Prusa i3 MK2S MMU1) = 1
GUI_App:load_current_presets check_printer_presets : (Original Prusa i3 MK2S MMU1) = 1
GUI_App:load_current_presets check_printer_presets : (Original Prusa i3 MK2S MMU1) = 1
BackgroundSlicingProcess:select_technology(1)
GUI_APP::Register wxEVT_IDLE

```
</details>

akospalfi, 2021-05-03:
> @akospalfi
> it seems like your test with build 246 it without any preset (empty setting directory)
> and your run with build 252 has a 'Anycubic Kossel Linear Plus' inside it.
> Is it because your [...]

Yeah sorry about that, when I was running build 246 I forgot to add the --datadir setting and it was running with an empty setting directory (you can see in my logs, I've included the cmd line params).

akospalfi, 2021-05-03:
Ok some more investigation work on my side:
Build 256 failed, but I've compiled it locally after fixing the casting problems.

<details><summary>Run log of build 256:</summary><p>

```
palfia@Akoss-iMac build % lldb src/superslicer
(lldb) target create "src/superslicer"
Current executable set to '/Volumes/WORK/myprojects/SuperSlicer/SuperSlicer/build/src/superslicer' (x86_64).
(lldb) process launch -- --loglevel 5 --datadir "/Users/palfia/Library/Application Support/SuperSlicer-2"
Process 25054 launched: '/Volumes/WORK/myprojects/SuperSlicer/SuperSlicer/build/src/superslicer' (x86_64)
After creation, the default FullPrintConfig:4613854208 has no printer tech 
set enum 4589736512 to monotonic: 0=>23
set enum 4589719072 to concentric: 0=>6
set enum 4589727184 to object: 0=>0
set enum 4589807056 to stars: 0=>3
set enum 4589827504 to sprinter: 0=>1
set enum 4589837904 to connected: 0=>0
set enum 4589840064 to connected: 0=>0
set enum 4589841184 to connected: 0=>0
set enum 4589838944 to connected: 0=>0
set enum 4589842176 to autoenlarged: 0=>2
set enum 4589851696 to top: 0=>0
set enum 4590683872 to time_estimate_only: 0=>1
set enum 4590715184 to none: 0=>0
set enum 4589759072 to rear: 0=>3
set enum 4590741200 to nearest: 0=>1
set enum 4589736528 to rectilineargapfill: 0=>22
set enum 4590774176 to plane: 0=>1
set enum 4590783040 to rectilinear: 0=>0
set enum 4590783024 to rectilinear: 0=>0
set enum 4589736576 to monotonic: 0=>23
set enum 4590814720 to linear: 0=>0
try to infer printer_technology
Don't find anything, return unknown
set enum 4590852416 to portrait: 0=>1
set enum 4590895472 to mCWS: 0=>1
set enum 4589666960 to FFF: 0=>1
set enum 4590873776 to dynamic: 0=>2
create default config @4613884928 with option printer_technology @4613884936 value 1
[2021-05-03 13:50:56.687979] [0x00000001116fbe00] [debug]   full path: 16849348648804176793
[2021-05-03 13:50:56.688254] [0x00000001116fbe00] [debug]   single instance: undefined. other params: /Volumes/WORK/myprojects/SuperSlicer/SuperSlicer/build/src/superslicer;--loglevel;5;--datadir;"/Users/palfia/Library/Application Support/SuperSlicer-2"
[2021-05-03 13:50:56.688288] [0x00000001116fbe00] [debug]   full lock path: /Users/palfia/Library/Application Support/SuperSlicer-2/cache/16849348648804176793.lock
[2021-05-03 13:50:56.688388] [0x00000001116fbe00] [info]    instance check: Another instance not found or single-instance not set.
2021-05-03 13:50:56.693590+0200 superslicer[25054:206303] SecTaskLoadEntitlements failed error=22 cs_flags=20, pid=25054
2021-05-03 13:50:56.693648+0200 superslicer[25054:206303] SecTaskCopyDebugDescription: superslicer[25054]/0#-1 LF=0
GUI_App::PresetBundle::new
ConfigOptionDef::create_default_option printer_technology
ConfigOptionDef::create_default_option it extists a printer_technology
ConfigOptionDef::create_default_option create a defautl with value 1 
Config:4622869616::apply_only printer_technology for the first time to default @4592238576 value 0 to nothing because it doesn't exist
ConfigOptionDef::create_default_option printer_technology
ConfigOptionDef::create_default_option it extists a printer_technology
ConfigOptionDef::create_default_option create a defautl with value 1 
Config:4622838816::apply_only printer_technology for the first time to default @4592259184 value 0 to value:0by conf @140732920714792 : 4592238576
create PresetBundle, does it has the printer tech? yes, value: 0 @4592259184
create PresetBundle with default printer as 0
ConfigOptionDef::create_default_option printer_technology
ConfigOptionDef::create_default_option it extists a printer_technology
ConfigOptionDef::create_default_option create a defautl with value 1 
Config:4622869792::apply_only printer_technology for the first time to default @4592265760 value 0 to value:1by conf @140732920714904 : 4613884936
set &this->printers.preset(1).printer_technology_ref() to SLA (2
PresetCollection::select_preset  => - default -
PresetCollection::select_preset  => - default -
PresetCollection::select_preset  => - default -
PresetCollection::select_preset  => - default -
PresetCollection::select_preset  => - default FFF -
end PresetBundle creation with default printer as 0
GUI_App:on_init_inner test point 1 : (- default FFF -) = 0
GUI_App:on_init_inner test point 2 : (- default FFF -) = 0
[... 2450 lines cut ...]
* Connected to files.prusa3d.com (185.115.1.124) port 443 (#0)
* TLS 1.2 connection using TLS_ECDHE_RSA_WITH_AES_256_GCM_SHA384
* Server certificate: files.prusa3d.com
* Server certificate: R3
* Server certificate: DST Root CA X3
> GET /wp-content/uploads/repository/PrusaSlicer-settings-master/live/Anycubic//index.idx HTTP/1.1
Host: files.prusa3d.com
User-Agent: SuperSlicer/2.3.56
Accept: */*

< HTTP/1.1 200 OK
< Server: nginx
< Date: Mon, 03 May 2021 11:59:34 GMT
< Content-Length: 628
< Connection: keep-alive
< Last-Modified: Mon, 03 May 2021 11:00:13 GMT
< ETag: "274-5c16addb38f92"
< Accept-Ranges: bytes
< 
* Connection #0 to host files.prusa3d.com left intact
[2021-05-03 13:59:33.953489] [0x00007000088a7000] [debug]   Got index for vendor: Anycubic: current version: 0.0.9, recommended version: 0.0.9
[2021-05-03 13:59:33.953505] [0x00007000088a7000] [warning] No such vendor: Creality

```
</p></details>

supermerill, 2021-05-03:
some last though:
 * on windows, the phony bool use 8byte (a int32) in the 64b build. And it may be the same on other platform, so moving it from boo to unit32 changes nothing
 * I'm going to output a last log build to try a last time to pinpoint & understand the macos problem. -> can someone publish a log from the 'Nigthly macos 25' ?
 * I'm moving to uint32 and make a release with that.

akospalfi, 2021-05-04:
> some last though:
> 
> * on windows, the phony bool use 8 bytes (a int32) in the 64b build anyway, for alignment reason I suppose (like in java). And it may be the same on other platform, so moving it from bool to unit32 changes nothing

Yes, the situation is the same on macOS, according to the debugger the ConfigOption class takes 16 bytes overall no matter if I use bool or uint32_t.

> * I'm going to output a last log build to try a last time to pinpoint & understand the macos problem. -> can someone publish a log from the 'Nigthly macos 25' ?

Attached below.

> * I'm moving to uint32 and make a release with that.

Sounds like a plan. :)

<details><summary>Run log of Nigthly macos #259:</summary><p>

```
palfia@Akoss-iMac build % /Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer --loglevel 5 --datadir "/Users/palfia/Library/Application Support/SuperSlicer-2"
After creation, the default FullPrintConfig:140415827275264 has no printer tech 
set enum 140415801041440 to monotonic: 0=>23
set enum 140415801024000 to concentric: 0=>6
set enum 140415801032112 to object: 0=>0
set enum 140415801111984 to stars: 0=>3
set enum 140415801132432 to sprinter: 0=>1
set enum 140415801142832 to connected: 0=>0
set enum 140415801144992 to connected: 0=>0
set enum 140415801146112 to connected: 0=>0
set enum 140415801143872 to connected: 0=>0
set enum 140415801147104 to autoenlarged: 0=>2
set enum 140415801156624 to top: 0=>0
set enum 140415801166688 to time_estimate_only: 0=>1
set enum 140415801197856 to none: 0=>0
set enum 140415801064000 to rear: 0=>3
set enum 140415801223872 to nearest: 0=>1
set enum 140415801041456 to rectilineargapfill: 0=>22
set enum 140415801256848 to plane: 0=>1
set enum 140415801265712 to rectilinear: 0=>0
set enum 140415801265696 to rectilinear: 0=>0
set enum 140415801041504 to monotonic: 0=>23
set enum 140415801297392 to linear: 0=>0
try to infer printer_technology
Don't find anything, return unknown
set enum 140415801335088 to portrait: 0=>1
set enum 140415801378144 to mCWS: 0=>1
set enum 140415800984832 to FFF: 0=>1
set enum 140415801356448 to dynamic: 0=>2
create default config @140415827305984 with option printer_technology @140415827305992 value 1
[2021-05-04 12:03:15.897365] [0x000000011d124e00] [debug]   full path: 15840400373692917632
[2021-05-04 12:03:15.897607] [0x000000011d124e00] [debug]   single instance: undefined. other params: /Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer;--loglevel;5;--datadir;"/Users/palfia/Library/Application Support/SuperSlicer-2"
[2021-05-04 12:03:15.897616] [0x000000011d124e00] [debug]   full lock path: /Users/palfia/Library/Application Support/SuperSlicer-2/cache/15840400373692917632.lock
[2021-05-04 12:03:15.897667] [0x000000011d124e00] [info]    instance check: Another instance not found or single-instance not set.
GUI_App::PresetBundle::new
ConfigOptionDef::create_default_option printer_technology
ConfigOptionDef::create_default_option it extists a printer_technology
ConfigOptionDef::create_default_option create a defautl with value 1 
ConfigOptionDef::create_default_option @140415799299104 => 1 
ref: phony140415799299112, value: => 140415799299116, keymapptr: 140415799299120 
ConfigOptionDef::create_default_option set it again:1 
DynamicConfig::optptr get ptr 140415799299104
DynamicConfig::optptr ptr after move into map 140415799299104 -> 1
Config:140415810797168::apply_only printer_technology for the first time to default @140415799299104==@140415799299104 value 0 to nothing because it doesn't exist
ConfigOptionDef::create_default_option printer_technology
ConfigOptionDef::create_default_option it extists a printer_technology
ConfigOptionDef::create_default_option create a defautl with value 1 
ConfigOptionDef::create_default_option @140415799319712 => 1 
ref: phony140415799319720, value: => 140415799319724, keymapptr: 140415799319728 
ConfigOptionDef::create_default_option set it again:1 
DynamicConfig::optptr get ptr 140415799319712
DynamicConfig::optptr ptr after move into map 140415799319712 -> 1
Config:140415818960928::apply_only printer_technology for the first time to default @140415799319712==@140415799319712 value 0 to value:0by conf @140732653054472 : 140415799299104
[... 1894 lines cut ...]
* TLS 1.2 connection using TLS_ECDHE_RSA_WITH_AES_256_GCM_SHA384
* Server certificate: files.prusa3d.com
* Server certificate: R3
* Server certificate: DST Root CA X3
> GET /wp-content/uploads/repository/PrusaSlicer-settings-master/live/Anycubic//index.idx HTTP/1.1
Host: files.prusa3d.com
User-Agent: SuperSlicer/2.3.56
Accept: */*

< HTTP/1.1 200 OK
< Server: nginx
< Date: Tue, 04 May 2021 10:03:17 GMT
< Content-Length: 628
< Connection: keep-alive
< Last-Modified: Tue, 04 May 2021 10:00:37 GMT
< ETag: "274-5c17e265ae796"
< Accept-Ranges: bytes
< 
* Connection #0 to host files.prusa3d.com left intact
[2021-05-04 12:03:17.156288] [0x0000700007db6000] [debug]   Got index for vendor: Anycubic: current version: 0.0.9, recommended version: 0.0.9
[2021-05-04 12:03:17.156302] [0x0000700007db6000] [warning] No such vendor: Creality

```

</p></details>

lbibass, 2021-05-08:
Any updates on this? It's still broken unfortunately.

supermerill, 2021-05-20:
didn't had the time to push a not-buggy code before my departure. I'm back now. will push it shortly

akospalfi, 2021-05-20:
> didn't had the time to push a not-buggy code before my departure. I'm back now. will push it shortly

nice to hear that you are back :)

just an FYI, I've been using my own build with the uint32_t "fix" for about a week now and I didn't notice any problem with it.

mvieleers, 2021-05-23:
Just wanted to let you know that the new version 2.3.56.0 seems to be working fine on Mac OS.

closed 2021-05-27 by supermerill