supermerill/SuperSlicer #1113: Using 2nd extruder -> hotbed settings beeing ignored

Phil1988, 2021-04-27:
**Describe the bug**
I have the problem, that when I set the 2nd extruder to be the default extruder like:
[image]

Or if I choose the 2nd extruder from the dropdown like:
[image]

**The bed does not heat up.**

The temperature setting at the filament looks like this:
[image]

I have this in the starting GCODE section:
M140 S[first_layer_bed_temperature] ; set bed temp

And this is what the generated GCODE looks like:
M140 S0 ; set bed temp

The project is attached as a zip file:
[2nd_extruder_coldBed.zip](https://github.com/supermerill/SuperSlicer/files/6383717/2nd_extruder_coldBed.zip)

SuperSlicer: Version 2.3.55
OS: Windows 10 x64

I hope you can help me out :)

supermerill, 2021-04-27:
Does it takes the temp of the first extruder instead of the second?

Phil1988, 2021-04-27:
It just sets it to 0.

In this profile, there is just the Temperature set for the nozzle and for a bed.
There is not a "bed temperature" for different extruders - at least not that I can see.

Am I doing something wrong?

Phil1988, 2021-05-05:
The problem is not existent on my 2nd notebook.
I will investigate further and let you know my findings :)

supermerill, 2021-06-11:
I found the problem. The bed temp is set to 0 for the second filament in the project file.
But because it's the same filament profile, it doesn't show on the gui. If you wap out & back the filament of the second extruder, everything goes back to normal.

I can reproduce the issue: If I go to the second filament profile by clicking on the cog, and change the first layer temp of it, then save the project. It will save with only the second filament first layer temp modified, even if you can't see that the first one keep its old value.