supermerill/SuperSlicer #1114: Thin walls on tiny nozzle causes printer to pause for a few seconds

CreeperShift, 2021-04-27:
**Describe the bug**
I've installed a 0.15mm nozzle and have been doing test prints to dial in settings. When using a very small layer height, in this case I'm printing at 0.04mm, once it tries printing the "thin wall" features as seen in the slicer preview, the printer will pause for 2-3 seconds completely ruining the section. It looks like it completes the thin walls a few layers, then the pause starts and then it continues again. Easily observable on the smokestack of the benchy, where it printed 3-4 layers, paused, printer another set of layers and paused again. 

In this case I have been using a benchy scaled down to 10% and there the issue happens as described. Once I scale it up to 12.5%, it no longer has thin walls and prints without the pause with identical settings.

**To Reproduce**
I have attached the Config files, the .3mf file and the gcode where I observed the issue the first time (some settings might have been changed since then, but the problem still exists)
[Tiny-Print.zip](https://github.com/supermerill/SuperSlicer/files/6385164/Tiny-Print.zip)

**Desktop (please complete the following information):**
Version: 2.3.55.5
Build: SuperSlicer_2.3.55.5
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build 19042), 64-bit edition
Total RAM size [MB]: 34,309MB

**Additional context**
Printer is a highly modified Ender-3 Pro running latest Klipper firmware.

supermerill, 2021-04-27:
Seems like your firmware had a hard time to cope with many very small extrusions.
You have to use the 'min length' from the printer->general section. But this setting is buggy in the current beta release, you have to use a nightly.

CreeperShift, 2021-04-27:
> Seems like your firmware has a hard time to cope with many very small extrusions.
> You have to use the 'min length' from the printer->general section. But this setting is buggy in the current beta release, you have to use a nightly (action menu in the to bar)

Thanks for the answer, I will give the nightly a try tomorrow and see if the setting helps. The only thing that surprises me is the fact that increasing the benchy size by just 2% fixes all of those issues. Does the thin walls setting do that many more extrusions compared to normal?

supermerill, 2021-04-28:
i made a chamfer on the tip of thin walls. on a 0.4 nozzle, it means ~10 sections of ~0.015mm, but just at the tip, to have a nice visual on the gcode viewer.

CreeperShift, 2021-04-28:
Okay thank you for pointing me into the right direction! It seems I was not printing off the virtual SD card with Klipper, causing the Stream from Octoprint -> Klipper to be overwhelmed. I've also noticed (from reading a few issues here) that "only 1 perimeter on top surfaces" can be buggy so I turned it off and now the top of the model looks a lot more clean as well. Currently reprinting a very detailed tiny model with thin walls enabled and it seems to work perfectly, so thank you this seems to be resolved! 

> i made a chamfer on the tip of thin walls. on a 0.4 nozzle, it means ~10 sections of ~0.015mm, but just at the tip, to have a nice visual on the gcode viewer.
Wow that's a lot, that would explain why octoprint couldn't stream it fast enough.

closed 2021-04-28 by CreeperShift