supermerill/SuperSlicer #1115: superslicer_console.exe only outputs 0 byte tmp file (prusa-slicer-console.exe works fine)

jerryfromearth, 2021-04-27:
**Describe the bug**
`superslicer\superslicer_console.exe -g square.stl -o super.gcode --loglevel 4` only generates a 0 byte ".gcode.tmp" file and no gcode file.

**To Reproduce**
Steps to reproduce the behavior:
Run this command:
`superslicer\superslicer_console.exe -g square.stl -o super.gcode --loglevel 4`
An empty "square.gcode.tmp" file is seen. No gcode file is seen.

However, 
`prusa3d\prusaslicer\prusa-slicer-console.exe -g square.stl -o prusa.gcode --loglevel 4` generates a proper gcode file. (PrusaSlicer version 2.3.1)

**3D Model and Slic3r Configuration Export**
[square.zip](https://github.com/supermerill/SuperSlicer/files/6385541/square.zip)
It happens with empty configuration as well.

**Expected behavior**
After running the command, there should be a functioning "square.gcode" file.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version 2.3.55.5

**Additional context**
[output.txt](https://github.com/supermerill/SuperSlicer/files/6385563/output.txt)

jerryfromearth, 2021-04-27:
Update:
superslicer_console.exe works fine, if I export settings via GUI, and then append "--load config.ini" in the command line.

Please close this issue if this is intended behavior. :)

supermerill, 2021-04-27:
I don't think it is.

closed 2021-06-04 by supermerill