supermerill/SuperSlicer #1116: Fiilament extrusion override is casted to integer

sttts, 2021-04-28:
**Describe the bug**

When adding an override for an object for the filament extrusion factor, every floating point value is casted to integer. When I enter 99.5, it becomes 99.

A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. add override: filament / extrusion
2. enter 99.5
3. change focus and come back
4. find 99

**Expected behavior**

99.5 to stay

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - MacOS 10.15
 - SuperSlicer from CI snapshot 2ish weeks ago

Pavulon87, 2021-04-28:
I can confirm it in the last version(windows).

Pavulon87, 2021-04-28:
This is the case for all settings of type coPercent.
For example fill_density or bridge_overlap.

Pavulon87, 2021-04-28:
In src\slic3r\GUI\ **OptionsGroup.cpp** there is:
> 	case coPercent:{
		double val = config.option<ConfigOptionPercent>(opt_key)->value;
		text_value = wxString::Format(_T("%i"), int(val));
		ret = text_value;// += "%";
	}

@supermerill Maybe a good option would be to change the format to:
>text_value = wxString::Format(_T("%g"), val);

supermerill, 2021-04-28:
@Pavulon87
got it
I prefer %f than %g for this case.
It also needs to not display 99.0000000005%

closed 2021-06-04 by supermerill