supermerill/SuperSlicer #1120: Coasting with Extra Wipe for external perimeters with negative value

macchinette, 2021-04-28:
** Is your feature request related to an issue? Please describe. **
Coasting to reduce imperfections of the Z-seam.

** Describe the solution you want **
The Extra Wipe for external perimeters function accepts only positive values, but with negative values the extruder should stop in advance of the end of the perimeter and do like the Coasting function present in other slicer software.

** Describe the alternatives you have considered **
None

** Additional context **
None

supermerill, 2021-04-28:
duplicate of #517

macchinette, 2021-04-29:
Sorry I had not seen that this request had already been talked about.
I'm not a software engineer so I don't understand much the difficulties that there may be, but I thought it would be easier to modify the "Extra Wipe for external perimeters" with negative values to stop the extrusion before the end of the path even in a brutal way without the two other parameters of volume and speed.

supermerill, 2021-04-29:
"Extra Wipe for external perimeters" add move, you want to remove some, entirely different though.

closed 2021-06-04 by supermerill