supermerill/SuperSlicer #1121: Machine limits in the G-code in wrong position.

macchinette, 2021-04-28:
** Describe the bug **
Machine limits in the G-code in wrong position.

** To play **
Steps to reproduce the behavior:
1. Go to "Printer Setting"
2. Click on "Machine Limits"
3. Enable "Also Emit Limits to G-code"
4. Export G-code

** Expected behavior **
Machine limits are written to the G-code after the extrusion width data.

**Screenshots**
[image]

** Desktop (complete the following information): **
  - OS: Windows
  - Version 2.3.56.0

** Additional context **
This bug is a problem for importing the settings profile from the G-code and for the preview of the G-code not processing the data correctly.

supermerill, 2021-04-29:
Can't reproduce. It's placed in the right place.
maybe with your project file?

macchinette, 2021-04-29:
In the meantime, thanks for the effort you put into developing the software, it's magnificent.
I have attached the project if it can be useful to you, the problem occurs to me from version 2.3 onwards.
Thanks so much 
[Coupler_Centrator_AI3M.zip](https://github.com/supermerill/SuperSlicer/files/6396546/Coupler_Centrator_AI3M.zip)

macchinette, 2021-04-29:
> Non riesco a riprodursi. È posto nel posto giusto.
> magari con il file di progetto?

I noticed that with just one of these two parameters fan_speedup_time and fan_kickstart different from 0 value it makes the mistake I described above.
I hope I was helpful.
Thank you

macchinette, 2021-04-29:
I also noticed that the G-code is exported with the wrong time indication two hours less than the actual export time.

supermerill, 2021-04-29:
It's exported at utc tiemzone.

closed 2021-06-04 by supermerill