supermerill/SuperSlicer #1122: italian translation doesn't work

Liger0, 2021-04-29:
### Do not report Prusa3D Slic3r bugs here without confirming it is a problem on a development release of Slic3r, or your issue will be closed. [Development builds of Slic3r](https://dl.slic3r.org/dev). [Prusa3D Issue Tracker](https://github.com/Prusa3D/Slic3r/issues)

**Describe the bug**
Italian language is not present

**To Reproduce**
Steps to reproduce the behavior:

1. Clean SS installation
2. Configuration -> Language
3. Say yes to restart required
4. Click on italian
5. No restart.
6. If you manually restart nothing happens.

**3D Model and Slic3r Configuration Export**
Please upload a ZIP archive containing a copy of the 3D model you are seeing this bug with and a CTRL-E export of the configuration used.

**Expected behavior**
Restart and apply italian language

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows 10 x64
 - Version 2.3.56.0 beta (as of 28 april)

**Additional context**
Add any other context about the problem here.

boromyr, 2021-04-30:
They should be my translation files, which I updated but I have not loaded.
Did you compilate the 56 version?

Liger0, 2021-04-30:
I downloaded from the git actions page.

boromyr, 2021-04-30:
Not I was aware, you can share the link, so I try the translation in version 56 and translate the new entries.

Liger0, 2021-04-30:
https://github.com/supermerill/SuperSlicer/actions/workflows/ccpp_win.yml

boromyr, 2021-04-30:
Great thanks. With updated files the language changes correctly. Check if there are new items to be translated and loaded.

Liger0, 2021-04-30:
I will once Merill will push the commit. Thanks.

supermerill, 2021-04-30:
I'm working on an updated tool in python to get all strings to translate, even the ones that are in conf files.

Liger0, 2021-04-30:
Are the issues of crash when entering cooling section like #1004 being fixed too?

boromyr, 2021-04-30:
[Update](https://github.com/supermerill/SuperSlicer/issues/1128)

> I'm working on an updated tool in python to get all strings to translate, even the ones that are in conf files.

Would be great!!!!!

Liger0, 2021-04-30:
I noticed something that can be improved, for example there is no need to say Cost(€) since on the value there is writen €/kg again, and a lot of stuff are using infinitive mood rather than imperative.

I did change the SuperSlicer.po file but I am not sure how they work, as even changing the SuperSlicer.po file, the changes aren't displayed in the gui, and I am not sure why, maybe the SuperSlicer.po file isn't the one that is used to show the language?

@supermerill is there a documentation of what each file in the localization directory does?
I also don't understand something like why the & here is in the middle of the words.

`#: src/slic3r/GUI/MainFrame.cpp:1297
msgid "&Collapse sidebar"
msgstr "Ridu&ci barra laterale"

#: src/slic3r/GUI/GUI_App.cpp:1645
msgid "&Configuration"
msgstr "&Configurazione"

#: src/slic3r/GUI/GUI_App.cpp:1480
msgid "&Configuration Snapshots"
msgstr "Istantanee di &Configurazione"

#: src/slic3r/GUI/MainFrame.cpp:1194
msgid "&Copy"
msgstr "&Copia"

#: src/slic3r/GUI/MainFrame.cpp:1178
msgid "&Delete selected"
msgstr "Eli&mina selezionati"`

closed 2021-05-27 by Liger0