supermerill/SuperSlicer #1123: Feature request: Automatic orientation adjustment according to the axis preference

dingyifei, 2021-04-29:
### Do not request features for the Prusa3D fork of Slic3r here without confirming it is applicable to the mainline Slic3r, or your issue will be closed. [Development builds of Slic3r](https://dl.slic3r.org/dev). [Prusa3D Issue Tracker](https://github.com/Prusa3D/Slic3r/issues)

**Is your feature request related to a problem? Please describe.**
Sometimes a printer has better quality printing on one axis over another. For example, a bed slinger sometimes has y-axis artifacts, lower y-axis acceleration, etc. Some axes have more advantage than others in surface quality, where a printer like a Voron switchwire could prefer x-axis over z, then over y. Additionally, adjusting the orientation for many prints on a large print surface is really tedius.

**Describe the solution you'd like**
A setting in the machine settings to configure the most preferred axis and the least preferred axis, where an "orient" button on the "plater" tab could be used to automatically orient all the parts by the multiple of 90 degrees so the longest axis of the parts is parallel to the most preferred axis and the shortest axis of the part is parallel to the least preferred axis.

**Describe alternatives you've considered**a
similar to the auto center feature, just do it all the time when it's enabled (probably a bad idea)

**Additional context**

supermerill, 2021-04-29:
There's so many factor on choosing the good orientation. Bridges, overhangs, layer line for strength...
I don't see an algorithm doing it.

Sineos, 2021-05-01:
I was looking for such a feature as well. Sometimes items exported from CAD will have a weird orientation depending on the origin in the CAD model.
IMHO it would be great to have something like `Place on Face` function for X and Y as well. This would allow a user defined orientation by choosing a Face and have it aligned straight to X or Y

supermerill, 2021-05-01:
"place on X" is something reasonable. 
by "place on X", it means "place on 0-X but only rotate along Z axis to not lift it from the build plate", right?

burner-, 2021-05-01:
How about it will place by given axis and define lift by outermost coordinate? Basically it is almost same than place on face but allow get axis aligned excatly 90 angle to bed.

Sineos, 2021-05-01:
> "place on X" is something reasonable.
> by "place on X", it means "place on 0-X but only rotate along Z axis to not lift it from the build plate", right?

Maybe we have the same understanding: 
"Place on X" would mean rotating around Z until the chosen face is parallel to the X-Axis. Of course contact to the build plate should remain