supermerill/SuperSlicer #1124: Small odd-shaped area does not get internal perimeters

JohanAR, 2021-04-29:
**Describe the bug**
The end of the area with sawteeth only gets a single external perimeter and no internal perimeters, with 3 perimeters configured. It works as expected with 2 perimeters, and with 4 perimeters the area without internal perimeters grows even larger.

**To Reproduce**
Steps to reproduce the behavior:
1. Load my STL
2. Change perimeters to 3
3. Slice
4. See error

**3D Model and Slic3r Configuration Export**

Rename to .3mf
[missing_perimeters.zip](https://github.com/supermerill/SuperSlicer/files/6398595/missing_perimeters.zip)

[t2_prong.stl.zip](https://github.com/supermerill/SuperSlicer/files/6398604/t2_prong.stl.zip)

**Expected behavior**
The whole area is filled with internal perimeters

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Xubuntu 20.04 LTS
 - Superslicer 2.3.55.5 (appimage)

**Additional context**
Not replicable in PrusaSlicer

neophyl, 2021-04-29:
Turn off only one perimeter on top layers.  It is known to cause issues with most of the 2.3 based versions.  The reason it doesn't happen in PS is that PS doesn't have that option at all.

supermerill, 2021-04-29:
can't reproduce on latest nightly.
99% neophyl is right.

closed 2021-06-04 by supermerill

JohanAR, 2021-06-05:
Just verified that 2.3.56 that the bug is gone