supermerill/SuperSlicer #1128: Updated Italian translations.

boromyr, 2021-04-30:
Working italian translation and no crash.
Version 2.3.55.5, I made a quick test on version 56 and seems to work.

Pretradotto automaticamente, con revisione personale durante l'utilizzo.
Alcuni termini specifici sono stati mantenunti in inglese, come i bridge, ironing, brim ecc.
La denominazione  delle aggiunzioni dei supporti è stata revisionata:
- "Aggiungi struttura di supporto" 
- "Aggiungi divieto di supporto" 

Soldi è stato sostituito con "€", varie correzioni di articoli, plurali e singolari e altre correzioni minori.

If you find errors you can post them below.
The non-translated entries must still be added to the .pot files.
Extract files to SuperSlicer\resources\localization\it.

[it.zip](https://github.com/supermerill/SuperSlicer/files/6407621/it.zip)

Liger0, 2021-04-30:
I noticed something that can be improved, for example there is no need to say Cost(€) since on the value there is writen €/kg again, and a lot of stuff are using infinitive mood rather than imperative.

I did change the SuperSlicer.po file but I am not sure how they work, as even changing the SuperSlicer.po file, the changes aren't displayed in the gui, and I am not sure why, maybe the SuperSlicer.po file isn't the one that is used to show the language?

@supermerill is there a documentation of what each file in the localization directory does?
I also don't understand something like why the & here is in the middle of the words.

`#: src/slic3r/GUI/MainFrame.cpp:1297
msgid "&Collapse sidebar"
msgstr "Ridu&ci barra laterale"

#: src/slic3r/GUI/GUI_App.cpp:1645
msgid "&Configuration"
msgstr "&Configurazione"

#: src/slic3r/GUI/GUI_App.cpp:1480
msgid "&Configuration Snapshots"
msgstr "Istantanee di &Configurazione"

#: src/slic3r/GUI/MainFrame.cpp:1194
msgid "&Copy"
msgstr "&Copia"

#: src/slic3r/GUI/MainFrame.cpp:1178
msgid "&Delete selected"
msgstr "Eli&mina selezionati"`

boromyr, 2021-04-30:
Yes I know is not perfect, base I translated with Bing and DeepL, Google gave problems with the syntax of the .po code, then I started to fix the translations that faded more frequently or those that were more annoying to read.

Regarding & in the middle of words not exactly what it is for.

supermerill, 2021-04-30:
> I also don't understand something like why the & here is in the middle of the words.

It's for shortcuts in the menus. ONly displayed if pressing alt and the '&' is before a majuscule.

> is there a documentation of what each file in the localization directory does?

I'll add a readme.
There is currently https://github.com/supermerill/SuperSlicer/wiki/Creating---updating-a-translation 
I've just finished the new tool. Just the readme to copy & update and I'll commit it..

Liger0, 2021-04-30:
Well those seem problematic for example then.

msgstr "Ridu&ci barra laterale"
msgstr "Eli&mina selezionati"`

supermerill, 2021-05-01:
It's not mandatory to keep it.
Look at the menu, and make sure there's no overlapping.

Also I'm not sure of it but it may be needed to write: `msgstr "Eli&Mina selezionati"` o show the shortcut. 
PrusaSlicer wrote `msgid "Re&load from disk"` but I think it may be a mistake and they should have written `msgid "Re&Load from disk"`

supermerill, 2021-05-01:
[localization_tool.zip](https://github.com/supermerill/SuperSlicer/files/6408553/localization_tool.zip)
Here is the updated tool & readme.
It may take me some time to push it on the master branch.

Liger0, 2021-05-01:
Riduci and Elimina  are two full words, they can't be separated. 
Can the & be placed anywhere before a majuscole or is there a difference before which one you place it?
If not it can just be the first character..

supermerill, 2021-05-01:
how to:
 * write al  translation without any '&'
 * compile the .po and start the app with it
 * open a menu, and white on paper on which letter you want the shortcut to be (try to be consistent with other app or get the same as the English version)
 * modify the .po
   * search for the menu string
   * add the & before the letter you want
   * if you want to the letter to by underlined, make it uppercase
 * test again, and with alt+key

Liger0, 2021-05-06:
I made a lot of changes.
I also fixed most of the infinitive mood that end in "are" to imperative. I probably missed some of them, and the "ere" and "ire" ones are still to do.
[SuperSlicer it 6 may 2021.zip](https://github.com/supermerill/SuperSlicer/files/6433226/SuperSlicer.it.6.may.2021.zip)

supermerill, 2021-05-21:
I fixed two little errors.

[it.zip](https://github.com/supermerill/SuperSlicer/files/6523933/it.zip)

Liger0, 2021-05-23:
I have a question about the localization tool (using 2.3.56 beta)
I inserted in the directory the prusa slicer.po of the last stable release, and left the slic3r.po already there. I did put in the settings 

`#data = C:/local/Slic3r/resources/localization/lang/TODO.po
#data = C:/local/Slic3r/resources/localization/lang/Slic3r.po
#data = Slic3r++.po
data = it/PrusaSlicer_it.po
data = it/SuperSlicer.po
data = it/Slic3r.po

ui_dir = ../ui_layout
allow_msgctxt = false
ignore_case = false

input = Slic3r.pot
todo = TODO.po
output = Slic3r.po
`

and executed `python pom_merger.py`

but the generated TODO.po has a lot less entries than the TODO.po already present in the 2.3.56 it localization directory, what am I doing wrong?

Liger0, 2021-05-23:
I made various changes.
The problems are though that for some reasons when compiling the characters aren't read correctly.
[image]
I also worked on the TODO file but I am failing at merging it
[image]

boromyr, 2021-05-23:
Strange, nothing happened to me. I used Merill's correct version.
In my opinion we are making too much confusion with translations, I would try to choose the latest stable version of the translation, update entries from the .pot and todo file, and use it as a starting file.

Liger0, 2021-05-23:
I based this on the corrected version of merill. Also did a lot on the TODO.
The translation works just fine, just if I compile it shows like that the accents.
[it.zip](https://github.com/supermerill/SuperSlicer/files/6528429/it.zip)

Liger0, 2021-05-23:
Nevermind, I fixed the characters.

boromyr, 2021-05-23:
If you use Poedit, you can go to "Translation" and "Update from Pot file"
This is the already updated file, the items with "requires verification" contain the new voices automatically translated, as well as those that were already present.
[SuperSlicer.po.txt](https://github.com/supermerill/SuperSlicer/files/6528448/SuperSlicer.po.txt)

Liger0, 2021-05-23:
[it.zip](https://github.com/supermerill/SuperSlicer/files/6528473/it.zip)
This is the working file, with the unmerged TODO (I fixed some errors and the characters). 
It works fine in SuperSlicer
In poedit I have this tho:
[image]

boromyr, 2021-05-23:
Red Poedit errors refer to a different letter after a "% ". So "% for" to "% per" cause a mistake that SuperSlicer has no.
This is your last file with the new non-translated entries. I propose to start from this.
[SuperSlicer.po.txt](https://github.com/supermerill/SuperSlicer/files/6528487/SuperSlicer.po.txt)

Liger0, 2021-05-23:
> Red Poedit errors refer to a different letter after a "% ". So "% for" to "% per" cause a mistake that SuperSlicer has no.
> This is your last file with the new non-translated entries. I propose to [...]

Oh, if it's the % it should be fine then since it's the % as a text.
Isn't my last file good? There shouldn't be any error.

Liger0, 2021-05-23:
> If you use Poedit, you can go to "Translation" and "Update from Pot file"
> This is the already updated file, the items with "requires verification" contain the new voices automatically translated, as [...]

Do you update from the last beta's Slic3r.pot? The old nightlies had various pots for each slic3r flavour.

Liger0, 2021-05-23:
> 
> 
> This is your last file with the new non-translated entries. I propose to start from this.
> [SuperSlicer.po.txt](https://github.com/supermerill/SuperSlicer/files/6528487/SuperSlicer.po.txt)

That one doesn't work, it makes the program crash.

Liger0, 2021-05-23:
> 
> 

> This is your last file with the new non-translated entries. I propose to start from this.
> [SuperSlicer.po.txt](https://github.com/supermerill/SuperSlicer/files/6528487/SuperSlicer.po.txt)
I started from this and worked a bit. Tell me what you think. It works for me, but I cannot merge the TODO, tell me if you succeed.
[it 23 may.zip](https://github.com/supermerill/SuperSlicer/files/6528690/it.zip)

supermerill, 2021-05-26:
found the bug.
[it_po_05_23_merged.zip](https://github.com/supermerill/SuperSlicer/files/6548765/it_po_05_23_merged.zip)

Liger0, 2021-05-26:
> 
> 
> found the bug.
> [it_po_05_23_merged.zip](https://github.com/supermerill/SuperSlicer/files/6548765/it_po_05_23_merged.zip)

I tried this translation. Just like the old ones, it still crashes the debug build to me like in #1166 , while working in old nightlies.
Same error
https://user-images.githubusercontent.com/20385116/119368180-00971000-bcb3-11eb-8694-5ca9fe9c2c0b.png

boromyr, 2021-05-27:
With the latest update everything works correctly, Liger I would suggest to start from the translation file present in 56.1 and work only on that to avoid confusion.
Regarding the Visual C++ error try to install the full package with all libraries from  https://github.com/abbodi1406/vcredist

Liger0, 2021-05-27:
> 
> 
> With the latest update everything works correctly, Liger I would suggest to start from the translation file present in 56.1 and work only on that to avoid confusion.
> Regarding the Visual C++ error try to install the full package with all libraries from https://github.com/abbodi1406/vcredist

I don't know if in that version he's inserted the fixed translation he just posted.
Anyway, I installed that software, and it's still crashing the same. Old nightly as I said doesn't....

supermerill, 2021-05-27:
Can't reproduce your crash. It happens at startup, right?
Did you try with a new empty setting directory (using the local-setting .bat for example)?

At least, the error seems to have changed. It's now maybe an error in the .po

Liger0, 2021-05-27:
Actually, if I tried the "debug" release that you posted some days ago, I still have the error  about the debug assertion failed.
While, if I try the "better spacing" last release, I get the "invalid string position".
Both using the last translation of "better spacing".

And using the local settings bat file, effectively there's no crash.
[SuperSlicer.zip](https://github.com/supermerill/SuperSlicer/files/6553748/SuperSlicer.zip)
This then should be the incriminated directory, maybe it's worth seeing if some settings cause the issue?
So I can import old profiles etc safely into the new version. Plus maybe some setting can make the new slicer crash.

supermerill, 2021-05-27:
Found the error. It's in the notification manager.
It creates an index on a long text string and then tries to apply it on a short one.

Prusa has already revamp this code, so maybe i'll grab their version. instead of just patching it.

Liger0, 2021-05-27:
Perfect, with the last debug release all the crashes are solved.
I think it's fine to close all the related issues now, leaving this thread for translation additions and updates of mistakes.

Liger0, 2021-05-27:
I've made a lot of updates based on the last release
[it 27 may.zip](https://github.com/supermerill/SuperSlicer/files/6555237/it.27.may.zip)
@supermerill I noticed the bar on the bottom "check for updates..." is in the translation, but its translation isn't used?

boromyr, 2021-05-27:
Liger, do you use Slicer.po or SuperSlicer.po? Let's pick one. I'd say Slicer.po since that was the one merill fixed.

Liger0, 2021-05-27:
I use superslicer.po

supermerill, 2021-05-27:
> @supermerill I noticed the bar on the bottom "check for updates..." is in the translation, but its translation isn't used?

that one?
[image]

Liger0, 2021-05-27:
Negative, I mean this one.
[image]

Plus can I ask the difference between the slic3r and prusaslicer? I thought the slic3r was just a reference to take from the configuration used in the slic3r project, and I just need to work on the superslicer.po, right?

supermerill, 2021-05-27:
Ah, yes, there is an error here, will fix.

there is no more superslicer.po
slic3r and superslicer will be the same things in the (near?) future.
To accomplish this goal, without disturbing the current state, I made superslicer name-agnostic. When a static name is needed, I use Slic3r and everything with SuperSlicer inside is generated at compile time (or runtime).
That's why the translation file is slic3r.po and the generated file is superslicer.mo

Liger0, 2021-05-27:
I've made a big update first, I hope it's fine as I used the superslicer PO. 
If you need to make the fix please to it in here :°)
There are alot of menus that look english while the translation is present.
[it 27 may v3.zip](https://github.com/supermerill/SuperSlicer/files/6556997/it.27.may.v3.zip)

Liger0, 2021-06-07:
Fixed various errors. The translation is complete and mostly checked and it's a big jump from the one present in the last stable, could this be updated?

Thanks. :)

[it 7 june.zip](https://github.com/supermerill/SuperSlicer/files/6609233/it.7.june.zip)

Liger0, 2021-06-07:
I totally forgot it still needed  to fix the infinite mood tences that ended by "ere" and "ire". Now it should be taken care of it too.
[it 7 june v2.zip](https://github.com/supermerill/SuperSlicer/files/6611531/it.7.june.v2.zip)

supermerill, 2021-06-24:
updated it translation in https://github.com/supermerill/SuperSlicer/blob/Nigthly/resources/localization/it/Slic3r.po
(you can get the latest nightly to download & test it)
I added small things that I could translate myself.
There is no more than ~ 70 things still to translate.

Sterbo88, 2021-10-30:
is it normal that in the print settings in the left column, width and flow is not translated?  and also perimeters and shells?  I ask to understand if I have correctly entered the latest language update

supermerill, 2021-10-30:
It's not up to date.
But you can help improve it.
If you can't follow the tutorial, I can send you a file to complete.

supermerill, 2021-10-31:
If someone is ready to add the 242 untranslated entries (2799 are already made)
here is my it working files.
You only have to update the TODO and give me back.
The Slic3r.po contains useful translation
The database contains all translations since the beginning, to be able to reuse them if needed, or as a comparison point.

How to modify the TODO?
```
#Similar to me: %s flow rate is maximized 
# 2    changes:  flow rate is maximized 
#  translation:  Il flusso viene massimizzato 
msgid "%s flow rate is maximized "
msgstr ""
```
there is a comment section, behind '#'
the last line is the one that need to be filled
The comment give us advice with similar translations already made.
Be careful! the number of '%' need to be the same or a crash may occur.
So here, the last line may become something like:
`msgstr "Il flusso %s viene massimizzato, ecco"`

macchinette, 2022-02-17:
> Se qualcuno è pronto ad aggiungere le 242 voci non tradotte (2799 sono già state fatte) ecco i miei file di lavoro. Devi solo aggiornare il TODO e restituirmi. Il Slic3r.po contiene traduzioni utili Il database contiene tutte le traduzioni fin dall'inizio, per poterle riutilizzare se necessario, o come punto di confronto.
> 
> Come modificare il TODO?
> 
> ```
> #Similar to me: %s flow rate is maximized 
> # 2    changes:  flow rate is maximized 
> #  translation:  Il flusso viene massimizzato 
> msgid "%s flow rate is maximized "
> msgstr ""
> ```
> 
> c'è una sezione commenti, dietro '#' l'ultima riga è quella che deve essere riempita Il commento ci dà consigli con traduzioni simili già fatte. Stai attento! il numero di '%' deve essere lo stesso o potrebbe verificarsi un arresto anomalo. Quindi, qui, l'ultima riga potrebbe diventare qualcosa del tipo: `msgstr "Il flusso %s viene massimizzato, ecco"`
> 
> [it_2021_10_31.zip](https://github.com/supermerill/SuperSlicer/files/7449617/it_2021_10_31.zip)
> 
> Nota anche che alcuni testi hanno visto molte righe aggiunte, e quindi l'algoritmo non ha riconosciuto nulla di simile, ma potresti essere in grado di trovare qualcosa da solo da una ricerca nel database.

Buongiorno Supermerill, siccome il tuo programma lo adoro e voglio essere di aiuto, mi sono permesso di tradurre i file allegati in italiano.
Siccome non sono capace di compilarlo in modo da generare il file necessario per controllarlo, ti chiedo se lo puoi fare tu e mi rendo disponibile alle nuove traduzioni per l'italiano che mancheranno con i futuri aggiornamenti, l'importante per me è avere un file da poter editare con blocco note.
Ho notato che nel file generale delle traduzioni ci sono dalla riga 15006 dei percorsi che fanno riferimento ad una unità specifica.
[image]

[it_2022_02_17.zip](https://github.com/supermerill/SuperSlicer/files/8085353/it_2022_02_17.zip)

Grazie Cristian

macchinette, 2022-04-02:
@supermerill
Good morning, I have a couple of questions:
1) I wanted to ask you if you have entered the translations I sent you?
2) Since in version 2.4.58.2 you mention the fact that translations are easily updatable, I was wondering which files I need to edit and if the compilation is automatic.

Thank you

supermerill, 2022-04-02:
> I wanted to ask you if you have entered the translations I sent you?

i guess. I'll check.

> Since in version 2.4.58.2 you mention the fact that translations are easily updatable, I was wondering which files I need to edit and if the compilation is automatic.

https://github.com/supermerill/SuperSlicer/tree/master_dev/resources/localization/it

I now not only commit the slic3r.po localization file but also the other ones that I use to create it.
All you have to do is to edit the TODO.po (in your computer or directly on github), and to ask for a pull request (or send it to me by another mean). 

https://github.com/supermerill/SuperSlicer/tree/master_dev/resources/localization/it

supermerill, 2022-04-13:
> Could the problem be above?

yes, forgot to add a `L` like that
```
def->enum_values.push_back(L("All surfaces"));
def->enum_values.push_back(L("Not on top"));
def->enum_values.push_back(L("Only on top"));
```
will fix
Note that you can add them manually to the po file.

macchinette, 2022-04-13:
Hello thanks for the reply, yes I look for them by hand in the program and insert them, I'm doing it with the whole program but because the .po file from which I started was not updated and the translations had already been done and I didn't want to redo all, unfortunately I can not do anything with Python does not work for me.  When I will not find any more by myself looking at the program and therefore I will be desperate, to be sure that they have all been translated we will have to do the work with Python to be able to know which translations are missing, I hope at that point you can do it.  thank you

Inviato da iPhone

supermerill, 2022-04-13:
I can run python for you, if you need to merge some .po together.
If python is install but refuse to run the merge script, try to install the dependencies: 
execute `python -m pip install python-Levenshtein` in a command prompt.

macchinette, 2022-04-18:
> I can run python for you, if you need to merge some .po together. If python is install but refuse to run the merge script, try to install the dependencies: execute `python -m pip install python-Levenshtein` in a command prompt.

Happy Easter!
I translated everything I could find, testing it on versions 2.3.57.12 and 2.4.58.3

Now I'm stopped because with Phyton I can't and also the Levenshtein module can't be installed.

So I ask you: can you do the magic to get the TODO file with the missing translations for both versions?

Attached is the ZIP file with the updated Slic3r.po file in Italian and the compiled SuperSlicer.mo.
[it_2022_04_18.zip](https://github.com/supermerill/SuperSlicer/files/8506144/it_2022_04_18.zip)

Once I have the TODO.po file with the missing translations I will translate them and insert them in the Slic3r.po file and I will compile with PoEdit (as I have done until now), at this point I will open the Pull Request for the Italian Translation and I will upload the updated and complete Slic3r.po and SuperSlicer.mo files ready to release.

Thank you for your collaboration, I love the SuperSlicer project!

macchinette, 2023-11-01:
Hi @supermerill Supermerill, I'm very happy about your new work adventure,
Since you will be returning to work on SuperSlicer, I wanted to ask you to insert this translation file into Italian.

[SuperSlicer ITA.zip](https://github.com/supermerill/SuperSlicer/files/13226962/SuperSlicer.ITA.zip)

I continued the Italian translation project and translated everything up to the last night you made and it seems like a good job.

My intention is to continue translating with every update you make, because for me it is important to make the most of your magnificent slicer.

Thank you very much and welcome back.

The image shows the only translation I can't do
[image]

supermerill, 2023-11-22:
Hi
Can you send me the .po instead?
I'm planning for the future to dedicate a repository for each language, so you won't need me anymore to update the itallian file.

macchinette, 2023-11-23:
> Hi Can you send me the .po instead? I'm planning for the future to dedicate a repository for each language, so you won't need me anymore to update the itallian file.

Hi SuperMerill,
here is the zip with both files (.PO and .MO) updated to the latest night, as I had previously indicated I inserted translations not present in the Slic3r.POT file, so if you make it process only that one the translations disappear, and it is for this reason I will also send you the .MO which contains all the translations including those of the PrusaSlicer.POT file
[it_2023_11_23.zip](https://github.com/supermerill/SuperSlicer/files/13452492/it_2023_11_23.zip)

The translations that don't work are these, one of which is missing translation even in PrusaSlicer
[image]
[image]
[image]

Thanks for collaboration.
Excuse my English

supermerill, 2023-11-27:
Thanks.
I have to do a 59.4 then.
>
You know, for the translation, there is a  big [it_database.po](https://github.com/supermerill/SuperSlicer/blob/master/resources/localization/it/it_database.po) file with the translations of everything since the start, and everything you send me replace / is added into it. When making the releases, each line is translated by finding the right one inside this big file. That way, if a line is reverted to a previous one, it doesn't need to be re-translated, and it provide a good chance for the [script](https://github.com/supermerill/SuperSlicer/blob/master/resources/localization/pom_merger.py) to find equivalent sentence to propose as baseline when creating the TODO.po.

macchinette, 2024-01-27:
> Thanks. I have to do a 59.4 then.
> 
> > had previously indicated I inserted translations not present in the Slic3r.POT file, so if you make it process only that one the translations disappear, and it is for this reason I will also send you the .MO which contains all the translations including those of the PrusaSlicer.POT file
> 
> You know, for the translation, there is a big [it_database.po](https://github.com/supermerill/SuperSlicer/blob/master/resources/localization/it/it_database.po) file with the translations of everything since the start, and everything you send me replace / is added into it. When making the releases, each line is translated by finding the right one inside this big file. That way, if a line is reverted to a previous one, it doesn't need to be re-translated, and it provide a good chance for the [script](https://github.com/supermerill/SuperSlicer/blob/master/resources/localization/pom_merger.py) to find equivalent sentence to propose as baseline when creating the TODO.po.

Hi Supermerill,
thank you for the effort you are putting into development.
I improved all the Italian translations (so all the Italian translations should be replaced) and translated all the missing sentences up to your last night.
Unfortunately the translations for the sentences in the photo below do not appear and I think the problem is in the code in "format_wxstr".
Then the other 4 in the photo are not in the code designed to be translated.

[image]

Attached with the translations. [it_2024_01_27.zip](https://github.com/supermerill/SuperSlicer/files/14073125/it_2024_01_27.zip)

Thank you

P.S. there would still be these:

[image]
[image]
[image]

supermerill, 2024-01-27:
1- I cleaned it up a bit, and re-generte the pot, there are inside. I give you the new pot below.
2- Weird that the other are ok. I changed a bit the code where it's tranlated, I wasn't sure how it managed to work before...
3- fixed.
All changes are available in next nightly.
[Slic3r.pot..zip](https://github.com/supermerill/SuperSlicer/files/14073379/Slic3r.pot.zip)

supermerill, 2024-01-27:
[TODO.zip](https://github.com/supermerill/SuperSlicer/files/14073384/TODO.zip)

macchinette, 2024-01-28:
> for current 2.5: [TODO.zip](https://github.com/supermerill/SuperSlicer/files/14073384/TODO.zip)

HI,
I translated everything that TODO said was missing.
I don't understand why your check detects missing translations of Z z X x Y y _ etc...

This part of the code is doubled in the PresetHints.cpp file
if (preset_fil.config.opt_int("top_fan_speed", 0) == 1)
          out += "\n\n" + _L("! 1 for maximum fan speed is not recommended, set it to 0 to stop the fan!");

In the code the translations that don't work that are in the photo below are set with format_wxstr while those that work boost::format
[image]

The files are attached. [it_2024_01_28.zip](https://github.com/supermerill/SuperSlicer/files/14076538/it_2024_01_28.zip)

Thank you

supermerill, 2024-02-01:
once I re-generate the .mo, these translation are translated for me. I'll check again in the release.