supermerill/SuperSlicer #1129: boost/multiprecision/cpp_int/limits.hpp:27:36: error: ambiguous conversion for functional-style cast from       'number<boost::multiprecision::backends::cpp_int_backend<128, 128, boost::multiprecision::unsigned_magnitude, boost::multiprecision::unchecked, void>, et_off>' to       'result_type' (aka 'number<boost::multiprecision::cpp_int_backend<128U, 128U, (cpp_integer_type)1U, (cpp_int_check_type)0U, void>

aep, 2021-05-01:
```
Darwin *.lan 20.3.0 Darwin Kernel Version 20.3.0: Thu Jan 21 00:06:51 PST 2021; root:xnu-7195.81.3~1/RELEASE_ARM64_T8101 arm64
```

```
[ 44%] Building CXX object src/libslic3r/CMakeFiles/libslic3r.dir/Arrange.cpp.o
In file included from /Users/aep/Downloads/SuperSlicer/src/libslic3r/Arrange.cpp:21:
In file included from /Users/aep/Downloads/SuperSlicer/deps/build/destdir/usr/local/include/boost/multiprecision/integer.hpp:9:
In file included from /Users/aep/Downloads/SuperSlicer/deps/build/destdir/usr/local/include/boost/multiprecision/cpp_int.hpp:1979:
/Users/aep/Downloads/SuperSlicer/deps/build/destdir/usr/local/include/boost/multiprecision/cpp_int/limits.hpp:27:36: error: ambiguous conversion for functional-style cast from
      'number<boost::multiprecision::backends::cpp_int_backend<128, 128, boost::multiprecision::unsigned_magnitude, boost::multiprecision::unchecked, void>, et_off>' to
      'result_type' (aka 'number<boost::multiprecision::cpp_int_backend<128U, 128U, (cpp_integer_type)1U, (cpp_int_check_type)0U, void>,
      (boost::multiprecision::expression_template_option)0U>')
   static const result_type val = -result_type(~ui_type(0));
                                   ^~~~~~~~~~~~~~~~~~~~~~~
/Users/aep/Downloads/SuperSlicer/deps/build/destdir/usr/local/include/boost/multiprecision/cpp_int/limits.hpp:129:22: note: in instantiation of function template specialization
      'std::detail::get_min<128, 128, boost::multiprecision::signed_magnitude, boost::multiprecision::unchecked, void, boost::multiprecision::et_off>' requested here
      return detail::get_min<MinBits, MaxBits, SignType, Checked, Allocator, ExpressionTemplates>(boost::multiprecision::backends::is_fixed_precision<backend_type>(), b...
                     ^
/Users/aep/Downloads/SuperSlicer/deps/build/destdir/usr/local/include/boost/integer/common_factor_rt.hpp:447:44: note: in instantiation of member function
      'std::__1::numeric_limits<boost::multiprecision::number<boost::multiprecision::backends::cpp_int_backend<128, 128, boost::multiprecision::signed_magnitude,
      boost::multiprecision::unchecked, void>, boost::multiprecision::et_off> >::min' requested here
    if(a == (std::numeric_limits<Integer>::min)())
                                           ^
/Users/aep/Downloads/SuperSlicer/src/libslic3r/Arrange.cpp:473:12: note: in instantiation of function template specialization
      'libnest2d::minAreaBoundingBox<ClipperLib::Polygon, long long, boost::rational<boost::multiprecision::number<boost::multiprecision::backends::cpp_int_backend<128, 128,
      boost::multiprecision::signed_magnitude, boost::multiprecision::unchecked, void>, boost::multiprecision::et_off> > >' requested here
    return minAreaBoundingBox<S, TCompute<S>, boost::rational<LargeInt>>(sh)
           ^
/Users/aep/Downloads/SuperSlicer/src/libslic3r/Arrange.cpp:516:26: note: in instantiation of function template specialization
      'Slic3r::arrangement::min_area_boundingbox_rotation<ClipperLib::Polygon>' requested here
            itm.rotation(min_area_boundingbox_rotation(itm.rawShape()));
                         ^
/Users/aep/Downloads/SuperSlicer/src/libslic3r/Arrange.cpp:643:5: note: in instantiation of function template specialization
      'Slic3r::arrangement::_arrange<libnest2d::_Box<ClipperLib::IntPoint> >' requested here
    _arrange(items, fixeditems, to_nestbin(bed), params, pri, cfn);
    ^
/Users/aep/Downloads/SuperSlicer/deps/build/destdir/usr/local/include/boost/multiprecision/number.hpp:120:25: note: candidate constructor [with Other =
      boost::multiprecision::backends::cpp_int_backend<128, 128, boost::multiprecision::unsigned_magnitude, boost::multiprecision::unchecked, void>, ET =
      boost::multiprecision::et_off]
   BOOST_MP_FORCEINLINE number(const number<Other, ET>& val,
                        ^
/Users/aep/Downloads/SuperSlicer/deps/build/destdir/usr/local/include/boost/multiprecision/number.hpp:126:13: note: candidate constructor [with Other =
      boost::multiprecision::backends::cpp_int_backend<128, 128, boost::multiprecision::unsigned_magnitude, boost::multiprecision::unchecked, void>, ET =
      boost::multiprecision::et_off]
   explicit number(const number<Other, ET>& val, typename boost::enable_if_c<
            ^
In file included from /Users/aep/Downloads/SuperSlicer/src/libslic3r/Arrange.cpp:8:
In file included from /Users/aep/Downloads/SuperSlicer/src/libnest2d/include/libnest2d/placers/nfpplacer.hpp:18:
In file included from /Users/aep/Downloads/SuperSlicer/src/libnest2d/include/libnest2d/placers/placer_boilerplate.hpp:4:
/Users/aep/Downloads/SuperSlicer/src/libnest2d/include/libnest2d/nester.hpp:839:47: warning: lambda capture 'this' is not used [-Wunused-lambda-capture]
        if(infl > 0) std::for_each(from, to, [this, infl](Item& item) {
                                              ^
/Users/aep/Downloads/SuperSlicer/src/libslic3r/Arrange.cpp:391:15: note: in instantiation of function template specialization
      'libnest2d::_Nester<libnest2d::placers::_NofitPolyPlacer<ClipperLib::Polygon, libnest2d::_Box<ClipperLib::IntPoint> >,
      libnest2d::selections::_FirstFitSelection<ClipperLib::Polygon> >::execute<std::__1::__wrap_iter<std::__1::reference_wrapper<libnest2d::_Item<ClipperLib::Polygon> > *> >'
      requested here
        m_pck.execute(from, to);
              ^
/Users/aep/Downloads/SuperSlicer/src/libslic3r/Arrange.cpp:518:5: note: in instantiation of function template specialization
      'Slic3r::arrangement::AutoArranger<libnest2d::_Box<ClipperLib::IntPoint>
      >::operator()<std::__1::__wrap_iter<std::__1::reference_wrapper<libnest2d::_Item<ClipperLib::Polygon> > *> >' requested here
    arranger(inp.begin(), inp.end());
    ^
/Users/aep/Downloads/SuperSlicer/src/libslic3r/Arrange.cpp:643:5: note: in instantiation of function template specialization
      'Slic3r::arrangement::_arrange<libnest2d::_Box<ClipperLib::IntPoint> >' requested here
    _arrange(items, fixeditems, to_nestbin(bed), params, pri, cfn);
    ^
In file included from /Users/aep/Downloads/SuperSlicer/src/libslic3r/Arrange.cpp:8:
In file included from /Users/aep/Downloads/SuperSlicer/src/libnest2d/include/libnest2d/placers/nfpplacer.hpp:18:
[... 19 lines cut ...]
In file included from /Users/aep/Downloads/SuperSlicer/src/libslic3r/Arrange.cpp:8:
In file included from /Users/aep/Downloads/SuperSlicer/src/libnest2d/include/libnest2d/placers/nfpplacer.hpp:18:
In file included from /Users/aep/Downloads/SuperSlicer/src/libnest2d/include/libnest2d/placers/placer_boilerplate.hpp:4:
/Users/aep/Downloads/SuperSlicer/src/libnest2d/include/libnest2d/nester.hpp:839:47: warning: lambda capture 'this' is not used [-Wunused-lambda-capture]
        if(infl > 0) std::for_each(from, to, [this, infl](Item& item) {
                                              ^
/Users/aep/Downloads/SuperSlicer/src/libslic3r/Arrange.cpp:391:15: note: in instantiation of function template specialization
      'libnest2d::_Nester<libnest2d::placers::_NofitPolyPlacer<ClipperLib::Polygon, ClipperLib::Polygon>, libnest2d::selections::_FirstFitSelection<ClipperLib::Polygon>
      >::execute<std::__1::__wrap_iter<std::__1::reference_wrapper<libnest2d::_Item<ClipperLib::Polygon> > *> >' requested here
        m_pck.execute(from, to);
              ^
/Users/aep/Downloads/SuperSlicer/src/libslic3r/Arrange.cpp:518:5: note: in instantiation of function template specialization
      'Slic3r::arrangement::AutoArranger<ClipperLib::Polygon>::operator()<std::__1::__wrap_iter<std::__1::reference_wrapper<libnest2d::_Item<ClipperLib::Polygon> > *> >'
      requested here
    arranger(inp.begin(), inp.end());
    ^
/Users/aep/Downloads/SuperSlicer/src/libslic3r/Arrange.cpp:643:5: note: in instantiation of function template specialization
      'Slic3r::arrangement::_arrange<ClipperLib::Polygon>' requested here
    _arrange(items, fixeditems, to_nestbin(bed), params, pri, cfn);
    ^
3 warnings and 1 error generated.
make[2]: *** [src/libslic3r/CMakeFiles/libslic3r.dir/Arrange.cpp.o] Error 1
make[1]: *** [src/libslic3r/CMakeFiles/libslic3r.dir/all] Error 2
make: *** [all] Error 2
```

supermerill, 2021-05-01:
I can't fix boost on macos.

aep, 2021-05-01:
2.2.53.4  worked fine. maybe it just needs a down or upgrade of boost?

supermerill, 2021-05-02:
maybe I have to merge with the latest prusa version to be compatible with macos 11.2.
Will do after the release of the current version.