supermerill/SuperSlicer #1130: SuperSlicer crash on Windows

panik988, 2021-05-01:
Hello,
I've got a problem with the new nighly in Win 10 64 bit.
When I slice a quite big file the program gave me an error and after crash.
I tryed on 3 pc and I had the same problem.
Thank you
Marco

[image]
[image]

Here my system information:
SuperSlicer
Version:   2.3.56.0
Build:     Slic3r_2.3.56.0

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19042), 64-bit edition
Total RAM size [MB]: 19,251MB
OpenGL installation
GL version:   4.6.13596 Compatibility Profile Context 20.10.26 27.20.1026.1
Vendor:       ATI Technologies Inc.
Renderer:     AMD Radeon(TM) Vega 8 Graphics
GLSL version: 4.60
Installed extensions:
GL_AMDX_debug_output
GL_AMD_blend_minmax_factor
GL_AMD_conservative_depth
GL_AMD_debug_output
GL_AMD_depth_clamp_separate
GL_AMD_draw_buffers_blend
GL_AMD_framebuffer_sample_positions
GL_AMD_gcn_shader
GL_AMD_gpu_shader_half_float
GL_AMD_gpu_shader_half_float_fetch
GL_AMD_gpu_shader_int16
GL_AMD_gpu_shader_int64
GL_AMD_interleaved_elements
GL_AMD_multi_draw_indirect
GL_AMD_name_gen_delete
GL_AMD_occlusion_query_event
GL_AMD_performance_monitor
GL_AMD_pinned_memory
GL_AMD_query_buffer_object
GL_AMD_sample_positions
GL_AMD_seamless_cubemap_per_texture
GL_AMD_shader_atomic_counter_ops
GL_AMD_shader_stencil_export
GL_AMD_shader_stencil_value_export
GL_AMD_shader_trace
GL_AMD_shader_trinary_minmax
GL_AMD_sparse_texture
GL_AMD_sparse_texture_pool
GL_AMD_stencil_operation_extended
GL_AMD_texture_cube_map_array
GL_AMD_texture_texture4
GL_AMD_transform_feedback3_lines_triangles
GL_AMD_transform_feedback4
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
GL_ARB_get_program_binary
GL_ARB_get_texture_sub_image
GL_ARB_gl_spirv
GL_ARB_gpu_shader5
GL_ARB_gpu_shader_fp64
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_imaging
GL_ARB_indirect_parameters
GL_ARB_instanced_arrays
GL_ARB_internalformat_query
GL_ARB_internalformat_query2
GL_ARB_invalidate_subdata
GL_ARB_map_buffer_alignment
GL_ARB_map_buffer_range
GL_ARB_multi_bind
GL_ARB_multi_draw_indirect
GL_ARB_multisample
GL_ARB_multitexture
GL_ARB_occlusion_query
GL_ARB_occlusion_query2
GL_ARB_parallel_shader_compile
GL_ARB_pipeline_statistics_query
GL_ARB_pixel_buffer_object
GL_ARB_point_parameters
GL_ARB_point_sprite
GL_ARB_polygon_offset_clamp
GL_ARB_program_interface_query
GL_ARB_provoking_vertex
GL_ARB_query_buffer_object
GL_ARB_robust_buffer_access_behavior
GL_ARB_sample_shading
GL_ARB_sampler_objects
GL_ARB_seamless_cube_map
GL_ARB_seamless_cubemap_per_texture
GL_ARB_separate_shader_objects
GL_ARB_shader_atomic_counter_ops
GL_ARB_shader_atomic_counters
GL_ARB_shader_ballot
GL_ARB_shader_bit_encoding
GL_ARB_shader_draw_parameters
GL_ARB_shader_group_vote
GL_ARB_shader_image_load_store
GL_ARB_shader_image_size
GL_ARB_shader_objects
GL_ARB_shader_precision
GL_ARB_shader_stencil_export
GL_ARB_shader_storage_buffer_object
GL_ARB_shader_subroutine
GL_ARB_shader_texture_image_samples
GL_ARB_shader_texture_lod
GL_ARB_shader_viewport_layer_array
GL_ARB_shading_language_100
GL_ARB_shading_language_420pack
GL_ARB_shading_language_packing
GL_ARB_shadow
GL_ARB_shadow_ambient
GL_ARB_sparse_buffer
GL_ARB_sparse_texture
GL_ARB_spirv_extensions
GL_ARB_stencil_texturing
GL_ARB_sync
GL_ARB_tessellation_shader
GL_ARB_texture_barrier
GL_ARB_texture_border_clamp
GL_ARB_texture_buffer_object
GL_ARB_texture_buffer_object_rgb32
GL_ARB_texture_buffer_range
GL_ARB_texture_compression
GL_ARB_texture_compression_bptc
GL_ARB_texture_compression_rgtc
GL_ARB_texture_cube_map
GL_ARB_texture_cube_map_array
GL_ARB_texture_env_add
GL_ARB_texture_env_combine
GL_ARB_texture_env_crossbar
GL_ARB_texture_env_dot3
GL_ARB_texture_float
GL_ARB_texture_gather
GL_ARB_texture_mirror_clamp_to_edge
GL_ARB_texture_mirrored_repeat
GL_ARB_texture_multisample
GL_ARB_texture_non_power_of_two
GL_ARB_texture_query_levels
GL_ARB_texture_query_lod
GL_ARB_texture_rectangle
GL_ARB_texture_rg
GL_ARB_texture_rgb10_a2ui
GL_ARB_texture_snorm
GL_ARB_texture_stencil8
GL_ARB_texture_storage
GL_ARB_texture_storage_multisample
GL_ARB_texture_swizzle
GL_ARB_texture_view
GL_ARB_timer_query
GL_ARB_transform_feedback2
GL_ARB_transform_feedback3
GL_ARB_transform_feedback_instanced
GL_ARB_transform_feedback_overflow_query
GL_ARB_transpose_matrix
GL_ARB_uniform_buffer_object
GL_ARB_vertex_array_bgra
GL_ARB_vertex_array_object
GL_ARB_vertex_attrib_64bit
GL_ARB_vertex_attrib_binding
GL_ARB_vertex_buffer_object
GL_ARB_vertex_program
GL_ARB_vertex_shader
GL_ARB_vertex_type_10f_11f_11f_rev
GL_ARB_vertex_type_2_10_10_10_rev
GL_ARB_viewport_array
GL_ARB_window_pos
GL_ATI_draw_buffers
GL_ATI_envmap_bumpmap
GL_ATI_fragment_shader
GL_ATI_separate_stencil
GL_ATI_texture_compression_3dc
GL_ATI_texture_env_combine3
GL_ATI_texture_float
GL_ATI_texture_mirror_once
GL_EXT_abgr
GL_EXT_bgra
GL_EXT_bindable_uniform
GL_EXT_blend_color
GL_EXT_blend_equation_separate
GL_EXT_blend_func_separate
GL_EXT_blend_minmax
GL_EXT_blend_subtract
GL_EXT_compiled_vertex_array
GL_EXT_copy_buffer
GL_EXT_copy_texture
GL_EXT_depth_bounds_test
GL_EXT_direct_state_access
GL_EXT_draw_buffers2
GL_EXT_draw_instanced
GL_EXT_draw_range_elements
GL_EXT_fog_coord
GL_EXT_framebuffer_blit
GL_EXT_framebuffer_multisample
GL_EXT_framebuffer_object
GL_EXT_framebuffer_sRGB
GL_EXT_geometry_shader4
GL_EXT_gpu_program_parameters
GL_EXT_gpu_shader4
GL_EXT_histogram
GL_EXT_memory_object
GL_EXT_memory_object_win32
GL_EXT_multi_draw_arrays
GL_EXT_packed_depth_stencil
GL_EXT_packed_float
GL_EXT_packed_pixels
GL_EXT_pixel_buffer_object
GL_EXT_point_parameters
GL_EXT_polygon_offset_clamp
GL_EXT_provoking_vertex
GL_EXT_rescale_normal
GL_EXT_secondary_color
GL_EXT_semaphore
GL_EXT_semaphore_win32
GL_EXT_separate_specular_color
GL_EXT_shader_image_load_store
GL_EXT_shader_integer_mix
GL_EXT_shadow_funcs
GL_EXT_stencil_wrap
GL_EXT_subtexture
GL_EXT_texgen_reflection
GL_EXT_texture3D
GL_EXT_texture_array
GL_EXT_texture_buffer_object
GL_EXT_texture_compression_bptc
GL_EXT_texture_compression_latc
GL_EXT_texture_compression_rgtc
GL_EXT_texture_compression_s3tc
GL_EXT_texture_cube_map
GL_EXT_texture_edge_clamp
GL_EXT_texture_env_add
GL_EXT_texture_env_combine
GL_EXT_texture_env_dot3
GL_EXT_texture_filter_anisotropic
GL_EXT_texture_integer
GL_EXT_texture_lod
GL_EXT_texture_lod_bias
GL_EXT_texture_mirror_clamp
GL_EXT_texture_object
GL_EXT_texture_rectangle
GL_EXT_texture_sRGB
GL_EXT_texture_sRGB_R8
GL_EXT_texture_sRGB_RG8
GL_EXT_texture_sRGB_decode
GL_EXT_texture_shared_exponent
GL_EXT_texture_snorm
GL_EXT_texture_storage
GL_EXT_texture_swizzle
GL_EXT_timer_query
GL_EXT_transform_feedback
GL_EXT_vertex_array
GL_EXT_vertex_array_bgra
GL_EXT_vertex_attrib_64bit
GL_IBM_texture_mirrored_repeat
GL_INTEL_fragment_shader_ordering
GL_KHR_context_flush_control
GL_KHR_debug
GL_KHR_no_error
GL_KHR_parallel_shader_compile
GL_KHR_robust_buffer_access_behavior
GL_KHR_robustness
GL_KTX_buffer_region
GL_NV_alpha_to_coverage_dither_control
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_copy_depth_to_color
GL_NV_copy_image
GL_NV_depth_buffer_float
GL_NV_explicit_multisample
GL_NV_float_buffer
GL_NV_half_float
GL_NV_primitive_restart
GL_NV_shader_atomic_int64
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_OES_EGL_image
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
GL_WIN_swap_hint
WGL_EXT_swap_control

supermerill, 2021-05-01:
You forgot the project file

panik988, 2021-05-01:
Here you go
[problem.zip](https://github.com/supermerill/SuperSlicer/files/6410173/problem.zip)

supermerill, 2021-05-01:
don"t crash on my dev build, so probably fixed in the next release.
I'll link here the last nightly in some hours

panik988, 2021-05-01:
Oh thats amazing!
I'm waiting for the new nightly and I'll let you know of the problem disappear.
Thank you so much!

panik988, 2021-05-03:
I can confirm that now all work on my PC with the new nightly.
Thankyou

closed 2021-05-03 by supermerill