supermerill/SuperSlicer #1133: Add print time estimate to cover custom G-Code

AfroHorse, 2021-05-02:
### Do not request features for the Prusa3D fork of Slic3r here without confirming it is applicable to the mainline Slic3r, or your issue will be closed. [Development builds of Slic3r](https://dl.slic3r.org/dev). [Prusa3D Issue Tracker](https://github.com/Prusa3D/Slic3r/issues)

**Is your feature request related to a problem? Please describe.**
The print estimate given by SuperSlicer is used by RRF to provide an estimate for time to go. Given I run bed levelling for every print in my custom G-Code, it is always out by a few minutes. 

**Describe the solution you'd like**
A variable on the custom GCode screen (for the printer) to add additional time to print estimate

**Describe alternatives you've considered**
Seeing if RRF/DuetWebControl can process the M73 messages and use them to estimate instead.

supermerill, 2021-05-02:
printer->limit->time estimation compensation?

AfroHorse, 2021-05-02:
Potentially could work although a fixed increase would be useful as an option rather than % so you don't end up sending a 24hour print way over.

This has now evolved as the latest DuetWeb reads the M73. The issue I have found is that SuperSlicer outputs an M73 every time the % changes so on short jobs, the time keeps being reset to X minutes 10 times each minute then snaps down to X-1 and starts getting reset again. 

The mouseover the setting indicates it should drop one in every minute so unsure if this is a bug or an artefact of it being useful for some printers to update every % but the time getting reset constantly is suboptimal on others

supermerill, 2021-05-02:
I can change it to float or percent.

for the second change (M73 in minutes instead of %), I'll see when I'll rewrite this part.

sprior, 2022-04-14:
Since my issue was marked a dup of this, can you clarify exactly what I would now do to tell SuperSlicer that the PRINT_START line in start_gcode will take 10 minutes?

supermerill, 2022-04-14:
[image]
next nightly

sprior, 2022-04-14:
Oooh, nice!

closed 2024-02-11 by supermerill