supermerill/SuperSlicer #1134: Invalid notebook index and other startup errors

khumarahn, 2021-05-02:
Hi. I run superslicer on arch linux, build it from aur. At the current git state I get errors at start
[image]

Then the print settings and filament settings are unavailable.
[image]

Same with the recent appimages. Older appimages run fine.

supermerill, 2021-05-02:
you can change the layout (preferecen -> gui) to old, for now.

makhtev, 2021-05-02:
Same issue on arch
Changing layout to old solved warning messages on start but print settings and filament settings are still unavailable

supermerill, 2021-05-02:
> Changing layout to old solved warning messages on start but print settings and filament settings are still unavailable

how is that possible? the third and fourth tab do nothing?

khumarahn, 2021-05-02:
they are not there. resizing does not help

tafli, 2021-05-03:
Same here on Manjaro. Building SuperSlicer from AUR. Changing layout doesn't fix the errors.

Additionally, the wizard does not run at all:
`~/.cache/yay/superslicer-git/src/SuperSlicer/src/slic3r/GUI/ConfigWizard.cpp(2595): assert ""prusa_it != p->bundles.cend()"" failed in ConfigWizard(): Vendor PrusaResearch not found`

[image]

I tried to delete the whole configuration folder but no success with that.

Additionally adding new printer is not possible. SuperSlicer crashes:
`usr/bin/superslicer: line 9: 31514 Segmentation fault      (core dumped) LD_LIBRARY_PATH="$LD_LIBRARY_PATH:/opt/wxgtk-dev/lib" /usr/share/SuperSlicer/superslicer "$@"`

The printer itself is beeing saved but when selecting SuperSlicer crashes again with above message written to the console.

frap129, 2021-05-14:
+1 on this. Running on Arch, built from source. Switching to the old layout doesn't resolve the error or make the print/fillament settings tabs appear.

supermerill, 2021-05-20:
@tafli
you have to update the profile git submodule now (see the updated how to build doc) to get the printer profiles.

I'll push a possible fix today

frap129, 2021-05-21:
Reporting that the latest nightly fixed this issue for me. Thanks supermerill!

khumarahn, 2021-05-21:
It will be fixed in arch linux AUR when nightly is merged to master?

frap129, 2021-05-21:
> It will be fixed in arch linux AUR when nightly is merged to master?

@khumarahn The `superslicer` package will be fixed when the next version is released, `superslicer-git` will be fixed when this is merged into master. I reinstalled `superslicer-git` but added `git checkout Nigthly` after `cd "$srcdir/SuperSlicer` in the PKGBUILD to install the latest nightly.

tafli, 2021-05-22:
Confirm to be working with AUR/superslicer-prerelease-bin:2.3.55.5-1 on Manjarko. Thanks for fixing!

closed 2021-06-04 by supermerill