supermerill/SuperSlicer #1135: Custom Bed Model justifies to middle of bed definition instead of 0,0

nionio6915, 2021-05-02:
I have attempted to add a custom bed model to account for defects in my bed. No matter how I define the stl origin, the import always places the bottom left hand corner at midpoint of the bed shape. 

Attempts to re-orientate the bed model or trick the geometery to align 0,0's results in the same result - what would be the 0,0 at the lower left corner ends up at the bed center. 

[image]

Printer Preset ->
  General ->
    Size & coordinates group: Bed Shape -> 'Set'
        Bed Shape Pop-up ->
           Model group. Select 'Load' and import stl model of bed. 

Expected outcome: Bed model would import and can be aligned to 0,0.

supermerill, 2021-05-02:
I don't have your project nor your model to try it out.
I don't have any problem with my own.

liftbag, 2021-05-04:
Maybe it depends on the CAD with which you create the model. I draw on fusion360, placing the exact center of the print area at the axes origin. No positioning problem in the viewport.

[image]

nionio6915, 2021-05-05:
Sorry for my delayed resonse. 
I drew the model in Autocad.  I have some defects/gouges in my bed that effect print quality. I am attaching a dxf and dwg and the stl.  

I started the drawing so that the 0,0 of the drawing was in the bottom left hand corner, and everyhitng was in +X & +Y dimensions. 

Here is screenshot of the stl viewed in 3d Builder- it shows the center at the right (Xmax/2) & (Ymax/2) location. the Z height of the objects are 0.1mm. 

Using the stl as a bed model does not work. It justfies as I described. 
However, if i load it as an object to print, it puts it in the right place. 

[image]

and here is a screenshot of the same stl loaded as the bed model and as an object to print. 
[image]

for now, my workaround is to load it as an object to print, then toggle the object's printable property. 

[EV2_BedModel2.zip](https://github.com/supermerill/SuperSlicer/files/6429348/EV2_BedModel2.zip)

liftbag, 2021-05-06:
In Autocad, try to move the model so that the origin of the axes corresponds to the exact center of the print area.  Then export the stl.  I am sure that when you import it as a bed model it will be correctly placed.

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

nionio6915, 2021-05-06:
I did try that before and redid it based on your suggestion. Same result. I would also point out that the origin is noted in the bottom left hand corner of the bed image it he window.  Snips of Acad and SS attached.

[image]
[image]

liftbag, 2021-05-06:
> I did try that before and redid it based on your suggestion. Same result. I would also point out that the origin is noted in the bottom left hand corner of the bed image it he window. Snips of Acad and SS attached.
> When I load the moved version int KISS, it locates it correctly as well.

Can you try with this bed model? I've placed your stl in the correct place with Fusion360.

[MeshBody1.stl.zip](https://github.com/supermerill/SuperSlicer/files/6437424/MeshBody1.stl.zip)

nionio6915, 2021-05-07:
Nope, no change.  I have not had this issue with any other slicer. My models import into KISS, Preflight, nanoDLP and *cough* S3D just fine.  I have multiple stls' for some prints that are defined for different colors, supports, print settings- all generated in Autocad. As long as they are all in positive X & Y & Z coordiante system.   
And stop pushing F360, its offensive.  It should not matter what the source of the stl is. The standard is 30+ years old.

liftbag, 2021-05-07:
> Nope, no change. I have not had this issue with any other slicer. My models import into KISS, Preflight, nanoDLP and _cough_ S3D just fine. I have multiple stls' for some prints that are defined for different colors, supports, print settings- all generated in Autocad. As long as they are all in positive X & Y & Z coordiante system.
> And stop pushing F360, its offensive. It should not matter what the source of the stl is. The standard is 30+ years old.
> 
> @supermerill can you clarify the onrmal import algorythm and how/why it seems to differe from the bed model import?

If your model is correctly imported into S3D, well it's not compatible with slicer and its forks, which require that, when generating the stl, the origin of the axes corresponds with the center of the model, and not one of its angle.
I may have inadvertently zipped your model instead of mine. Unfortunately I will only be able to check this on Monday, but I can assure you that it is correctly positioned when imported as a bed model in PrusaSlicer and SuperSlicer.
I never expected to offend anyone just by quoting a semi-parametric CAD. There is always something to learn.

liftbag, 2021-05-09:
Here's what your model looks like after putting the origin of the axes in the right place.

[image]

The same model, used on S3D, is not placed correctly. SuperSlicer, PrusaSlicer and I believe also Slicer, unlike other slicers, want the origin of the stl model axes to be positioned in the center of the print area.

[image]

nionio6915, 2021-05-13:
There is a difference in how SS is importing the same stl as a print model and as you describe as a bed model to be put at the 'center'  Delta printer people understand the 'center'.  Cartesian printer people don't know what 'center' is. 

The image in the bed definition window shows the bed size and origin as defined. I don't believe there is an official standard, but the convention seems to be the bottom left hand corner is 0,0. I highlighted it in my first screenshot.  
https://user-images.githubusercontent.com/17693150/117359291-cb9a5900-ae7c-11eb-8098-a0e237326715.png

By your description, the bed needs to be defined like this- which I do not belive is correct by the .stl standard. 
[image]

I will admit that its been a while since I look at it, but I am pretty certain that the .stl standard is the object needs to be in the 1st quadrant and all position data 'positive'.

This really needs to be clarified and documented better.

nionio6915, 2021-05-20:
So this is an artifact of the originating software? 
How does a user like me push the request for documentation change to at least put 2 axis arrows in that window for prin bed 0,0 and bed model 0,0.
Ceratinly I can not be the only one who has encountered this,

supermerill, 2021-05-21:
I can add a tooltip
`The origin of the model is the center of the build plate.` ?

parallyze, 2021-05-24:
> So this is an artifact of the originating software?

Yes, it's mentioned here:
https://help.prusa3d.com/en/article/custom-bed-models_158431

> "The position of the model origin (point with coordinates x:0, y:0, z:0) needs to be in the middle of the print bed area. If you load a custom model and it appears misaligned, the origin is not set properly."

nionio6915, 2021-05-24:
no. not all CAD programs honor this anomaly. ACAD seems to put all objects in positive space as per the original STL standard. doing it this way puts coords/objects in (-x,-y) which was not previously premitted.

the bed shape configuration window shows the bed size and the bed origin axis but there is no mention of the custom bed model origin from the prusa doc linked.

what we have NOT looked at is what affect there is on the "Texture Model" for X, Y or Z!