supermerill/SuperSlicer #1136: Remove preheat nozzle and bed when using klipper macros

MrNoob0, 2021-05-02:
I have a klipper macro for my start gcode and when i use super slicer i notice that it automatically heats up the bed and nozzle if there is no heat up commands present.

I might've just missed a setting but is there a way to get rid of the heat up section without manually removing it from the gcode?

[image]

supermerill, 2021-05-02:
there will be a switch to remove them

kmarty, 2021-05-10:
The very same "issue" is in PrusaSlicer too. Workaround for this might be the same:
```
M140 S0 ; To avoid auto-adding bed heating from slicer
M104 S0 ; To avoid auto-adding nozzle heating from slicer
START_PRINT BED_TEMP=[first_layer_bed_temperature] EXTRUDER_TEMP=[first_layer_temperature]
```

closed 2021-05-17 by MrNoob0