supermerill/SuperSlicer #1138: Extra perimeters inside the fill layer.

TheAndr0id, 2021-05-05:
**Describe the bug**
Extra Perimeters are being generated under External perimeters on the top layer (see image).

**To Reproduce**

Steps to reproduce the behavior:
1. Import model
2. Slice

**3D Model and Slic3r Configuration Export**
[TestModel.zip](https://github.com/supermerill/SuperSlicer/files/6424878/TestModel.zip)

**Expected behavior**
The top surface should have a constant infill.

**Desktop (please complete the following information):**
 - OS: Linux x86
 - Version: 2.2.53.4

Correct slicing (PrusaSlicer 2.2.0):
[image]

SuperSlicer (2.2.53.4):
[image]

If one steps down through the layers you can see even under the top item it has an extra Perimeter under the top.
[image]

(I'm not expecting this to be fixed in the 2.2.x range, just to get it fixed in the new stuff... :) )

neophyl, 2021-05-05:
Its caused by having only one perimeter on top enabled.  This settings is known to cause issues.  Unfortunately its still a problem in the latest nightly I have installed (2.3.56.0).

If you turn that setting off it will slice correctly in 2.2.x

TheAndr0id, 2021-05-05:
Thank you!  I just had that setting on since it is a default.  I actually like having the multiple perimeters on top.  This fixes my issue, but still leaves an issue for SuperSlicer...  :/

supermerill, 2021-05-05:
It's due to the combinaison of "min width">0 and "one peri on top". it tries to see where it is needed from the letters, by growing & reducing them. In the process, they are merged and formed a big blob that can't be shrunk back to nothing.

TheAndr0id, 2021-05-05:
So which "min width" are you talking about?  The min_width_top_surface is set to the default value of 200%,  thin_walls_min_width is set to it's default of 100%, and I tried the slicing with thin_walls on and off and it made no difference.  (thin_walls is kind of my go-to setting to check when things slice sideways)

I'm sorry to drag this out, but it's kind of bugging me.  Is it because it's trying to create a single perimeter around the letters and since they are too small it gets confused?  Should it be doing that in any case since it's not an external perimeter?  (I've read the popup text for min_width_top_surface about 5 times now and it's making my head hurt)

I'm not trying to be a pain, I'm happy just turning that setting off to avoid the problem.  I have seen many postings that the "One perimeter on top" feature is the main reason many people are using SuperSlicer.  (I came for the ironing, stayed for the extra infill patterns... :D )

supermerill, 2021-05-20:
> So which "min width" are you talking about?

that one
[image]

To comply this setting, it has to get the upper layer, and do some transformations on it. These transformations can produce weirds shapes if there are only thins things next to each others, like letters.

TheAndr0id, 2021-05-21:
Ok, that's fair.  I was doing weird things with thin walls, and the "one perimeter" code didn't like it.  I gave it a shot with a Minimum Width  setting of 0 and that removed the artifacts on the layer under the text.  It's still sliced different than if the one perimeter setting is turned off, but the results seem to be usable.   I'll play around with it the next time I'm printing this model.

Thanks for your help!

closed 2021-05-21 by supermerill