supermerill/SuperSlicer #1140: Last nightly doesn't start

Liger0, 2021-05-07:
### Do not report Prusa3D Slic3r bugs here without confirming it is a problem on a development release of Slic3r, or your issue will be closed. [Development builds of Slic3r](https://dl.slic3r.org/dev). [Prusa3D Issue Tracker](https://github.com/Prusa3D/Slic3r/issues)

**Describe the bug**
The last nightly (commit cf80732) doesn't work

**To Reproduce**
Steps to reproduce the behavior:
1. Try to open it

**3D Model and Slic3r Configuration Export**
Please upload a ZIP archive containing a copy of the 3D model you are seeing this bug with and a CTRL-E export of the configuration used.

**Expected behavior**
Start

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10 x64
 - Version [e.g. 1.3.0]: Nightly cf80732

**Additional context**
Add any other context about the problem here.

th33xitus, 2021-05-08:
Can't confirm.
Works for me.

Sineos, 2021-05-08:
https://github.com/supermerill/SuperSlicer/runs/2444500043 works here as well (Win64)

salocain, 2021-05-08:
he doesn't start on linux --> splash screen and.....nothing

Liger0, 2021-05-08:
> 
> 
> https://github.com/supermerill/SuperSlicer/runs/2444500043 works here as well (Win64)

Well that's an old build, not the one I reported.

supermerill, 2021-05-20:
It's weird as the only change on that is a deletion of a macos-only section.

Liger0, 2021-05-20:
Not totally sure if it's of any meaning, but all the path is just reported as 'a'

Liger0, 2021-05-20:
I confirm that the commit 23a708f is also broken.

th33xitus, 2021-05-21:
> 
> 
> I confirm that the commit [23a708f](https://github.com/supermerill/SuperSlicer/commit/23a708f075c72f971761ee3786b5a5a7c17d5396) is also broken.

https://github.com/supermerill/SuperSlicer/suites/2791890190/artifacts/62002145

I can't confirm (again).
Works fine for me... Win10 Pro x64

neophyl, 2021-05-21:
Works for me on Win 10 too.  Running in english though and not italian like the OP

Liger0, 2021-05-21:
I'm running english.

neophyl, 2021-05-21:
Sorry,I really shouldn't make assumptions but I saw Nascondi dettagli from your pic I thought it sounded/looked Italian.

Liger0, 2021-05-21:
Well, the windows installation is italian, the slicer is english.

Liger0, 2021-05-21:
> 
> 
> he doesn't start on linux --> splash screen and.....nothing

Indeed, the splash screen is displayed, the crash happens after it.

Liger0, 2021-05-22:
The beta  2.3.56.0 is unfortunately also broken :(

supermerill, 2021-05-22:
I just replicate this bug on one of my computer

closed 2021-05-27 by Liger0