supermerill/SuperSlicer #1141: Layer heights with variable layer height

Liger0, 2021-05-07:
### Do not report Prusa3D Slic3r bugs here without confirming it is a problem on a development release of Slic3r, or your issue will be closed. [Development builds of Slic3r](https://dl.slic3r.org/dev). [Prusa3D Issue Tracker](https://github.com/Prusa3D/Slic3r/issues)

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Printer cr10s
2. Enable variable layer height
3. Layer height limits: 0.04 to 0.32
4. 0.12 layer height base

**3D Model and Slic3r Configuration Export**

[config.zip](https://github.com/supermerill/SuperSlicer/files/6442791/config.zip)

**Expected behavior**
Use layer heights between the limits

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10 x64
 - Version Nightly 580669e 
 - 
**Additional context**
Add any other context about the problem here.

neophyl, 2021-05-12:
The zip file contains only a config.ini file.  You need to save your project from in SuSi, then take the saved 3mf file and zip that up and attach it here.  The 3mf contains the model and all your slicer settings, the model position, any modifiers etc etc.

Without a 3mf file its hard to debug things.

Liger0, 2021-05-12:
For some reasons now it's different, and it only doesn't respect the max layer height, filling with something greater. I don't know if that's expected behaviour or not, and how to decide to allow it or not.
[image]
[Max LH.zip](https://github.com/supermerill/SuperSlicer/files/6467692/Max.LH.zip)

neophyl, 2021-05-12:
Everything is working as it should, there is no issue with SuSi.

Both PS/Susi treat overhangs the same way.  As overhangs are not printed on the layer below they are assumed to be printed at your nozzle diameter.  As such no matter what your printed layer height is configured to they are always (in this case ) assumed to be 0.4 thick.  If you look at the normal preview the dark blue of overhang perimeters match your red areas when looking at the layer height preview.  The same is true of the bottom of the model.  As the bottom isn't flat you are printing in midair for over half the bottom.  I suggest you use either the place to face tool or the cut function to trim off 0.2-0.4mm from the bottom.  I'd use cut as the base and the model look poorly combined and cut will give you a proper flat first layer.

Support layer height is NOT tied in any way to model print height.  It is totally independent in most circumstances, the slicer will vary the height of supports even if variable layer height on the model is not used.  It combines layers etc to get what it considers a best fit to speed up printing while still providing support.  This is completely normal.  If you look at the preview slider for the layers you will see the total number of layers change depending on if you are using support or not.  They only match the model when synchronize to object layers is enabled.  Which I believe is only an option with soluble supports (ie contact z distance=0)

Good luck printing this as I suspect your supports are going to be a royal pain in the rear to remove, especially as you have them configured.

Liger0, 2021-05-12:
I understand, thanks. Why doesn't it go as low as 0.04mm as configured though?

neophyl, 2021-05-12:
The project you attached has not had any variable layer heights applied to the model so why should it go below your configured layer height ?  or above it ?  All of the object layers are your base layer height of 0.12mm except for the overhang perimeters which as explained are 0.4 thick.

Support layers can and do vary without anything being applied to them as that's how the software works.  

If you want the model to have variable layer heights then you need to select it and use the Variable Layer Height tool on it.  You can then manually or automatically change layer heights at different parts of the model height.

As for the extrusion that is overhang Super Slicer and Prusa Slicer have always printed unsupported layers at the nozzle diameter.  Think about how your printer actually works.  It squirts out a line of plastic that is pretty much the diameter of your nozzle.  Its only the fact that its normally squirted out with a thin lower gap below (your desired layer height) it that makes it that layer height.  The plastic hits the layer below and squishes out sideways.  When it lays down the next perimeter it does so with a slight overlap to the perimeter next to it, thats how things stick together.  If there's no plastic below then it cant make it thin like a normal layer can it.  
With support below the layer it could squish it into that but if you do that your supports are then welded to the part and wont come off, they will be the same as a normal lower layer.  So even when there's support below it leaves a gap between support and extrusion so its still assuming that its the nozzle diameter as it comes out. It plans for that in the slicing engine.

Liger0, 2021-05-13:
Oh well, thanks for explaining. Tho the height being 0.4 means it will extrude more and weld too much with the supports than it does with other slicers, that's weird.

closed 2021-05-13 by Liger0