supermerill/SuperSlicer #1143: e637a9e Nigthly  in linux (not tested in windows) gui part lost ..

salocain, 2021-05-08:
**Describe the bug**

**To Reproduce**
just load the nightly 

**Desktop (please complete the following information):**
 - OS:Linux

firtst capture --> the last working nightly (Merge branch 'merill-merge' C/C++ CI ubuntu #237)
the nightly after have this problems

[image]

the second --> the result --> some gui part lost

[image]

nerdCopter, 2021-05-18:
+1   ...i waited for settling of `master`, but no changes for sometime...
I can live with bed-graphic missing, but the Filament and Printer menus fail completely.

Ubuntu 20.04.2 LTS.   
[Separately, i can no longer even launch it in Debian -- segfault]

for completeness sake, when starting with no profile/config the following unusable dialog shows as well:
[image]

salocain, 2021-05-27:
The last release work well 😁😁

closed 2021-05-27 by supermerill

nerdCopter, 2021-05-27:
Confirmed working Debian 10.
TYVM!