supermerill/SuperSlicer #1145: poor overhang performance, but only when the slicer labels it as an "Overhang perimeter"

actuallylemoncurd, 2021-05-08:
### Do not report Prusa3D Slic3r bugs here without confirming it is a problem on a development release of Slic3r, or your issue will be closed. [Development builds of Slic3r](https://dl.slic3r.org/dev). [Prusa3D Issue Tracker](https://github.com/Prusa3D/Slic3r/issues)

**Describe the bug**
When printing a 3D Benchy, it labels the bottom part of the overhang in the front of the ship as an "Overhang Perimeter"
You can see on the printed part exactly where it is labelled as such in the slicer, you can tell the slicer is doing something different there, appears to be extrusion width related but I am unsure at this time

https://discord.com/channels/771316156203270154/771319294113742858/840705638559776809

**To Reproduce**
Steps to reproduce the behavior:
Open 3dbenchy stl
slice and observe the overhang perimeter in the front
Print it, see error/artifact

**3D Model and Slic3r Configuration Export**
In the discord link above

**Expected behavior**
for this area to be printed like the rest of the perimeters

**Screenshots**
In the discord link above

**Desktop (please complete the following information):**
Windows 10, 20H2 build

**Additional context**
Add any other context about the problem here.

actuallylemoncurd, 2021-05-08:
[image]
[image]

CCS86, 2021-05-13:
Check your setting for this:

[image]

If it is set below 100%, than an overhanging wall can get bridge flow applied, which is a huge increase in flow rate.

actuallylemoncurd, 2021-05-13:
Ok interesting point

I was unable to find that exact settings panel you posted but I did find this
[image]
the number circled used to be 50%. i bumped it up by 5% until i got this on the benchy
[image]
no longer detected as a "bridge" so the flow rate is the same

thank you for pointing this out, i guess this was not a bug but incompetence as this slicer has WAY more variables than any other i've used

awesome work

closed 2021-05-13 by actuallylemoncurd

CCS86, 2021-05-13:
You might want to make that 100%, so only true bridges are detected as bridges. 

Any of those lines in blue, are treated as bridges, so the flow rate is calculated on making a 0.4mm diameter bead, instead of a 0.4 x [layer height] bead. It's a big increase.

actuallylemoncurd, 2021-05-13:
ehh, i quite like how it slows the flow rate of the nozzle down when on the harse overhangs so ill just change that as needed per print

CCS86, 2021-05-13:
It doesn't "slow the flow rate". If you are using a 0.4mm nozzle and 0.15mm layer height, using bridge flow (100%) doubles your flow rate.

supermerill, 2021-05-20:
maybe a "min length" for overhang detection may be useful to avoid these "spikes" when it's just at the threshold limit.

CCS86, 2021-05-21:
I think that's a good idea.