supermerill/SuperSlicer #1146: vertical shell thickness not as configured

tcurdt, 2021-05-09:
**Describe the bug**

Does not ensure vertical shell thickness despite being checked.

**To Reproduce**

I was slicing https://www.thingiverse.com/thing:3714978

**3D Model and Slic3r Configuration Export**

[Archive.zip](https://github.com/supermerill/SuperSlicer/files/6447432/Archive.zip)

**Expected behavior**

Just on the left circled wall, the right circled wall should have the configured amount of perimeters and be solid.

**Screenshots**

[image]

**Desktop (please complete the following information):**
 - OS: macOS 10.15.7
 - Version 2.3.55.5
 
**Additional context**

neophyl, 2021-05-12:
Turn OFF one Only perimeter on top surfaces and it will slice it correctly and fill that area with the extra perimeters and gapfill as you want it.  

btw this seems to slice ok with the nightly (2.3.56.0) so looks like its already fixed.

tcurdt, 2021-05-12:
> Turn OFF one Only perimeter on top surfaces and it will slice it correctly and fill that area with the extra perimeters and gapfill as you want it.

I didn't think that the top surface setting would be of interest here. Interesting.
So it was error-sly assuming it was all a top surface on the right?

> btw this seems to slice ok with the nightly (2.3.56.0) so looks like its already fixed.

Oh, perfect. That means we can close this.
Is there a ETA for 2.3.56?

neophyl, 2021-05-12:
Yeah that setting has been very buggy in recent versions.  There have been several issues opened where just turning it off fixed things.  Its the first thing I check for now lol.

No idea on the release schedule but there were so many changes in PS2.3.0 that seem to need refactoring for SuSi that I'm not surprised its taking awhile.  I guess we might get a release just around the same time PS goes to 2.4, murphy's law ;)

tcurdt, 2021-05-12:
> Yeah that setting has been very buggy in recent versions. There have been several issues opened where just turning it off fixed things. Its the first thing I check for now lol.

I'll keep that in mind for the next issue! 

> No idea on the release schedule but there were so many changes in PS2.3.0 that seem to need refactoring for SuSi that I'm not surprised its taking awhile. I guess we might get a release just around the same time PS goes to 2.4, murphy's law ;)

Of course! :)

closed 2021-05-12 by tcurdt