supermerill/SuperSlicer #1148: App crashes when slicing attached file

Miksu80, 2021-05-12:
Superslicer crashes without any notification when slicing the project attached. Also when doing 0.7mm default extrusion width, there are gaps in slicing.

Superslicer Version 2.3.55 daily precompiled pack + Win10

**To Reproduce**
Steps to reproduce the behavior:
1. Open attached project
2. slice

[Arrowpack.zip](https://github.com/supermerill/SuperSlicer/files/6464242/Arrowpack.zip)
[config.txt](https://github.com/supermerill/SuperSlicer/files/6464270/config.txt)

neophyl, 2021-05-12:
fyi if you save the 3mf then that contains the model and all your settings used to slice it so the stl and config arent usually required but in this case I'm glad they are there - see later.

Using Windows 10 and Super Slicer 2.3.55.5 I can duplicate the crash.  It seems to generate the basic preview but crashes before it can display the colour coded display.  
I did find that turning off One loop perimeters allows it to slice fine and work for me though with 2.3.55.5

I then tried the nightly from a week or so ago (v2.3.56.0).  When I load the 3mf file into 2.3.56 the single loop perimeter option is unticked for some reason. Which is strange as the project loaded into 2.3.55.5 has it ticked and the saved config.txt also has it enabled.  

What is also strange is that if I load the 3mf into 2.3.56.0 and then enable the one loop perimeter it slices fine and does not crash.  
However if I load the project in 2.3.56.0 and then load the config from the text file (which does enable the one loop perimeter setting) and THEN slice it crashes out like 2.3.55.5 does.

The only differences I can find between config.txt and the SuperSlicer config in the 3mf file are the 4 'print_host = ' settings which are there but not defined in the text file but missing from the 3mf config.  That may be normal though.

Theres other weirdness going on too for me.  If I load in the project and then select a different print profile I get the usual do you want to discard or keep values box, but that box seems to think the projects setting for Bridge Flow Ratio is set to 1 not 100%.  If I cancel and go back to the gui the gui is showing 100 but its also showing the reset arrow.  If I click on that reset arrow it does briefly change to 1 and then back to 100.  Not sure whats going on there.

Guess this is for someone who knows more than me :)

On the extrusion width issue do you have gaps in your printed parts too ?  I'm wondering if its a preview parser issue due to the fact you are using a 0.4mm nozzle with such a wide extrusion width.

Miksu80, 2021-05-12:
It crashes by these settings, if changing printing width, its ok and other setting changes make it work also so hard to tell which one is causing it to fail..

I didn't check gap version by printing, but the uncolored preview showed the bridges OK and they dissapeared (along with the layers where they are) on colored preview. When manually examining the gcode produced, there weren't any layer losses so it's got something to do with gcode interpreter.

Pavulon87, 2021-05-12:
I sliced the attached file without a problem in version 2.3.56.

supermerill, 2021-05-20:
> What is also strange is that if I load the 3mf into 2.3.56.0 and then enable the one loop perimeter it slices fine and does not crash.
However if I load the project in 2.3.56.0 and then load the config from the text file (which does enable the one loop perimeter setting) and THEN slice it crashes out like 2.3.55.5 does.

can't even makes it crash with that procedure.

closed 2021-06-09 by Miksu80