supermerill/SuperSlicer #1151: "Fillamentum PLA" profile not showing up for many printers

EV3R4, 2021-05-14:
**Describe the bug**
"Fillamentum PLA" isn't showing up for some printer profiles (it's only showing up for "Original Prusa i3 MK3S MMU2S **Single**").

**To Reproduce**
* Add "Fillamentum PLA" as filament profile.
* Go to/add some other printer than "Original Prusa i3 MK3S MMU2S **Single**" (e.g. "Original Prusa i3 MK3S MMU2S").
* Look if its available.

**Expected behavior**
I expected the filament profile to show up on other printer profiles than the "Original Prusa i3 MK3S MMU2S **Single**".

**Desktop (please complete the following information):**
 - OS: Windows
 - Version: 2.3.55.5-alpha

**Additional context**
I think the problem is this line.
`compatible_printers_condition = ! (printer_notes=~/.*PRINTER_VENDOR_PRUSA3D.*/ and printer_notes=~/.*PRINTER_MODEL_MK(2.5|3).*/ and single_extruder_multi_material)`
Because its only available for the **single** version of the "Original Prusa i3 MK3S MMU2S" profile and it might not be interpreting the **exclamation mark** correctly.

neophyl, 2021-05-14:
I've just added the Mk3s MMU2s to my list of printers using PS2.3.1 and Susi 2.3.55.  I also added the Filamentum PLA too.  

In both programs when I have the MMU2 Single selected the filament list has "Fillamentum PLA" listed and available.  When I select MMU2 multi printer though and go to the filament list the filament listed is "Fillamentum PLA @MMU2" so its a different profile for the same filament that differs in some way between single and multi.  On my install it looks like the MMU multi profiles are ALL different as far as the software is concerned.  That's Prusa vendor spaces for you.
Make sure you have both installed. 

From a quick scan it looks like settings for multimaterial are different between single/multi profiles, even the printing temps are slightly different so they are by no means identical profiles.

supermerill, 2021-05-20:
That's the way prusa do their profiles.
Is there is a big problem with the prusa profiles? If not I'm not going to tune theirs.