supermerill/SuperSlicer #1152: Infill tab gui error

Liger0, 2021-05-14:
### Do not report Prusa3D Slic3r bugs here without confirming it is a problem on a development release of Slic3r, or your issue will be closed. [Development builds of Slic3r](https://dl.slic3r.org/dev). [Prusa3D Issue Tracker](https://github.com/Prusa3D/Slic3r/issues)

**Describe the bug**
The options for connection lenght and perimeter anchor in the infill tab are too much on the right and the dropdown is too small

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Print settings
2. Go to the infill tab

**3D Model and Slic3r Configuration Export**
Please upload a ZIP archive containing a copy of the 3D model you are seeing this bug with and a CTRL-E export of the configuration used.

**Expected behavior**
Aligned style and big enough dropdown

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windwos 10 x64
 - Version Nightly 2.3.56

**Additional context**
Add any other context about the problem here.

supermerill, 2021-05-20:
You can change that by modifying the `resources/ui_layout/print.ui` file.
If you succeed to get a better layout, I'll merge it.

Sirage48, 2021-05-24:
This is my print.ui updated for the latest "Spacing edition".  A work-in-progress and occasionally changed as I set and forget some settings.
I did a global insert "simple" into all settings, changed the settings used once or rarely to expert, and then change some occasionally to advanced if I think they are not used very often.
The widths are set for a 27in 4k monitor with non-modal gui.
For each update I grep to cleanly extract and sort all the settings in this file and the updated version and take a difference to find any new settings.
[print.ui.zip](https://github.com/supermerill/SuperSlicer/files/6529853/print.ui.zip)

Liger0, 2021-05-24:
> 
> 
> This is my print.ui updated for the latest "Spacing edition". A work-in-progress and occasionally changed as I set and forget some settings.

Everything looks like this here

[image]

Sirage48, 2021-05-25:
The issue is translation and your Italian has different length words / expressions to English.  There is also a correction I will do in the Extras and Overhangs as I missed a few widths in this latest version.
Probably the default layout has lots of flexibility in widths  that work across different languages. I prefer to have a cleaner layout that suits the way I think when I create and test profiles and that requires settings with lots of fixed widths.  Access to the UI is a fantastic feature of Superslicer.

If you like this layout I could do an Italian version - it shouldn't take too long - mainly fixing a few widths, and maybe splitting a few lines if the text is too long.

A bit of fiddling with widths and font spacing.  Might be easier with a mono spaced font but that is not available and maybe cause other problems. Best to let supermerril focus on fixing bugs at the moment.

Feedback on layout and grouping of settings appreciated - obviously different to the default.

On my screen.

[image]

Sirage48, 2021-05-25:
Is this any good?
italian_print.ui.zip

Liger0, 2021-05-25:
I honestly use the expert tab and I feel good with the stock organization of it.