supermerill/SuperSlicer #1153: Clearer width & flow

Liger0, 2021-05-14:
### Do not report Prusa3D Slic3r bugs here without confirming it is a problem on a development release of Slic3r, or your issue will be closed. [Development builds of Slic3r](https://dl.slic3r.org/dev). [Prusa3D Issue Tracker](https://github.com/Prusa3D/Slic3r/issues)

**Describe the bug**
I think how the width and flow are managed is too confusing.
I simply tried to reach a manual width = nozzle diameter = spacing like all other slicers do, so not having overlap between lines, and I am not being able to do so. Some option have a positive overlap, some have negative one.
I tried setting the width manually to 0.4, 0 (to default from nozzle diameter) and even 100%, and the spacing too.
But even so I wasn't able to reach the "default" wanted result of extruding simply the nozzle diameter width with no advanced overlap, it either was lower or higher than 0.4.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Print settings
2. Go do width and flow
3. Try to have extrusion width = nozzle diameter = spacing between lines
4. Can't find a way to do so

**3D Model and Slic3r Configuration Export**
Please upload a ZIP archive containing a copy of the 3D model you are seeing this bug with and a CTRL-E export of the configuration used.

**Expected behavior**
Being able to set a simpled default behaviour of manual width = nozzle diameter = spacing

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows 10 x64
 - Version nightly 2.3.56

**Additional context**
Add any other context about the problem here.

CCS86, 2021-05-17:
We have already had some in-depth discussion on this. The next release will have better clarity/control on this. Or, grab a nightly if you are impatient.

Liger0, 2021-05-18:
I am using the last nightly already.

CCS86, 2021-05-19:
What you are asking for won't functionally work well. Read through this:

https://manual.slic3r.org/advanced/flow-math

Liger0, 2021-05-19:
> 
> 
> What you are asking for won't functionally work well. Read through this:
> 
> https://manual.slic3r.org/advanced/flow-math

Why so? It's literally what every other slicer does.

CCS86, 2021-05-19:
It's right here: 

[image]

Liger0, 2021-05-19:
but why can't it be set to 0 anyway?

CCS86, 2021-05-19:
Overlap being set to 100% means all that yellow area gets filled. Setting it to 0% means none of it gets filled.

supermerill, 2021-05-20:
[image]'ll add some pictures instead of the old text.

I'm open to a more detailed & comprehensive help /explanation text.

Liger0, 2021-05-20:
The part that confused me is that it stated that changing one of the 2 disables the other, which was counterintuitive since it is not true, as they apply together to be the same in the end result to follow the other.

supermerill, 2021-05-21:
I can replace `Setting the spacing will deactivate the width setting, and vice versa`
by `Setting the spacing will make the width setting computed from this one`
Someone can create a better sentence?

CCS86, 2021-05-23:
"You can set **either** 'Spacing', **or** 'Width'; the other will be calculated, using the specified 'Overlap' percentages"

closed 2021-06-04 by supermerill