supermerill/SuperSlicer #1154: Cannot set Z speed independently

martinbudden, 2021-05-15:
This is the same bug as https://github.com/prusa3d/PrusaSlicer/issues/2945 for which I have submitted a PR in PrusaSlicer , but not had a reply.

**Describe the bug**
The z travel speed is the same as the xy travel speed,  so when the xy travel speed is high, you get z-banding or even z-axis skipping.

I have a fix for this and so this bug report is just to see if you are interested in the fix - if so I will submit the PR. I'm also happy to submit it for review without any expectation that it will be accepted.

For reference, the PrusaSlicer PR is here https://github.com/prusa3d/PrusaSlicer/pull/6369

supermerill, 2021-05-20:
Do you think it has any consequences on a print profile?
Isn't it more useful on the printer machine limits tab?

balkin, 2021-05-26:
@supermerill Hi! I'm not the TS but we just had a discussion about this issue and I'm willing to let you know what I've found. Here's a fragment of code generated by slic3r++, actually prusa does exactly the same:
```
;LAYER_CHANGE
;Z:0.2
;HEIGHT:0.2
; custom gcode: before_layer_gcode
;LAYER:0
M117 INDICATOR-Layer0
; custom gcode end: before_layer_gcode

G1 Z0.200000 F7200.000 ; move to next layer (0)
G1 E-4.00000 F2100.000 ; retract
```

Perhaps `G1 Z0.200000 F7200.000` is really not very beautiful. Although I'm perfectly fine with the superslicer's gcode, my printer's firmware ignores velocities greater than upper limit of my printer.
Maybe instead of using `travel_speed` for Z velocity you could use `machine_max_feedrate_z`?

supermerill, 2021-05-26:
> Maybe instead of using travel_speed for Z velocity you could use `machine_max_feedrate_z`?

Maybe, or i can create a `machine_z_travel_speed` next to it.
I'll have to see if it's useful to create an other one. I have some 3D-path that may want to use a slow `machine_max_feedrate_z`, and travel that want to use a fast `machine_z_travel_speed`

balkin, 2021-05-26:
If that helps, here are my settings from the gcode file:
```; allow_empty_layers = 0
; avoid_crossing_not_first_layer = 1
; avoid_crossing_perimeters = 0
; avoid_crossing_perimeters_max_detour = 0
; bed_custom_model = C:\\Program Files\\Ultimaker Cura 4.6\\resources\\meshes\\ulti_steel_platform.stl
; bed_custom_texture = 
; bed_shape = 0x0,200x0,200x200,0x200
; bed_temperature = 105
; before_layer_gcode = ;LAYER:[layer_num]\n
; between_objects_gcode = 
; bottom_fill_pattern = monotonic
; bottom_solid_layers = 5
; bottom_solid_min_thickness = 0
; bridge_acceleration = 1000
; bridge_angle = 0
; bridge_fan_speed = 100
; bridge_flow_ratio = 35%
; bridge_overlap = 100%
; bridge_speed = 25
; bridge_speed_internal = 150%
; bridged_infill_margin = 200%
; brim_ears = 1
; brim_ears_detection_length = 1
; brim_ears_max_angle = 125
; brim_ears_pattern = concentric
; brim_inside_holes = 0
; brim_offset = 0
; brim_width = 6
; brim_width_interior = 0
; chamber_temperature = 0
; clip_multipart_objects = 0
; color_change_gcode = M600
; complete_objects = 0
; complete_objects_one_skirt = 0
; complete_objects_sort = object
; cooling = 1
; cooling_tube_length = 5
; cooling_tube_retraction = 91.5
; curve_smoothing_angle_concave = 0
; curve_smoothing_angle_convex = 0
; curve_smoothing_cutoff_dist = 2
; curve_smoothing_precision = 0
; default_acceleration = 3000
; default_filament_profile = ""
; default_print_profile = 
; deretract_speed = 35
; disable_fan_first_layers = 3
; dont_support_bridges = 1
; draft_shield = 0
; duplicate_distance = 6
; end_filament_gcode = "; Filament-specific end gcode \n;END gcode for filament\n"
; end_gcode = M104 S0 ; turn off temperature\nM140 S0 ; turn off heatbed\nM107 ; turn off fan\n{if layer_z < max_print_height}G1 Z{z_offset+min(layer_z+10, max_print_height)} F240{endif} ; Move print head up\nG1 X0 Y0 F5000;\nM84 X Y Z E ; disable motors
; enforce_full_fill_volume = 1
; ensure_vertical_shell_thickness = 0
; exact_last_layer_height = 0
; external_infill_margin = 150%
; external_perimeter_cut_corners = 0%
; external_perimeter_extrusion_spacing = 
; external_perimeter_extrusion_width = 0
; external_perimeter_fan_speed = -1
; external_perimeter_overlap = 100%
; external_perimeter_speed = 45
; external_perimeters_first = 0
; external_perimeters_hole = 1
; external_perimeters_nothole = 1
; external_perimeters_vase = 0
; extra_loading_move = -2
; extra_perimeters = 1
; extra_perimeters_odd_layers = 0
[... 304 lines cut ...]
; wiping_volumes_matrix = 0
; xy_inner_size_compensation = 0
; xy_size_compensation = 0
; z_offset = 0
; z_step = 0.005
; BEGIN DISPLAYLAYERPROGRESS SETTINGS
; DisplayLayerProgress_layerIndicatorProcessed = true
; END   DISPLAYLAYERPROGRESS SETTINGS
```

And actually the second layer is exactly the same:
```;LAYER_CHANGE
;Z:0.3
;HEIGHT:0.1
; custom gcode: before_layer_gcode
;LAYER:1
M117 INDICATOR-Layer1
; custom gcode end: before_layer_gcode

@Objectstop robocar - Helly.stl id:0 copy 0
M204 S3000 ; adjust acceleration
G1 Z0.300000 F7200.000 ; move to next layer (1)
```

I'm using klipper (and it ignores 120mm/s on z movements), but probably some firmwares may not ignore it.

martinbudden, 2021-05-31:
Sorry the delay in replying.

> Maybe instead of using travel_speed for Z velocity you could use `machine_max_feedrate_z`?

That was my initial idea, however, from the (limited) discussion on the PrusaSlicer fork, there seemed a reluctance to use `machine_max_feedrate_z` for this purpose.

Also, `machine_max_feedrate_z` is not directly accessible by `GCodeWriter`, so it would certainly be a more complicated PR to do it this way.

My feeling is that an additional option of `travel_speed_z` is more likely to get accepted as a PR for the PrusaSlicer.

Note that Marlin (unlike Klipper, it seems) definitely does not honour the value of `machine_max_feedrate_z`, so the z feedrate does need to be limited by the slicer.

supermerill, 2021-06-02:
I'm ok for a pr to add `travel_speed_z`

martinbudden, 2021-06-02:
OK, I'll get my PR from PrucaSlicer moved over here.

martinbudden, 2021-06-07:
Shouldn't this be closed now?

closed 2021-06-07 by supermerill