supermerill/SuperSlicer #1156: Not really an issue... Where's Supermerill?

LaserBee, 2021-05-16:
Not an issue or a bug... 

The dev used to update quite regularly. There hasn't been an update since Feb. Does anyone have contact with Supermerill? Is he okay? Is he still working on this, or is he busy with other stuff? 

I'm not concerned in that my slicer works great, I just am curious if it's still being developed, and want to make sure the dev is doing well! :)

Thanks.

ReubenBTalbott, 2021-05-16:
Someone on the SuperSlicer Discord said that Supermerill’s computer died and that he’ll be back soon.

Liger0, 2021-05-16:
last nightly is from 10 days ago (altough it doesn't start so you have to use the one before it)

LaserBee, 2021-05-17:
Ah okay, thanks guys. I didn't even realise nightlies were a thing!

supermerill, 2021-05-20:
problem is there are too many bugs to output a proper release.

LaserBee, 2021-05-20:
That's okay, there's no rush. I;m glad you're still around! :)

closed 2021-06-04 by supermerill