supermerill/SuperSlicer #1157: Completely different overhang / infill behavior when part is rotated 90 degrees

DasBootU96, 2021-05-16:
### Do not report Prusa3D Slic3r bugs here without confirming it is a problem on a development release of Slic3r, or your issue will be closed. [Development builds of Slic3r](https://dl.slic3r.org/dev). [Prusa3D Issue Tracker](https://github.com/Prusa3D/Slic3r/issues)

**Describe the bug**
Completely different overhang / infill behavior when part is rotated 90 degrees

**To Reproduce**
Steps to reproduce the behavior:
1. copy / download https://github.com/VoronDesign/Voron-2/blob/Voron2.4/STLs/VORON2.4/Gantry/X_Axis/X_Carriage/Printheads/TriangleLab_Dragon/printhead_front_dragon.stl

2. drag drop the printhead_front_dragon.stl in SS. Make a copy of printhead_front_dragon.stl or drag & drop printhead_front_dragon.stl again in SS. You now have identical printhead_front_dragon.stl in SS.

3. rotate one of the printhead_front_dragon.stl parts 90 degrees and slice

4. examine layer 130 (@0.2mm) with overhangs. For the rotated part, overhangs are printed in mid air and shows perimeter lines where there should be infill. 

**3D Model and Slic3r Configuration Export**
Please upload a ZIP archive containing a copy of the 3D model you are seeing this bug with and a CTRL-E export of the configuration used.
[printhead_front_dragon.zip](https://github.com/supermerill/SuperSlicer/files/6489714/printhead_front_dragon.zip)

**Expected behavior**
Expected behavior is that overhangs are not printed in mid air and infill is not printed as perimeters.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

[printhead_front_dragon.emf.zip](https://github.com/supermerill/SuperSlicer/files/6489723/printhead_front_dragon.emf.zip)

**Desktop (please complete the following information):**
 - OS: [e.g. MacOS, Linux, Windows]
 - Version [e.g. 1.3.0]
     - _Do not report Prusa3D Slic3r bugs here without confirming it is a problem on a development release of Slic3r, or your issue will be closed. *Only* use normal Slic3r version IDs._
     - _For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_
     - _Use `About->About Slic3r` for release versions._

**Additional context**
Issue seen in SS 2.3.56.0 and SS 2.3.55.2. Issue not present in PS2.3.1

DasBootU96, 2021-05-17:
Yes, turning off ``extra perimeters: [v] on overhangs`` gives the *exact* behavior you mentioned:

[image]

thanks @neophyl

supermerill, 2021-05-20:
that's a bug in the bridge detection, as it shouldn't fill bridge area.

supermerill, 2021-10-02:
I checked and it's because the bridge angle override is set to 180°, so bridges can only have one direction, hence the different behavior if rotated.