supermerill/SuperSlicer #1158: I would like to ask for the update of the reference of the profiles folder

jfestrada, 2021-05-17:
Hi, 

We want to use SuperSlicer as our main slicer for the [VERTEX XL](https://github.com/Vertex3dPrinters/VERTEX-XL) because of that I opened a PR with our profiles to [slic3r-profiles](https://github.com/slic3r/slic3r-profiles) and was accepted and merged into the main repository. So now we want to start the testing with the MACOS, Linux, and Windows x64 versions. Because of that, I'm asking you to update the reference of the profiles folder which is inside of [resources folder](https://github.com/supermerill/SuperSlicer/tree/master/resources) which would be very helpful.

Best regards
Juanfra

supermerill, 2021-05-20:
It's done automatically. If your local build doesn't have it, do a git-submodule update.

jfestrada, 2021-05-23:
Hi, many thanks for the tip.

closed 2021-05-23 by jfestrada